libretiny-eu/libretiny #183: ERROR Error auth result: Error: Authentication invalid.

kurtdefeyter, 2023-10-09:
Please help the noob
I have no idea why i get this , would someone be kind enough
[logs_kickstart-bk7231n_run.txt](https://github.com/libretiny-eu/libretiny/files/12850541/logs_kickstart-bk7231n_run.txt)
 to point me in a direction 
Many thanks

Cossid, 2023-10-09:
It appears you're trying to OTA on a device the currently is running ESPHome with an OTA password.
You must include the old password, (and set the new one to blank if that is your intention)

https://esphome.io/components/ota.html#updating-the-password use simply `""` for a blank password.

kurtdefeyter, 2023-10-09:
ok , allow me to ask one more stupid question ...at what stage did i set that password , i did put esp home on it with cloutcutter ?

Cossid, 2023-10-09:
Yes, if you compiled a custom ESPHome firmware, it was included in that.

kurtdefeyter, 2023-10-10:
i am afraid that i lost the device , i went thru all the logs in cloudcutter and i never have set a password during the process., so its end of the line i am afraid