libretiny-eu/libretiny #184: Compilation error when using WiFiClientSecure

TuanAIoT, 2023-10-10:
I use CBU module with platformio.ini configuration
[env:cbu]
platform = libretiny
board = cbu
framework = arduino
upload_speed = 921600
monitor_speed = 115200
lib_compat_mode = off
lib_deps = 256dpi/MQTT@^2.5.1

Compile and Program operate normally and declare:
WiFiClient WifiClient;

But got a compilation error when declaring:
WiFiClientSecure WifiClientSecure;

Can someone help me fix it

[image]
[image]
[image]

TuanAIoT, 2023-10-10:
I saw an article where the same error occurred with Esp32, but with the CBU module and libretiny platform, I don't know how to fix it.
https://github.com/espressif/arduino-esp32/issues/5977

kuba2k2, 2023-10-10:
The error from the linked issue is not related and completely different.

This issue has been found before and resolved, but it's in a different branch (not finished yet). For now, you can paste the following block of code into your program:
```cpp
__attribute__((weak)) String ipToString(const IPAddress &ip) {
	char szRet[16];
	sprintf(szRet, "%hhu.%hhu.%hhu.%hhu", ip[0], ip[1], ip[2], ip[3]);
	return String(szRet);
}
```
Make sure to put it in a `.cpp` file, and not in a `.c` file.

TuanAIoT, 2023-10-10:
What is that code and where should I paste it?

kuba2k2, 2023-10-10:
This is the missing function. Paste it somewhere in your `.cpp` program file. For example after the line `WiFiClientSecure WifiClientSecure`. Also, it's better to name it something like `clientSecure` because your current name is too similar to the class name.

TuanAIoT, 2023-10-10:
You mean paste it right after the line declaring WiFiClientSecure clientSecure;
However, there is still a compilation error
[image]
[image]

kuba2k2, 2023-10-10:
Yes, that's what I meant, However, the secure client is not tested and not properly implemented on Beken. You might want to use raw mbedTLS directly instead.

TuanAIoT, 2023-10-11:
I did as you said and continued to comment line 241 and compiled successfully
[image]

TuanAIoT, 2023-10-11:
But Error connecting to MQTTS server
[image]

closed 2023-10-16 by TuanAIoT

kuba2k2, 2023-10-16:
You can't just comment out code - it's there for a reason. As I mentioned earlier, WifiClientSecure is not fully implemented and might not work at all, fail to compile, or even crash the chip.

TuanAIoT, 2023-10-17:
> Bạn không thể chỉ nhận xét mã - nó có lý do. Như tôi đã đề cập trước đó, WifiClientSecure chưa được triển khai đầy đủ và có thể không hoạt động, không biên dịch được hoặc thậm chí làm hỏng chip.
So is there any way I can connect with MQTT SSL/TLS?

kuba2k2, 2023-10-17:
Try using ESPHome instead. Maybe it supports MQTT with TLS.

TuanAIoT, 2023-10-17:
I think I prefer programming on the libretiny platform.
MQTT SSL/TLS connectivity is essential for IoT projects.
Are you continuing to develop on this topic, because I think the MbedTLSClient library is quite complete, there's just a little more left.
Everyone also hopes that your project can replace Tuya SDK. I don't like TuyaOS and the Tuya WindIDE compiler