libretiny-eu/libretiny #185: Error connect MQTT SSL/TLS

TuanAIoT, 2023-10-12:
I use CBU module, with 256dpi/MQTT@^2.5.1 library
I connected MQTT using the command client.begin("broker.emqx.io",1883 , client);
But I have problems connecting with MQTT SSL/TLS.
Don't know what the error is and how to fix it.

Thank you very much!

[image]
[image]
[image]

TuanAIoT, 2023-10-13:
I use esp32 wroom with the PubSubClient library and it can completely connect to MQTTS.
But PubSubClient doesn't support libretiny.
That's because libretiny doesn't have the WiFiClientSecure library but instead the MbedTLSClient library. The CA_Cert, Client_Cert, Client_Key authentication program is having problems.
Can anyone with an MQTTS connection help me replace the PubSubClient library with another similar library?

TuanAIoT, 2023-10-14:
@kuba2k2 Can you help me, any answer can help me continue the project.
i want to be connected with MQTT SSL/TLS
Thanks a lot!

TuanAIoT, 2023-10-14:
When I use the PubSubClient and WiFiClient libraries, when SSL/TLS is not needed, the CBU module can still connect to MQTT normally.
Only when using WiFiClientSecure does it return an error status of authentication failed

TuanAIoT, 2023-10-16:
Can someone help me with this problem, please
@kuba2k2

TuanAIoT, 2023-11-01:
Hello @kuba2k2 ,
Can the MQTT SSL/TLS connection error be fixed and can I submit the cost to your team to fix it?

TuanAIoT, 2023-11-25:
Hi @kuba2k2 ,
I posted on the TuyaOS forum to ask, but it seems that the BK7231N does not support connecting to MQTT SSL/TLS
https://www.tuyaos.com/viewtopic.php?t=2432

Cossid, 2023-11-25:
That is a TuyaOS/SDK limitation, not a BK7231N/T limitation.  MQTT on TuyaOS does support TLS/SSL, just not with external servers, as they data they send over that wire is encrypted separately with private keys set up by their OS.

kuba2k2, 2023-11-25:
We're not using TuyaOS in LibreTiny. Of course Beken can support TLS - the mbedTLS library is there. It's just a matter of configuring it right and adapting the MQTT library to use it properly.

TuanAIoT, 2023-12-01:
How to properly configure and tune the MQTT library so it can work with SSL/TLS? Sadly, I couldn't find documentation that could resolve this issue

HieuNM59, 2024-08-07:
Hi @TuanAIoT , I also have a project that needs MQTT with TLS on the beken BK7231N chip. Have you succeeded with this? Can you give me the documentation or instructions for connecting MQTTS?