libretiny-eu/libretiny #186: Tuya WBR3

guestisp, 2023-10-15:
Guys, is the Tuya WBR3 chipset (i think it's a RTL8720CM) supported ? 
https://solution.tuya.com/hardware/detail/59006

I can see support for the RTL8720CM but in the Tuya devices list the WBR3 is not present.

kuba2k2, 2023-10-15:
RTL8720, whether is CM or CF, is not supported yet. See #44 .

closed 2023-10-15 by kuba2k2

guestisp, 2023-10-15:
So, why RTL8720CF is listed here as supported ? 
[image]

kuba2k2, 2023-10-15:
Because a development version (generic chip) was added to test and fix support for it. The core features work, but many important things such as OTA are not supported. See [Feature Support](https://docs.libretiny.eu/docs/status/supported/#feature-support) - RTL8720C is listed there.

guestisp, 2023-10-15:
Uhm, probably i' could work with that anyway. I just need to trigger a change on a GPIO. (it's the ilonda wifi fish feeder, tuya based, not sure which chipset anyway, i'll know better tomorrow. It could be and esp8266 or this RTL)