libretiny-eu/libretiny #188: Difference between RTL877710BN 468k and 7788k

Weissnix4711, 2023-10-18:
Hi, sorry if I'm being exceptionally dumb here, but what's the difference between the `generic-rtl8710bn-2mb-468k` and `generic-rtl8710bn-2mb-788k` boards? The board definition json for each seems identical (other than the name), and I can't figure out what the numbers actually refer to.

Cossid, 2023-10-18:
The difference is the partition layout, which affects how large the app partition is, and in effect, how large of an OTA the device can handle.

You can see the corresponding flash layout on each format's docs page;
468k: https://docs.libretiny.eu/boards/generic-rtl8710bn-2mb-468k/#flash-memory-map
788k: https://docs.libretiny.eu/boards/generic-rtl8710bn-2mb-788k/#flash-memory-map

The 468k layout has a large user storage section.

If you are only OTAing, you are stuck with and bound to your current partition sizing.  If you are serial flashing, you can choose the optimal layout for your needs.

There are no pin or hardware differences between the two.

closed 2023-10-18 by Cossid

Weissnix4711, 2023-10-19:
Gotcha, thanks.