libretiny-eu/libretiny #190: Lost The Ability to Pair RF433

banter7702, 2023-10-24:
Hi, I have a generic 4 channel smart relay:
[image]

I have successfuly backed up the original firmare dump and flashed it with esphome. I had to break button 3 and 4 connections temporarily because they are connected to rx and tx.

However, I lost the ability to pair it with RF remote. Previously, in tuya app, in the device settings, there is an option to add rf remote per button.

I guess there is no pairing method thru holding button.

Is there anything I could do to pair remote for each relay?

kuba2k2, 2023-10-24:
You probably need to add the "remote_receiver" component to ESPHome. This device doesn't look like it has TuyaMCU, so the 433 MHz receiver chip is probably directly connected to a GPIO. You need to find that GPIO and add an appropriate component.

banter7702, 2023-10-24:
> You probably need to add the "remote_receiver" component to ESPHome. This device doesn't look like it has TuyaMCU, so the 433 MHz receiver chip is probably directly connected to a GPIO. You need to find that GPIO and add an appropriate component.

Is there a way to find the gpio thru firmware dump? Just like ltchiptool that grabs automatically? Is there a detailed guide about this?

kuba2k2, 2023-10-24:
We might be able to find the GPIO if you post the firmware dump here (or "Source data" from ltchiptool).

kuba2k2, 2023-10-24:
If you've paired it to Tuya before dumping, your wifi SSID and password will be there. They're encrypted, so you can't redact it easily.

However, you can post the source data from ltchiptool. It is unencrypted, and you'll find the SSID and password fields there - just mask them out and you'll be fine.

banter7702, 2023-10-24:
> If you've paired it to Tuya before dumping, your wifi SSID and password will be there. They're encrypted, so you can't redact it easily.
>
> However, you can post the source data from ltchiptool. It is [...]

Hi,

Is this what we need? I dont know what I'm looking at, sorry. I hope we can find what we need to restore the RF functionality.
Please tell me if I masked a lot of info so I could restore it:

```
{
    "gw_bi": {
        "uuid": "redacted",
        "psk_key": "redacted",
        "auth_key": "redacted",
        "ap_ssid": "SmartLife",
        "ap_passwd": null,
        "country_code": "CN",
        "bt_mac": null,
        "bt_hid": null,
        "prod_test": false,
        "fac_pin": "redacted"
    },
    "user_param_key": {
        "bt1_lv": 0,
        "bt1_pin": 8,
        "bt2_lv": 0,
        "bt2_pin": 9,
        "bt3_lv": 0,
        "bt3_pin": 10,
        "bt4_lv": 0,
        "bt4_pin": 11,
        "ch_cddpid1": 9,
        "ch_cddpid2": 10,
        "ch_cddpid3": 11,
        "ch_cddpid4": 12,
        "ch_dpid1": 1,
        "ch_dpid2": 2,
        "ch_dpid3": 3,
        "ch_dpid4": 4,
        "ch_num": 4,
        "crc": 109,
        "cyc_dpid": 43,
        "inch_dp": 44,
        "init_conf": 38,
        "jv": 100,
        "module": "CB3S",
        "net_trig": 4,
        "net_type": 0,
        "netled1_lv": 0,
        "netled1_pin": 22,
        "netn_led": 0,
        "nety_led": 0,
        "on_off_cnt": 10,
        "onoff_clear_t": 6,
        "onoff_rst_m": 0,
        "onoff_rst_type": 2,
        "onoff_type": 2,
        "rand_dpid": 42,
        "remote_add_dp": 49,
        "remote_io": 20,
        "remote_list_dp": 50,
        "remote_select": 1,
        "reset_t": 5,
        "rf_width": 345,
        "rl1_lv": 1,
        "rl1_pin": 24,
        "rl2_lv": 1,
        "rl2_pin": 6,
        "rl3_lv": 1,
[... 223 lines cut ...]
            "type": "raw"
        },
        {
            "mode": "rw",
            "id": 49,
            "type": "raw"
        },
        {
            "mode": "rw",
            "id": 50,
            "type": "raw"
        }
    ],
    "ble_beaconkey": "redacted",
    "em_sys_env": "bk7231n",
    "wd_cfg_mem": {
        "power_cfg": 0,
        "key_type_cfg": 2
    },
    "mf_test_close": true,
    "wd_lock_mem": {
        "ctrl_lock_cfg": "AA8AAAAAAAA="
    }
}
```

kuba2k2, 2023-10-24:
Yes, that's it. I believe the `remote_io` parameter might be the GPIO of the RF module - in this case it's P20. That matches your photo:
[image]

banter7702, 2023-10-24:
> Yes, that's it. I believe the `remote_io` parameter might be the GPIO of the RF module - in this case it's P20. That matches your photo: [image]

Please be patient with me. I have no idea what to do with that info. What should I do? 

This is some part of my esp.yaml, if that info has something to do with it:
```
text_sensor:
  - platform: libretiny
    version:
      name: LibreTiny Version

binary_sensor:
  - platform: gpio
    id: binary_switch_1
    pin:
      number: P8
      inverted: true
      mode: INPUT_PULLUP
    on_press:
      then:
        - switch.toggle: switch_1
  - platform: gpio
    id: binary_switch_2
    pin:
      number: P9
      inverted: true
      mode: INPUT_PULLUP
    on_press:
      then:
        - switch.toggle: switch_2
  - platform: gpio
    id: binary_switch_3
    pin:
      number: P10
      inverted: true
      mode: INPUT_PULLUP
    on_press:
      then:
        - switch.toggle: switch_3
  - platform: gpio
    id: binary_switch_4
    pin:
      number: P11
      inverted: true
      mode: INPUT_PULLUP
    on_press:
      then:
        - switch.toggle: switch_4
  - platform: gpio
    id: binary_switch_all
    pin:
      number: P7
      inverted: true
      mode: INPUT_PULLUP
    on_press:
      then:
        - switch.toggle: switch_1
        - switch.toggle: switch_2
        - switch.toggle: switch_3
        - switch.toggle: switch_4

switch:
  - platform: gpio
    id: switch_1
    name: Relay 1
    pin: P24
  - platform: gpio
    id: switch_2
    name: Relay 2
    pin: P6
  - platform: gpio
    id: switch_3
    name: Relay 3
    pin: P26
  - platform: gpio
    id: switch_4
    name: Relay 4
    pin: P14

status_led:
  pin:
    number: P22
    inverted: true

``` 

Im sorry, Im so new to this. Please bear with me. I really need to restore the rf functionality and at the same time use esphome.

kuba2k2, 2023-10-24:
You need to add the Remote Receiver component on pin P20. Here are the docs:
https://esphome.io/components/remote_receiver

I don't know what kind of remotes you want to use with this. You'll have to try your luck and test various protocols of that component. The docs have everything explained for you.

banter7702, 2023-10-24:
Ok, I'll update this topic tomorrow. It seems doable for me. Thank you very much!

banter7702, 2023-10-25:
> You need to add the Remote Receiver component on pin P20. Here are the docs: https://esphome.io/components/remote_receiver
>
> I don't know what kind of remotes you want to use with this. You'll have [...]

Hi, update on this topic.

I've figured it out. Before posting this topic, I was about to give up already and buy another tuya relay. Thank you very much. I've learned a lot!

badrpc, 2024-01-06:
> I've figured it out. Before posting this topic, I was about to give up already and buy another tuya relay. Thank you very much. I've learned a lot!

Would you mind posting configuration that worked for you? I have what looks like identical relay model. I will really appreciate it if you can share your setup.

RaW79, 2024-03-19:
@banter7702 @badrpc 

I´m testing with same type of module and may I ask for your feedback what code worked for you?

In addition, could the RF chip be identified what is used on this boards?  On my boards the RF SOT8 chip is blank/unprinted. Would be great to know what is used. Thx in advance!

badrpc, 2024-05-03:
On my board RF chip is labeled 590R-2153. I suppose this is SYN590R - the logo on the chip mathces the one you can find in internet.

I currently have the following configuration:

```
# Example device configuration for TUYA TYWB 4CH
# 4 channel relay with WIFI
esphome:
  name: test-tywb-4ch

bk72xx:
  board: cb3s

# From original firmware boot log:
#
# ******* ch[0] *******
# [RELAY] - pin:24,        io_drive:2
# [KEY] - pin:8,   io_drive:1,     type:0,         long_press:5000
# ******* ch[1] *******
# [RELAY] - pin:6,         io_drive:2
# [KEY] - pin:9,   io_drive:1,     type:0,         long_press:5000
# ******* ch[2] *******
# [RELAY] - pin:26,        io_drive:2
# [KEY] - pin:10,  io_drive:1,     type:0,         long_press:5000
# ******* ch[3] *******
# [RELAY] - pin:14,        io_drive:2
# [KEY] - pin:11,  io_drive:1,     type:0,         long_press:5000
#
# ******* system *******
# [WIFI LED] - pin:22,     io_drive:1
# [TOTAL KEY] - pin:7,     io_drive:0,     long_press:5
# [433 REMOTER] - pin:20,  freq_min:345
# [WIFI MTHD]-5, [WIFI LED]-1, [KEY TRIG]-0

switch:
  - platform: gpio
    name: Relay 1
    pin: P24

  - platform: gpio
    name: Relay 2
    pin: P6

  - platform: gpio
    name: Relay 3
    pin: P26

  - platform: gpio
    name: Relay 4
    pin: P14

binary_sensor:
  - platform: gpio
    pin:
      number: P8
      inverted: true
    name: "Button 1"

  - platform: gpio
    pin:
      number: P9
      inverted: true
    name: "Button 2"

  - platform: gpio
    pin:
      number: P10
      inverted: true
    name: "Button 3"
    filters:
      - delayed_on: 50ms
      - delayed_off: 50ms

  - platform: gpio
    pin:
      number: P11
      inverted: true
    name: "Button 4"
    filters:
      - delayed_on: 50ms
      - delayed_off: 50ms

  - platform: gpio
    pin:
      number: P7
      inverted: true
    name: "Mode switch key"

light:
  - platform: status_led
    name: "Status LED"
    pin:
      number: P22
      inverted: true

# Not entirely clear how to use this.
# Requires further investigation.
remote_receiver:
  pin:
    number: P20
    #inverted: true
    mode:
      input: true
      pullup: true
  dump:
    # - aeha
    # - byronsx
    # - canalsat
    # - canalsatld
    # - coolix
    # - dish
    # - drayton
    # - jvc
    # - haier
    # - lg
    # - magiquest
    # - midea
    # - nec
    # - nexa
    # - panasonic
    # - pioneer
    # # - pronto
    # # - raw
    # - rc5
    # - rc6
    - rc_switch
    # - samsung
    # - samsung36
    # - sony
    # - toshiba_ac
```

With this configuration when I press buttons on the remote I can see the following in the logs:
```
19:49:10 | [I] | [remote.rc_switch:261] | Received RCSwitch Raw: protocol=1 data='101011010011010111101000'
19:49:10 | [I] | [remote.rc_switch:261] | Received RCSwitch Raw: protocol=1 data='101011010011010111101000'
19:49:10 | [I] | [remote.rc_switch:261] | Received RCSwitch Raw: protocol=1 data='101011010011010111'
19:49:13 | [I] | [remote.rc_switch:261] | Received RCSwitch Raw: protocol=1 data='101011010011010111100100'
19:49:13 | [I] | [remote.rc_switch:261] | Received RCSwitch Raw: protocol=1 data='101011010011010111100100'
19:49:13 | [I] | [remote.rc_switch:261] | Received RCSwitch Raw: protocol=1 data='101011010011010111100100'
19:49:13 | [I] | [remote.rc_switch:261] | Received RCSwitch Raw: protocol=1 data='101011010'
19:49:17 | [I] | [remote.rc_switch:261] | Received RCSwitch Raw: protocol=1 data='10101101001'
19:49:17 | [I] | [remote.rc_switch:261] | Received RCSwitch Raw: protocol=1 data='101011010011010111100010'
19:49:17 | [I] | [remote.rc_switch:261] | Received RCSwitch Raw: protocol=1 data='101011010011010111100010'
19:49:17 | [I] | [remote.rc_switch:261] | Received RCSwitch Raw: protocol=1 data='1010110100'
19:49:18 | [I] | [remote.rc_switch:261] | Received RCSwitch Raw: protocol=1 data='101011010011010111100001'
19:49:18 | [I] | [remote.rc_switch:261] | Received RCSwitch Raw: protocol=1 data='101011010011010111'
```

I didn't go further then simple testing with this device yet, but I think the following worked for me at some point:
```
binary_sensor:
  # https://esphome.io/components/remote_transmitter#remote-transmitter-rc-switch-protocol
  - platform: remote_receiver
    name: "Remote button A"
    rc_switch_raw:
      code: '101011010011010111101000'
      protocol: 1
```

badrpc, 2024-09-24:
> Hi, I have a model exactly like yours, I don't know how to configure it yet, can you share?

@truongbber please clarify your question. I inserted almost a full config in my post - should be enough to get started with configuration. If you have general problems flashing this device with ESPHome it would be better to post your question on https://community.home-assistant.io/c/esphome/ (feel free to tag me there if you'd like to here my specific experience - same user name as here).

In general this specific device is rather straightforward with the exception that one need to isolate console TX/RX from resistors and capacitors which interfere with console. I chose to de-solder the two resistors and two capacitors from the board (don't really need on-device buttons but I can still use them with some software de-bouncing). Alternatively as mentioned in https://github.com/libretiny-eu/libretiny/issues/190#issue-1958583197 you can cut the traces and then restore them later.

Please make sure that you have exactly the same chip (my board has [CB3S](https://developer.tuya.com/en/docs/iot/cb3s?id=Kai94mec0s076) which is based on BK7231N) and check relevant documentation:
* https://docs.libretiny.eu/
* https://esphome.io/components/libretiny