libretiny-eu/libretiny #192: Generic RGBWW Ceiling Light Issues

banter7702, 2023-10-30:
Hi, its me with some remote issues again.

I flashed a generic rgbww ceiling light, identical with this one:
[[DUMP] [CB3S] [BK7231N] Eacam 24W RGBCW Ceiling ](https://github.com/tuya-cloudcutter/tuya-cloudcutter/issues/361)

So, I had this same initial problem in that topic where pins cant be found or extracted, I used esphome kickstarter and ltchiptool to extract it but same problem.

So I manually search it and found all the rgb cw ww pwms.

But it has problems. The rgb is not optimal. Sometimes it will turn off if I fully go to 2700k or 6500k, I had to go to red to make color temp work properly. Same goes to fully going down to lowest brightness, it turns off.

And my biggest gripe is that, I lost the remote functionality. We'll, if it would be me, its fine, but the people in the house needs it. How do I restore it? And some of its fancy features such as scenes?

Here is my esphome config:

```
esphome:
  name: ceiling-lamp
  friendly_name: Ceiling Lamp

bk72xx:
  board: generic-bk7231n-qfn32-tuya

# Enable logging
logger:

preferences:
  flash_write_interval: 10s

# Enable Home Assistant API
api:
  encryption:
    key: "redacted"

ota:
  password: "redacted"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot in case wifi connection fails
  ap:
    ssid: "Ceiling-Lamp Fallback Hotspot"
    password: "redacted"

output:
  - platform: ledc
    id: output_red
    pin: P8
  - platform: ledc
    id: output_green
    pin: P24
  - platform: ledc
    id: output_blue
    pin: P26
  - platform: ledc
    id: output_cold
    pin: P7
  - platform: ledc
    id: output_warm
    pin: P6

light:
  - platform: rgbww
    id: light_rgbww
    name: Light
    color_interlock: true
    cold_white_color_temperature: 6500 K
    warm_white_color_temperature: 2700 K
    red: output_red
    green: output_green
    blue: output_blue
    cold_white: output_cold
    warm_white: output_warm
    restore_mode: RESTORE_DEFAULT_OFF
``` 

Here are some pics for reference:
[image]
[image]

ferbulous, 2023-10-30:
Is that using rf or ir remote? 
you’ll need to enable receiver, capture the codes for each button and do automation for each button in the yaml.

banter7702, 2023-10-30:
> Is that using rf or ir remote? you’ll need to enable receiver, capture the codes for each button and do automation for each button in the yaml.

Im not really sure. How to confirm? The remote emits visible blue light when pressing. Is if possible its bluetooth?

ferbulous, 2023-10-30:
This [link](https://shopee.com.my/product/193268172/22130317290?gad_source=1&gclid=Cj0KCQjwqP2pBhDMARIsAJQ0Czp_7QMna_5JAQKyqK3EFvQT9DBPAeDvhEdt7JI-8_lX3JNeu3UbWo4aAr68EALw_wcB) seems to be similar to your model, and apparently it's rf.

[image]

Did you see any rf receiver on the pcb?

banter7702, 2023-10-30:
> This [link](https://shopee.com.my/product/193268172/22130317290?gad_source=1&gclid=Cj0KCQjwqP2pBhDMARIsAJQ0Czp_7QMna_5JAQKyqK3EFvQT9DBPAeDvhEdt7JI-8_lX3JNeu3UbWo4aAr68EALw_wcB) seems to be similar to your model, and apparently it's rf.
> 
> [image]
> 
> Did you see any rf receiver on the pcb?

Yes, exactly that, exactly same picture of ads. Im confident they are the same product. So this must be IR, correct?

Here is the full picture of the board, nothing else at the back. Im not really sure if there is an IR receiver. IR receiver should be obvious, like some coil of antenna wire, but there might some other form:

[image]

[image]

ferbulous, 2023-10-31:
I don't see any, have you tried capturing the remote codes with wifi remote/broadlink just to verify if it's rf or ir?

banter7702, 2023-10-31:
Do I need some equipment with that?

I think we can rule out IR since the remote is emitting blue light?

Is it possible that remote is via wifi/bluetooth? Can I confirm this with esphome?