libretiny-eu/libretiny #196: Wifi disconnecting and not coming back up

lmcd, 2023-11-03:
I'm working with ESPHome on an RGB bulb - and if I interact with the bulb for a bit, scrubbing through colours etc, the connection will eventually drop and not come back up without a power cycle.

Sometimes I get about 20 seconds out of it, sometimes a few minutes, but will always eventually disconnect.

I can run a continuous effect on the bulb and trigger a disconnect, and the effect will continue running, so I think this confirms it hasn't crashed or anything.

Unfortunately I can't debug over serial as the UART pads are inaccessible. I flashed the device with tuya-cloudcutter.

The chip is a `bk7231n` based. Specifically it's a CB2L: https://developer.tuya.com/en/docs/iot/cb2l-module-datasheet?id=Kai2eku1m3pyl

lmcd, 2023-11-03:
Also worth noting that I was using these bulbs for months with the stock Tuya firmware, and never experienced any connectivity issues.

lmcd, 2023-11-06:
I have noticed this is only occurring when when the logger is active via `esphome run`
If I ctrl+c after an `esphome run`, the bulb will never go down. If I leave the logger running, it'll last a minute or two before requiring a power cycle.

alexkramer98, 2023-12-20:
Update: looks like it's a wifi issue. Some lights that have <=2/5 bars of wifi randomly disconnect. Some never get back online. The 2.4Ghz wifi channels are very crowded where I live, so maybe that has something to do with it. 

Never experienced this issue using the stock Tuya firmware (or other devices) though (same location). Solved by placing an additional wifi access point closer to the disconnecting bulbs.

coltc50, 2023-12-23:
I had the same experience and fixed it also with a additional access point but I wish it get fixed from the ground...