libretiny-eu/libretiny #197: Issues when adding more than 7 devices?

grax71, 2023-11-03:
Hi! I have installed the add-on and then added a number of devices based on BEKEN BK7231N.
The devices are named from tuya-01 to tuya-07.
All works fine, except when adding the 8th device tuya-08. The add-on starts behaving  strange. This is what I experienced:

1. The Device console of Home Assistant start visualizing few of the already provisioned "tuya-n" devices as to be re-verified and asks for encryption key. Even if the encryption key is given, there is an error message. Then the device disappears from the list of devices to be verified.
2. The new device "tuya-08" appears in the device list, but then tuya-01 disappears.

So, it looks that these is some kind of (addressing?) issue in the number of devices. Or maybe the naming? Not sure.

Thanks.

kuba2k2, 2023-11-03:
I'm not sure I understand what you mean. What kind of add on are you using? What device list do you mean? How can it just disappear from the list without removing it explicitly?

That being said, it's most likely not a LibreTiny issue, but ESPHome (I assume that's what you're using).

grax71, 2023-11-03:
Hi! Yes let me be more detailed. I am using the "Libretiny ESP home add-on":

[image]

Then, the devices added are of course devices which are using that ADD-on and this behaviour is experienced in the "Device List" of ESP Home:

[image]

So, it is clear that there is some issue in the merge. So do you suggest me to post this in ESPHome or Home Assistant thread?

Thanks!

Cossid, 2023-11-03:
You should migrate all your devices to the official ESPHome addon, which now fully support LibreTiny, and is the only one being updated moving forward.

grax71, 2023-11-03:
Thanks for the hint. But I have one issue. You said "You should migrate all the devices to ESPHome addon which now supports Libretiny". But if I try to add a libretiny-supported device, then in the ESPhome add-on I don´t have the option. This is the list of devices type supported by official ESPHome add-on. As you see, the "Libretiny" device type is not reported in the list. I have latest version of ESPHome add.on installed.

[image]

While if I try to add one device on the "LibreTiny ESP Home" add-on, then the list allows me to add one device of the category "Libretiny". here is the list:

[image]

And then I can go through the further list of LibreTiny devices type:

[image]

So, is it my standard ESPHome add-on that has some issue and does not shows the LIbreTiny devices types, or it is my correct understanding that in order to add LibreTiny devices, I have to add the LibreTiny ESP Add-on, which is a different add-on than the standard ESP-Home add-on?

[image]

Thanks for kind help.

Cossid, 2023-11-03:
You can just pick any, you're going to replace the entire config after creation anyway.

grax71, 2023-11-03:
Sorry @Cossid  I did not get your comment. Is there a way to extend my standard ESPHome Add-on including the management of libretiny-compliant boards? Today I have two add-ons with different devices listed. If I could have both ESP and non-ESP devices in the same add-on this will be simpler. Is there a description on how to get there? Thanks

Cossid, 2023-11-03:
The ESPHome official addon already supports LibreTiny, it just doesn't have an option in the new device list.  You can selected any of the ones it does present, and after it creates the device, you just edit and replace the entire config with what you had in LibreTiny.

After you move your configs, you will not need a separate LibreTiny addon any longer.

grax71, 2023-11-03:
Thanks @Cossid , crystal clear!
Ok I think this issue can be closed, I will migrate all in the standard ESPhome and then check if I have same kind of issue. Thanks again for kind help!

closed 2023-11-03 by Cossid