libretiny-eu/libretiny #198: restore_mode not working

nedyarrd, 2023-11-04:
It looks like restore mode, or anything else what needs to be writen to persistent memory is not working. I have LSC bulbs flashed with LibreTiny ESPHome. When I try to set restore_mode it makes only "on" and "off" like stated, but don't remember last state and state of color temperature.

Cossid, 2023-11-04:
`restore_mode` works fine on every device I have.  "on" and "off" are not valid settings, see https://esphome.io/components/light/index.html?highlight=restore_mode.  Additionally, settings are not saved instantly, by default, they are only updated every 60 seconds to avoid flash wearing.  Additionally if you use anything like groups, they may be overriding the restored settings when they sync.

zoidy, 2024-10-31:
> Edit: there are also quirks as to when save triggers when the mode changes, but the on/off state does not. You may want to consider calling the preferences save when you set the state you desire.

Like in the original post, `restore_mode` doesn't work for me (I have `CMA19-60W-AL-9ACK ` lights). Could you elaborate on how to do preferences save?

Cossid, 2024-10-31:
There was a [bug found](https://github.com/esphome/esphome/pull/7642) and patched in 2024.10.1 that sometimes led to preferences not being saved, so start by updating to that version or newer.

Additionally, you can add a button to manually save preferences on demand;
```yaml
button:
  - platform: template
    id: save_preferences
    name: "Save Preferences"
    entity_category: diagnostic
    on_press:
      - lambda: global_preferences->sync();
```

zoidy, 2024-10-31:
Thanks, I was running the September build. I upgraded to 2024.10.2 and it seems to have fixed the issue. Using `restore_mode = RESTORE_DEFAULT_ON` now restores the color and brightness of the bulb after cutting power and restoring it.

felipecrs, 2024-11-05:
Is there a way to completely disable saving preferences to flash, like [`restore_from_flash`](https://esphome.io/components/esp8266#:~:text=package%20to%20use.-,restore_from_flash,-(Optional%2C%20boolean)?

Cossid, 2024-11-05:
> Is there a way to completely disable saving preferences to flash, like [`restore_from_flash: false`](https://esphome.io/components/esp8266)?

I'm not certain if there is a way to completely disable it, but there are two things you can do to reduce it in ESPHome.

First is changing the [`flash_write_interval`](https://esphome.io/components/esphome#adjusting-flash-writes) to `never` (note, `0s` just removes delays, `never` is a special case).

Secondly, you can set `restore_mode` to either `ALWAYS_ON` or `ALWAYS_OFF`, which should reduce state tracking.

felipecrs, 2024-11-05:
> Secondly, you can set `restore_mode` to either `ALWAYS_ON` or `ALWAYS_OFF`, which should reduce state tracking.

This is what I'm doing currently.

Thanks for the hint about [flash_write_interval](https://esphome.io/components/esphome#adjusting-flash-writes).

kuba2k2, 2024-11-05:
No - this option is specific to ESP8266, where there apparently is some other kind of (semi-)non-volatile memory that is used to persist settings across reboots and OTA updates. There is no other kind of memory in LibreTiny, so this option would be equivalent to setting `restore_mode` and/or `flash_write_interval`.

felipecrs, 2024-11-05:
Awesome. Thanks a lot.

felipecrs, 2024-11-08:
@Cossid, @kuba2k2, there's no `never` for `flash_write_interval`:

[image]

And also, I have set all my switches and lights to either `ALWAYS_ON` and `ALWAYS_OFF`, but I still see this in my device logs:

```
00:59:02 | [I] | [safe_mode:041] | Boot seems successful; resetting boot loop counter
00:59:02 | [D] | [lt.preferences:104] | Saving 1 preferences to flash...
00:59:02 | [D] | [lt.preferences:132] | Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
```

It seems to only happen in initialization/startup though.

Cossid, 2024-11-08:
Interesting, https://esphome.io/components/esphome.html#adjusting-flash-writes it is a `Time` object, and `never` is a valid option according to https://esphome.io/guides/configuration-types#config-time
I believe it just sets it to int.Max or int.Max-1 though.  They must have missed a validation case (or don't have it properly documented when it can an cannot be used)

Preferences will also write when given a reboot command (like before an OTA).  I guess just set it to a high value.