libretiny-eu/libretiny #202: BL0937 not working with wb2s

brunosalezze, 2023-11-08:
I have a smart power plug with a WB2S chipset and a BL0937 power/current/voltage monitor. Libretiny works well to sense button and control LED and Relay, but the BL0937 readings only return zeros. The pinout is correct since I opened the device and tested with OpenBeken. With OpenBeken the readings are perfect. My YAML is the following:

bk72xx:
  board: wb2s

logger:
  level: DEBUG

sensor:
  # Power sensor
  - platform: hlw8012
    model: BL0937
    sel_pin:
      number: PWM4
      inverted: true
    cf_pin: PWM1
    cf1_pin: PWM2
    change_mode_every: 1
    update_interval: 5s
    # Current sensor
    current:
      name: BL0937 Current
      accuracy_decimals: 8
      filters:
        - multiply: 1
    # Voltage sensor
    voltage:
      name: BL0937 Voltage
      accuracy_decimals: 8
    # Power sensor
    power:
      id: power
      name: BL0937 Power
      accuracy_decimals: 8
    # Energy sensor
    energy:
      id: energy
      name: BL0937 Energy
      accuracy_decimals: 8

  - platform: adc
    pin: ADC3
    name: WB2S ADC
    update_interval: 5s
    filters:
        - multiply: 100

binary_sensor:
  # Binary sensor for the button press
  - platform: gpio
    name: button
    pin:
      number: RX1
      inverted: false
    on_press:
      - switch.toggle: relay

switch:
  # Switch to toggle the relay
  - platform: gpio
    id: relay
    name: switch
    pin: PWM5
    inverted: true
    on_turn_on:
      - switch.turn_on: led
    on_turn_off:
      - switch.turn_off: led

  - platform: gpio
    id: led
    name: led
    pin: PWM0
    inverted: false`

and the OpenBeken config is the following:

`"vendor": "Tuya",
  "bDetailed": "0",
  "name": "Full Device Name Here",
  "model": "enter short model name here",
  "chip": "BK7231T",
  "board": "TODO",
  "flags": "33554432",
  "keywords": [
    "TODO",
    "TODO",
    "TODO"
  ],
  "pins": {
    "6": "LED;57",
    "7": "BL0937CF;56",
    "8": "BL0937CF1;55",
    "10": "Btn_n;53",
    "23": "ADC;0",
    "24": "BL0937SEL;9",
    "26": "Rel;37"
  },`

Any help for further debuging?

Cossid, 2023-11-08:
It has been the general experience that you need to invert the pins for sel, cf, and cf1 on the Beken platform.  No idea why, but that is how Tuya generally has the configs as well.

brunosalezze, 2023-11-08:
Thank you, with this config the device works as expected:

sensor:
  # Power sensor
  - platform: hlw8012
    model: BL0937
    voltage_divider: 883.278
    current_resistor: 0.000908 ohm
    sel_pin:
      number: PWM4
      inverted: true
    cf_pin: 
      number: PWM1
      inverted: true
    cf1_pin:
      number: PWM2
      inverted: true
    change_mode_every: 3
    update_interval: 5s

closed 2023-11-08 by brunosalezze