libretiny-eu/libretiny #205: New board: T112_v1.1

Weissnix4711, 2023-11-11:
Found in Teckin smart plug model SP23. Posting this info here in the hopes someone else is gonna add this board, or at the very least that it may be useful to someone in the future. I don't currently have the time, nor frankly the inclination, to do so myself.

It can of course be flashed as a generic rtl8710bn, but adding to the docs would be nice nonetheless.

Flash dump: [teckin_sp23_factory_fw.zip](https://github.com/libretiny-eu/libretiny/files/13325304/teckin_sp23_factory_fw.zip)

CPU: Realtek W302 / rtl8710bn

Pinout:

| 5 | 18 | 12 | 23 | 22 | 30/TX2 | GND |
| --- | --- | --- | --- | --- | --- | --- |
| NC | 0 | 14 | 15 | 19 | 29/RX2 | 3v3 |

Photos:

[image]

[image]

kuba2k2, 2023-11-11:
Hi
What are the board's dimensions? And what is the pin raster?
Knowing that, I can add the appropriate drawings and definitions.

Weissnix4711, 2023-11-11:
Hi. Dimensions are approximately this:

[image]

Thanks

kuba2k2, 2023-11-14:
Here you go:
https://docs.libretiny.eu/boards/t112-v1.1/
Mind that the flash layout might not match if your module was preloaded with non-Tuya firmware - I've set it to 788k, as that's what we'll be leaning towards in the future.

Weissnix4711, 2023-11-15:
:+1: Brilliant.