libretiny-eu/libretiny #207: respond to physical switch even if WiFi isn't connected

ntzb, 2023-11-13:
hi,
after flashing libretiny-esphome, when there's no WiFi connection (e.g. for some reason the switch fails to connect after a power blackout), the relay is not switchable.
how can I set it up, so it listens to the physical switch, even if no WiFi is connected?
I'd expect it to act as a dumb switch, if no WiFi is available.

Cossid, 2023-11-13:
This isn't really a LibreTiny issue, and without a config, there is no advice that can really be given.  All button/binary_sensor/switch/relay components are local by default unless you tie them into a service.  Wifi state should not matter for them, though if you lose wifi, the device will likely reboot every 15 minutes if HA doesn't connect to it by default.

ntzb, 2023-11-13:
I know it could be esphome related, but I could only find https://www.reddit.com/r/homeassistant/comments/12dmyrh/help_needed_for_esphome_to_work_without_home/
which points to esphome's Binary Sensor Automation, but since it's working when WiFi is connected, I don't think I can should override `on_press`..

here's my yaml, it's the extracted/converted yaml produced after using tuya-cloudcutter, and after I calibrated the current/power etc:
```
esphome:
  name: "local-smart-plug"

bk72xx:
  board: generic-bk7231n-qfn32-tuya

logger:

web_server:

captive_portal:

mdns:

api:
  password: ""

ota:
  password: ""

wifi:
  ssid: SSID
  password: PASSWORD
  ap:

button:
  - platform: restart
    name: Restart

debug:
  update_interval: 30s

text_sensor:
  - platform: debug
    reset_reason:
      name: Reset Reason
  - platform: libretiny
    version:
      name: LibreTiny Version

sensor:
  - platform: uptime
    name: Uptime
  - platform: hlw8012
    model: BL0937
    cf_pin:
      number: P7
      inverted: true
    cf1_pin:
      number: P6
      inverted: true
    sel_pin:
      number: P24
      inverted: true
    current:
      name: BL0937 Current
      filters:
        - multiply: 0.45
        - lambda: if (x < 0.01) {return 0;} else {return x;}
    voltage:
      name: BL0937 Voltage
    power:
      name: BL0937 Power
    energy:
      name: BL0937 Energy
    voltage_divider: 820
    current_resistor: 0.00103 ohm
    update_interval: 3s

binary_sensor:
  - platform: gpio
    id: binary_switch_1
    pin:
      number: P10
      inverted: true
      mode: INPUT_PULLUP
    on_press:
      then:
        - switch.toggle: switch_1

switch:
  - platform: gpio
    id: switch_1
    name: Relay 1
    pin: P26
    on_turn_on:
      - light.turn_on: light_status
    on_turn_off:
      - light.turn_off: light_status

light:
  - platform: status_led
    id: light_status
    pin:
      number: P8
      inverted: true
```

Cossid, 2023-11-13:
That should always work locally, regardless of wifi state.  Are you sure you weren't in a reboot (which happens every 15 minutes when not conntected with `api:` present)?  Also, every 10 reboots, if it still can't connect, it will fallback to safemode for a period of time.  Again, that is all stock ESPHome functionality.

You might want to consider changing [reboot_timeout](https://esphome.io/components/api.html?highlight=reboot_timeout) to 0.

Cossid, 2023-11-13:
For the case you're describing, you might need it in both places.

ntzb, 2023-11-14:
by normally I meant forcing a reboot to normal operation (vs safe mode) even within the timeframe of the safe mode.

thanks for your help! I'll close this for now, and continue with esphome support.

closed 2023-11-14 by ntzb