libretiny-eu/libretiny #211: RTL8710BN Wifi reconnection loop after esphome OTA upload

fredericmorin, 2023-11-26:
Chip: RTL8710BN
Board: BEELITE smart plug

Issue: Wifi reconnection loop after OTA upload 

Current 
Power cycle fix the issue
Reboot after too many wifi connection error fix the issue as well.

Log from uart2 tx:
```
[D][wifi:428]: Starting scan...
[D][wifi:443]: Found networks:
[I][wifi:486]: - 'oxio-6B00' (6C:F3:7F:E7:2D:E2) ▂▄▆█
[D][wifi:488]:     Channel: 11
[D][wifi:489]:     RSSI: -54 dB
[D][wifi:491]: - 'T.Fam' (90:AA:C3:C9:33:58) ▂▄▆█
[D][wifi:491]: - 'Reseau Videotron' (C0:94:35:F4:E6:E6) ▂▄▆█
[D][wifi:491]: - 'NETGEAR86' (B0:39:56:D8:0C:28) ▂▄▆█
[I][wifi:277]: WiFi Connecting to 'oxio-6B00'...
E [    445.470] WIFI: Connection failed; ret=-1
[W][wifi_lt:119]: esp_wifi_connect failed! 4
[E][wifi:323]: wifi_sta_connect_ failed!
[W][wifi_lt:288]: Event: Disconnected ssid='' bssid=00:00:00:00:00:00 reason='Unspecified'
[D][wifi:428]: Starting scan...
[D][wifi:443]: Found networks:
[I][wifi:486]: - 'oxio-6B00' (6C:F3:7F:E7:2D:E2) ▂▄▆█
[D][wifi:488]:     Channel: 11
[D][wifi:489]:     RSSI: -54 dB
[D][wifi:491]: - 'T.Fam' (90:AA:C3:C9:33:58) ▂▄▆█
[D][wifi:491]: - 'Reseau Videotron' (C0:94:35:F4:E6:E6) ▂▄▆█
[D][wifi:491]: - 'NETGEAR86' (B0:39:56:D8:0C:28) ▂▄▆█
[D][wifi:491]: - 'Drou_2' (D8:0D:17:B9:6A:7C) ▂▄▆█
[I][wifi:277]: WiFi Connecting to 'oxio-6B00'...
E [    451.953] WIFI: Connection failed; ret=-1
[W][wifi_lt:119]: esp_wifi_connect failed! 4
[E][wifi:323]: wifi_sta_connect_ failed!
[W][wifi_lt:288]: Event: Disconnected ssid='' bssid=00:00:00:00:00:00 reason='Unspecified'
wIP_DHCP: dhcp stop.
Deinitializing WIFI ...0m

WIFI deinitialized[W][wifi_lt:288]: Event: Disconnected ssid='' bssid=00:00:00:00:00:00 reason='Unspecified'
[D][wifi:428]: Starting scan...

Initializing WIFI ...
WIFI initialized
                [D][wifi:443]: Found networks:
[I][wifi:486]: - 'oxio-6B00' (6C:F3:7F:E7:2D:E2) ▂▄▆█
[D][wifi:488]:     Channel: 11
[D][wifi:489]:     RSSI: -54 dB
[D][wifi:491]: - 'RedMaJuVer4' (F8:79:0A:11:17:96) ▂▄▆█
[D][wifi:491]: - 'NETGEAR86' (B0:39:56:D8:0C:28) ▂▄▆█
[D][wifi:491]: - 'Drou_2' (D8:0D:17:B9:6A:7C) ▂▄▆█
[I][wifi:277]: WiFi Connecting to 'oxio-6B00'...
E [    458.721] WIFI: Connection failed; ret=-1
[W][wifi_lt:119]: esp_wifi_connect failed! 4
[E][wifi:323]: wifi_sta_connect_ failed!
[W][wifi_lt:288]: Event: Disconnected ssid='' bssid=00:00:00:00:00:00 reason='Unspecified'
[D][wifi:428]: Starting scan...
[D][wifi:443]: Found networks:
[I][wifi:486]: - 'oxio-6B00' (6C:F3:7F:E7:2D:E2) ▂▄▆█
[D][wifi:488]:     Channel: 11
[D][wifi:489]:     RSSI: -54 dB
[D][wifi:491]: - 'Helix 1175 2.4' (A0:FF:70:DC:62:9C) ▂▄▆█
[D][wifi:491]: - 'Reseau Videotron' (C0:94:35:F4:E6:E6) ▂▄▆█
[D][wifi:491]: - 'NETGEAR86' (B0:39:56:D8:0C:28) ▂▄▆█
[D][wifi:491]: - 'Drou_2' (D8:0D:17:B9:6A:7C) ▂▄▆█
[I][wifi:277]: WiFi Connecting to 'oxio-6B00'...
E [    465.204] WIFI: Connection failed; ret=-1
[W][wifi_lt:119]: esp_wifi_connect failed! 4
[E][wifi:323]: wifi_sta_connect_ failed!
[W][wifi_lt:288]: Event: Disconnected ssid='' bssid=00:00:00:00:00:00 reason='Unspecified'
[D][wifi:428]: Starting scan...
```

Kaldek, 2023-12-22:
I can confirm the same behaviour.  Tuya WR4 running ESPHome 2023.12.0.

rishabmehta7, 2024-04-23:
Seeing the same

geonux, 2025-03-28:
Hello, I also have the same problem, tested on two different wifi networks (board: wr2; framework: arduino; platform: libretiny @ 1.7.0). 
The scan correctly see available networks, but I have this connect error :
```
E [    391.151] WIFI: Connection failed; ret=-1
[W][wifi_lt:127]: esp_wifi_connect failed! 4
[E][wifi:369]: wifi_sta_connect_ failed!
[W][wifi_lt:296][wifievent]: Event: Disconnected ssid='' bssid=00:00:00:00:00:00 reason='Unspecified'
[V][wifi_lt:264][wifievent]: Event: WiFi Scan Done status=0 number=2 scan_id=0
```

What surprises me is that the AP network works perfectly.

geonux, 2025-03-28:
Probably also linked to https://github.com/libretiny-eu/libretiny/issues/247

For very obvious reason, I've just seen that the wifi connection has just been initialized... and this connection seems to be stable for several minutes. 
This initialization of the connection with an existing wifi network is therefore functional, but (very) random, whereas the initialization of the AP mode is done the first time.
A timing problem when initializing the RTL8195A  ?