libretiny-eu/libretiny #213: RTL8711AM - possible to support?

szupi-ipuzs, 2023-11-28:
I've found this chip in the wifi adapter of my air conditioner (HiSense) and, judging by it's name, it should be of the same group as the other,  already supported Realtek chips.
Do you think there is a chance it can be supported by LibreTiny as well?
I've found its [datasheet](https://datasheetspdf.com/pdf-file/1301574/apm/RTL8711AM/1) and there is an [sdk on github](https://github.com/pvvx/RtlDuino/tree/master/development/rtl87xx/system/sdk/project) that references this chip.

The board (wifi adapter) itself [looks like this](https://fcc.report/FCC-ID/2AGCCAEH-W4E1/4378065.pdf) and is called AEH-W4E1. It communicates with the air conditioner via a 4 pin cable, which I believe is an UART (the pins are marked as RX/TX/VCC/GND). The protocol used is unknown so far, (could be TuyaMCU?)... The module is connected to the air conditioner only via this cable, so it is also powered by it.
I know it would probably be best to get a flashdump from this thing, but I doubt i will be able to solder cable to the chip itself, its pins are extremely tiny. There are however some pads on both sides of the board, but I have no clue what they are for...

So all I can do for now is to take some detailed pictures and attach them here.

Btw. I've got the air conditioner to run locally via [this great HA integration](https://github.com/deiger/AirCon), however I was not able to cut it from the cloud completely, as it starts to exhibit strange behaviors (going zombie state), typical for many Tuya devices in such scenario. This is why I wanted to investigate if there's a possibility to flash it with esphome.

kuba2k2, 2023-11-28:
See here:
https://docs.libretiny.eu/docs/platform/realtek-amb/
RTL87xxA is Ameba1 series - it's different than AmebaZ and AmebaZ2, which are currently (more or less) supported in LT.

The SDK used for AmebaZ is the same as for Ameba1, so adding support would be possible. However, it's not a plan for the near future, as these chips are very rarely found these days.

Taking pictures would help for future reference.

szupi-ipuzs, 2023-11-30:
[image]
[image]
[image]

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Here are the pictures, I have done some (minor, I hope) damage while removing the shielding (?), as I did not expect it would be glued.

kuba2k2, 2023-11-30:
It wasn't glued, it was soldered - and solder is a pretty good glue 😆
Thanks anyway, I hope you didn't damage your device.

szupi-ipuzs, 2023-11-30:
Haha, that explains a lot :) Don't worry, this was my spare part.

danielg4, 2024-06-20:
I came to this page simply by Googling "RTL8711AM" after seeing it mentioned here:
https://fcc.report/FCC-ID/BEJ-LCW007/3962072.pdf
That is _the_ Wi-Fi module inside practically every LG appliance, e.g. refrigerators, clothes washers, etc.
I just thought you should know.

RneeJ, 2024-09-09:
bump.
I opend up a brandnew bought wifi/zigbee router and found a WRG1.
According to Tuya it consists of a RTL8711AM for Wifi

Technical details here : https://developer.tuya.com/en/docs/iot/wrg1-datasheet?id=K97rig6mscj8e

This board is inside : 
[image]

[image]

RneeJ, 2024-09-11:
It seems this wifi board rtl8711-am is also used in this development board https://www.iot-store.com.au/products/creator-pro-rtl8711am-wiscreator-473ma-realtek-ameba-wifi-module?srsltid=AfmBOorW3UNn2KWqMBD5JSWJi5IhUdkmbKfAC0DPW77Kr-ddgRB1g5nU

Github : https://github.com/RAKWireless/CREATOR-Arduino-SDK

mihsu81, 2024-09-11:
@RneeJ There's a project for the Tuya GW018-DM which has a RTL8721CSM, if you want a custom firmware to use in conjunction with Z2MQTT. https://github.com/jasperw1996/ambd_sdk_GW018-DM

RneeJ, 2024-09-12:
@mihsu81 thank you! That looks very promising, i will try that.

webysther, 2026-01-11:
Just connect related info from tasmota: https://github.com/arendst/Tasmota/discussions/9262