libretiny-eu/libretiny #214: RTL8710BX Static IP not working

MiguelAReis, 2023-11-29:
Hello, I have flashed a bunch of Smart Sockets, some with esp8285's, BK7231N's and RTL8710BX's. I have set static IPs through esphome yaml shown below. Every device but the RTL8710BX has taken the static ip I set. 

Example yaml code:
```
esphome:
  name: socket010
  friendly_name: Socket 010

rtl87xx:
  board: generic-rtl8710bx-4mb-980k

web_server:
  
# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "XXXXXXXXXXXXXXXXXXXXXX"
    
ota:
  password: "XXXXXXXXXXXXXXXXXXXX"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  fast_connect: true
  manual_ip:
    static_ip: 192.168.0.20
    gateway: 192.168.0.1
    subnet: 255.255.255.0

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Socket Hotspot"

captive_portal:

    
debug:
  update_interval: 30s

binary_sensor:
  - platform: gpio
    id: binary_switch_1
    pin:
      number: PA18
      inverted: true
      mode: INPUT_PULLUP
    on_press:
      then:
        - switch.toggle: switch_1

switch:
  - platform: gpio
    id: switch_1
    name: Relay
    pin: PA00
    restore_mode: RESTORE_DEFAULT_OFF

status_led:
  pin:
    number: PA23
    inverted: true

sensor:
  - platform: uptime
    name: Uptime
  - platform: hlw8012
    model: BL0937
    cf_pin:
      number: PA05
      inverted: false
    cf1_pin:
      number: PA12
      inverted: false
    sel_pin:
      number: PA15
      inverted: true
    current:
      name: Current
    voltage:
      name: Voltage
    power:
      name: Power
    energy:
      name: Energy
    voltage_divider: 1530
    current_resistor: 0.001 ohm
    update_interval: 5s
    change_mode_every: 3

```

Below is an excerpt of the UART output. Notice the different IP's reported in it by [wifi] (the actual IP)  and by [web_server],[ota] and [api] (the one i wanted to set):

```
[I][wifi:560]: WiFi Connected!
[C][wifi:382]:   Local MAC: XXXXXXXXXXXXXX
[C][wifi:383]:   SSID: 'XXXXXXXXXXXXXXXXX'
[C][wifi:384]:   IP Address: 192.168.0.249
[C][wifi:385]:   BSSID: XXXXXXXXXXXXXXX
[C][wifi:387]:   Hostname: 'socket010'
[C][wifi:389]:   Signal strength: -52 dB ▂▄▆█
[C][wifi:393]:   Channel: 1
[C][wifi:394]:   Subnet: 255.255.255.0
[C][wifi:395]:   Gateway: 192.168.0.1
[C][wifi:396]:   DNS1: 195.130.131.1
[C][wifi:397]:   DNS2: 195.130.131.1
[D][wifi:569]: Disabling AP...
[C][web_server:116]: Setting up web server...
[C][ota:097]: Over-The-Air Updates:
[C][ota:098]:   Address: 192.168.0.20:8892
[C][ota:101]:   Using Password.
[C][api:025]: Setting up Home Assistant API server...
[I][app:062]: setup() finished successfully!
[I][app:102]: ESPHome version 2023.10.6 compiled on Nov 29 2023, 23:31:58
[C][status_led:019]: Status LED:
[C][status_led:020]:   Pin: 23
[C][wifi:546]: WiFi:
[C][wifi:382]:   Local MAC: XXXXXXXXXXXXX
[C][wifi:383]:   SSID: 'XXXXXXXXXXXXXX'
[C][wifi:384]:   IP Address: 192.168.0.249
[C][wifi:385]:   BSSID: B4:F2:67:5A:E6:1E
[C][wifi:387]:   Hostname: 'socket010'
[C][wifi:389]:   Signal strength: -51 dB ▂▄▆█
[C][wifi:393]:   Channel: 1
[C][wifi:394]:   Subnet: 255.255.255.0
[C][wifi:395]:   Gateway: 192.168.0.1
[C][wifi:396]:   DNS1: 195.130.131.1
[C][wifi:397]:   DNS2: 195.130.131.1
[C][logger:416]: Logger:
[C][logger:417]:   Level: DEBUG
[C][logger:418]:   Log Baud Rate: 115200
[C][logger:420]:   Hardware UART: UART2
[C][switch.gpio:068]: GPIO Switch 'Relay'
[C][switch.gpio:090]:   Restore Mode: restore defaults to OFF
[C][switch.gpio:031]:   Pin: 0
[C][uptime.sensor:031]: Uptime Sensor 'Uptime'
[C][uptime.sensor:031]:   Device Class: 'duration'
[C][uptime.sensor:031]:   State Class: 'total_increasing'
[C][uptime.sensor:031]:   Unit of Measurement: 's'
[C][uptime.sensor:031]:   Accuracy Decimals: 0
[C][uptime.sensor:031]:   Icon: 'mdi:timer-outline'
[C][gpio.binary_sensor:015]: GPIO Binary Sensor 'binary_switch_1'
[C][gpio.binary_sensor:016]:   Pin: 18
[C][hlw8012:037]: HLW8012:
[C][hlw8012:038]:   SEL Pin: 15
[C][hlw8012:039]:   CF Pin: 5
[C][hlw8012:040]:   CF1 Pin: 12
[C][hlw8012:041]:   Change measurement mode every 3
[C][hlw8012:042]:   Current resistor: 1.0 mΩ
[C][hlw8012:043]:   Voltage Divider: 1530.0
[C][hlw8012:044]:   Update Interval: 5.0s
[C][hlw8012:045]:   Voltage 'Voltage'
[C][hlw8012:045]:     Device Class: 'voltage'
[C][hlw8012:045]:     State Class: 'measurement'
[C][hlw8012:045]:     Unit of Measurement: 'V'
[C][hlw8012:045]:     Accuracy Decimals: 1
[C][hlw8012:046]:   Current 'Current'
[C][hlw8012:046]:     Device Class: 'current'
[C][hlw8012:046]:     State Class: 'measurement'
[C][hlw8012:046]:     Unit of Measurement: 'A'
[C][hlw8012:046]:     Accuracy Decimals: 2
[C][hlw8012:047]:   Power 'Power'
[C][hlw8012:047]:     Device Class: 'power'
[C][hlw8012:047]:     State Class: 'measurement'
[C][hlw8012:047]:     Unit of Measurement: 'W'
[C][hlw8012:047]:     Accuracy Decimals: 1
[C][hlw8012:048]:   Energy 'Energy'
[C][hlw8012:048]:     Device Class: 'energy'
[C][hlw8012:048]:     State Class: 'total_increasing'
[C][hlw8012:048]:     Unit of Measurement: 'Wh'
[C][hlw8012:048]:     Accuracy Decimals: 1
[C][captive_portal:088]: Captive Portal:
[C][web_server:161]: Web Server:
[C][web_server:162]:   Address: 192.168.0.20:80
[C][mdns:115]: mDNS:
[C][mdns:116]:   Hostname: socket010
[C][ota:097]: Over-The-Air Updates:
[C][ota:098]:   Address: 192.168.0.20:8892
[C][ota:101]:   Using Password.
[C][api:139]: API Server:
[C][api:140]:   Address: 192.168.0.20:6053
[C][api:142]:   Using noise encryption: YES
[C][lt.component:013]: LibreTiny:
[C][lt.component:014]:   Version: v1.4.1 on generic-rtl8710bx-4mb-980k, compiled at Nov 29 2023 23:31:10, GCC 10.3.1 (-Os)
[C][lt.component:015]:   Loglevel: 3
[C][debug:064]: Debug component:
```

fasax1284, 2024-01-13:
Same issue with RTL8710 static ip

Raddie12, 2024-03-10:
Can you update it via ota?

I´m always getting an error at the end:

`RAM:   [==        ]  20.4% (used 53608 bytes from 262144 bytes)
Flash: [=====     ]  51.5% (used 516836 bytes from 1003520 bytes)
========================= [SUCCESS] Took 3.96 seconds =========================
INFO Successfully compiled program.
INFO Connecting to 10.8.199.81
INFO Uploading /data/build/smart-plug-rtl8710bx-01/.pioenvs/smart-plug-rtl8710bx-01/firmware.uf2 (1034752 bytes)
Uploading: [============================================================] 100% Done...

INFO Upload took 9.25 seconds, waiting for result...
ERROR Error Update end: Error: Finishing update failed. See the MQTT/USB logs for more information.`

jon-daemon, 2025-02-09:
I have the same issue with `board: wr3`. Is there any way to set a static IP? My router doesn't support it.