libretiny-eu/libretiny #221: E: TimeoutError: Timeout while linking - Readingc

ntblade, 2023-12-11:
Hi,
I'm unable to read RTL8710B. Here's the cli output:
norrie@norrie-G50-70:~$ ltchiptool -v flash read ambz kettle
D: Found Local data snapshot (v1.4.1) - /home/norrie/.local/lib/python3.10/site-packages/ltchiptool
I: Available COM ports:
I: |-- ttyUSB0 - CP2102 USB to UART Bridge Controller - CP2102 USB to UART Bridge Controller - Silicon Labs (10C4/EA60)
I: |   |-- Selecting this port. To override, use -d/--device
I: Connecting to 'Realtek AmebaZ' on /dev/ttyUSB0 @ 1500000
D: Linking: stage 0
D: Linking: stage 1
D: Linking: stage 2
D: Linking: stage 3
I: Connect UART2 of the Realtek chip to the USB-TTL adapter:
I: 
I:     --------+        +---------------------
I:          PC |        | RTL8710B            
I:     --------+        +---------------------
I:          RX | ------ | TX2 (Log_TX / PA30) 
I:          TX | ------ | RX2 (Log_RX / PA29) 
I:             |        |                     
I:         GND | ------ | GND                 
I:     --------+        +---------------------
I:  
I: Using a good, stable 3.3V power supply is crucial. Most flashing issues
I: are caused by either voltage drops during intensive flash operations,
I: or bad/loose wires.
I:  
I: The UART adapter's 3.3V power regulator is usually not enough. Instead,
I: a regulated bench power supply, or a linear 1117-type regulator is recommended.
I:  
I: In order to flash the chip, you need to enable download mode.
I: This is done by pulling CEN to GND briefly, while still keeping the TX2 pin
I: connected to GND.
I:  
I: Do this, in order:
I:  - connect CEN to GND
I:  - connect TX2 to GND
I:  - release CEN from GND
I:  - release TX2 from GND
E: TimeoutError: Timeout while linking
E: |-- File "/home/norrie/.local/lib/python3.10/site-packages/ltchiptool/soc/ambz/util/ambztool.py", line 188, in link

Can anyone help please?

kuba2k2, 2023-12-12:
Well, did you do what the guide says? Is the download mode enabled?

ntblade, 2023-12-13:
Yest I did several times. If I power-cycle the chip here's what I can see in minicom:
`ROM:[V0.1]
FLASHRATE:4
BOOT TYPE:0 XTAL:40000000
IMG1 DATA[1168:10002000]
IMG1 ENTRY[8000541:100021ef]
IMG1 ENTER
CHIPID[000000ff]
read_mode idx:0, flash_speed idx:0
calibration_result:[1:9:9][5:d] 
calibration_result:[2:13:7][1:d] 
ROM:[V0.1]
FLASHRATE:4
BOOT TYPE:0 XTAL:40000000
IMG1 DATA[1168:10002000]
IMG1 ENTRY[8000541:100021ef]
IMG1 ENTER
CHIPID[000000ff]
read_mode idx:0, flash_speed idx:0
calibration_result:[1:9:9][5:d] 
calibration_result:[2:13:7][1:d] 
calibration_result:[3:1:1][1:1] 
calibration_ok:[2:13:7] 
FLASH CALIB[NEW OK]
OTA2 ADDR[80d0000]
OTAx SELE[fffffffc]
OTA1 USE
IMG2 DATA[0x80985e0:6684:0x10005000]
IMG2 SIGN[RTKWin(10005008)]
IMG2 ENTRY[0x10005000:0x800b17d]
===== Enter Image 2 ====
System_Init1
OSC8M: 8383135 
boot reason: 0 
System_Init2

11111111111111111111111111
interface 0 is initialized
interfac[01-01 1e 1 is init8:12:15 TUYAialized

Initializi_log_serv.c:ng WIFI ...280] log sta
LDO Mts ufread faode, BD_Infil.
[01-0o: 0 
1 18:12:15 TUYA Debug][o
nline_log_LDO Mode, BDserv.c:540]_Info: 0 
 log serv init success
[01-01 18:12:15 TUYA Debug][tuya_device.c:22] < TUYA IOT SDK V:2.0.0 BS:30.07_PT:2.2_LAN:3.3_CAD:1.0.2_CD:1.0.0 >
< tuya_iot_lib BUILD AT:2019_10_31_21_06_38 BY tuya_iot_team AT 8710_2M >
IOT DEFS < WIFI_GW:1 DEBUG:1 KV_FILE:0 SHUTDOWN_MODE:0 LITTLE_END:1 TLS_MODE:2 ENABLE_LOCAL_LINKAGE:0 ENABLE_CLOUD_OPERATION:0 ENABLE_SUBDEVICE:0 ENABLE_ENGINEER_TO_NOR>
                                                                                                 
[01-01 18:12:15 TUYA Debug][tuya_device.c:23] rtlbn_tls_common_9600:1.0.5                        
[01-01 18:12:15 TUYA Notice][simple_flash.c:430] key_addr: 0x1eb000   block_sz 4096              
[01-01 18:12:15 TUYA Notice][simple_flash.c:498] get key:                                        
0xe0 0x8c 0x65 0xb2 0x82 0x83 0x9e 0x47 0x94 0x1f 0x84 0xef 0x63 0xd9 0x84 0x94 

WIFI initialized

init_thread(55), Available heap 0x11328[01-01 18:12:15 TUYA Notice][tuya_uart.c:125] 1   9600
[01-01 18:12:15 TUYA Notice][tuya_main.c:369] mf_init succ
[01-01 18:12:15 TUYA Notice][uart_common.c:3598] uart baud rate:9600----firmware key:keym5sjxrqcryptn
[01-01 18:12:15 TUYA Notice][tuya_uart.c:125] 0   9600
[01-01 18:12:15 TUYA Notice][uart_common.c:3695] uart_task_init ok
[01-01 18:12:15 TUYA Notice][uart_common.c:602] send jump_pack
`
Does this help?

kuba2k2, 2023-12-13:
Then you didn't do what the guide says. This log output is NOT download mode, it's a normal boot.

You need to connect TX2 to GND and then power on the chip. After you disconnect TX2 from GND it shouldn't print normal logs. Then run the flashing program.

kuba2k2, 2023-12-13:
After that you need to disconnect CEN from ground, and then disconnect TX2 from ground. If you don't do this, it will never work. The instruction message includes these steps too.

ntblade, 2023-12-13:
Sorry I hit comment before I was finished. Yes, I'm disconnecting them in the correct order :)

ntblade, 2023-12-14:
Thanks for trying to help me. I dropped the board and tore the wire I had soldered onto TX2 off pulling the pad with it. :(