libretiny-eu/libretiny #227: PA15 not working on T102_V1.1 Board RTL8710BN

ModuloFS, 2023-12-23:
Hello, 

i am using ESPhome to flash my Tekin SP22 smart plug. It has a T102_V1.1 board with an RTL8710BN chip on it. Sadly GPIO PA15, which is driving the Relay is not working. If i change the output in my switch-section of the config yaml of esphome to PA14 the output works perfectly fine. So i think it is not a config problem. 
I also tried the configuration "gpio_recover: true" but this also not helped.

Does anyone have the same issue?

BTW. If i want to add the power sensor i also get the same change interrupt error as described in this issue:

https://github.com/libretiny-eu/libretiny/issues/155

Here is my esphome config:

`esphome:
  name: esphome-tekin-sp22

rtl87xx:
  board: t102-v1.1

    
logger:

web_server:
    version: 1

mdns:

ota:
  password: ""

wifi:
  ssid: "MY-NET"
  password: "MY-PW"

output:
  - platform: libretiny_pwm
    id: output_led_1
    pin: PA5

light:
  - platform: monochromatic
    id: light_switch_1
    output: output_led_1

binary_sensor:
  - platform: gpio
    id: binary_switch_1
    pin:
      number: PA23
      inverted: true
      mode: INPUT_PULLUP
    on_press:
      then:
        - switch.toggle: switch_1

switch:
  - platform: gpio
    id: switch_1
    name: Relay 1
    pin:
        number: PA15
        mode: OUTPUT
    on_turn_on:
      - light.turn_on: light_switch_1
    on_turn_off:
      - light.turn_off: light_switch_1

status_led:
  pin: PA18
`

fasax1284, 2024-01-14:
some problem with PA15 pin not work

alexl2709, 2024-01-21:
I had got the same problem. After the changing name board to wr2 - works