libretiny-eu/libretiny #229: Adafruit NeoPixels linker errors on BK7231N

hannescam, 2023-12-26:
I tried to upload and compile the example for Adafruit Neopixel on my board but I got this error:

```hannes@hannes-GUSCHTAV ~/D/new_fw> platformio run -t upload
Processing generic-bk7231n-qfn32-tuya (platform: libretiny; board: generic-bk7231n-qfn32-tuya; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/libretiny/generic-bk7231n-qfn32-tuya.html
PLATFORM: LibreTiny (1.4.1+sha.eed39c9) > Generic - BK7231N (Tuya QFN32)
HARDWARE: BK7231N 120MHz, 256KB RAM, 1.03MB Flash
DEBUG: Current (openocd) External (custom, openocd)
PACKAGES: 
 - framework-arduino-api @ 2022.8.24+sha.237b10a 
 - framework-beken-bdk @ 0.0.0+v2021.06.07.sha.6491b8c 
 - library-flashdb @ 1.2.0+sha.d5c892f 
 - library-freertos @ 9.0.0+sha.95cc959 
 - library-freertos-port @ 2023.5.23+sha.a917d93 
 - library-lwip @ 2.1.3-bdk+sha.4ee4d34 
 - library-printf @ 6.1.0+sha.28a79bd 
 - library-uf2ota @ 5.0.0+sha.f955412 
 - tool-openocd @ 2.1100.211028 (11.0) 
 - toolchain-gccarmnoneeabi @ 1.100301.220327 (10.3.1)
PLATFORM VERSIONS:
 - libretiny @ 1.4.1+sha.eed39c9.dirty
 - ltchiptool @ 4.10.1
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 1 compatible libraries
Scanning dependencies...
Dependency Graph
|-- Adafruit NeoPixel @ 1.12.0
Building in release mode
Compiling .pio/build/generic-bk7231n-qfn32-tuya/src/main.cpp.o
Archiving .pio/build/generic-bk7231n-qfn32-tuya/libarduino_api.a
Indexing .pio/build/generic-bk7231n-qfn32-tuya/libarduino_api.a
Compiling .pio/build/generic-bk7231n-qfn32-tuya/core_bk72xx_arduino_libraries/WiFi/WiFiAP.cpp.o
Compiling .pio/build/generic-bk7231n-qfn32-tuya/core_bk72xx_arduino_libraries/WiFi/WiFiEvents.cpp.o
Compiling .pio/build/generic-bk7231n-qfn32-tuya/core_bk72xx_arduino_libraries/WiFi/WiFiGeneric.cpp.o
Compiling .pio/build/generic-bk7231n-qfn32-tuya/core_bk72xx_arduino_libraries/WiFi/WiFiSTA.cpp.o
Compiling .pio/build/generic-bk7231n-qfn32-tuya/core_bk72xx_arduino_libraries/WiFi/WiFiScan.cpp.o
Archiving .pio/build/generic-bk7231n-qfn32-tuya/libcore_bk72xx_arduino_fixups.a
Indexing .pio/build/generic-bk7231n-qfn32-tuya/libcore_bk72xx_arduino_fixups.a
Compiling .pio/build/generic-bk7231n-qfn32-tuya/core_bk72xx_arduino/main.cpp.o
Compiling .pio/build/generic-bk7231n-qfn32-tuya/core_bk72xx_arduino/wiring.c.o
Compiling .pio/build/generic-bk7231n-qfn32-tuya/core_bk72xx_arduino/wiring_analog.c.o
Compiling .pio/build/generic-bk7231n-qfn32-tuya/core_bk72xx_arduino/wiring_digital.c.o
Compiling .pio/build/generic-bk7231n-qfn32-tuya/core_bk72xx_arduino/wiring_irq.c.o
Compiling .pio/build/generic-bk7231n-qfn32-tuya/core_common_arduino_libraries/api/Serial/Serial.cpp.o
Archiving .pio/build/generic-bk7231n-qfn32-tuya/libcore_bk72xx_arduino.a
Indexing .pio/build/generic-bk7231n-qfn32-tuya/libcore_bk72xx_arduino.a
Compiling .pio/build/generic-bk7231n-qfn32-tuya/core_common_arduino_libraries/api/SoftwareSerial/SoftwareSerial.cpp.o
Compiling .pio/build/generic-bk7231n-qfn32-tuya/core_common_arduino_libraries/api/WiFi/WiFi.cpp.o
Compiling .pio/build/generic-bk7231n-qfn32-tuya/core_common_arduino_libraries/api/WiFi/WiFiAP.cpp.o
Compiling .pio/build/generic-bk7231n-qfn32-tuya/core_common_arduino_libraries/api/WiFi/WiFiEvents.cpp.o
Compiling .pio/build/generic-bk7231n-qfn32-tuya/core_common_arduino_libraries/api/WiFi/WiFiGeneric.cpp.o
Compiling .pio/build/generic-bk7231n-qfn32-tuya/core_common_arduino_libraries/api/WiFi/WiFiSTA.cpp.o
Archiving .pio/build/generic-bk7231n-qfn32-tuya/libcore_bk72xx_arduino_libraries.a
Indexing .pio/build/generic-bk7231n-qfn32-tuya/libcore_bk72xx_arduino_libraries.a
Compiling .pio/build/generic-bk7231n-qfn32-tuya/core_common_arduino_libraries/api/WiFi/WiFiScan.cpp.o
Compiling .pio/build/generic-bk7231n-qfn32-tuya/core_common_arduino_libraries/common/FS/FS.cpp.o
Compiling .pio/build/generic-bk7231n-qfn32-tuya/core_common_arduino_libraries/common/IPv6Address/IPv6Address.cpp.o
Compiling .pio/build/generic-bk7231n-qfn32-tuya/core_common_arduino_libraries/common/MD5/MD5MbedTLSImpl.cpp.o
Compiling .pio/build/generic-bk7231n-qfn32-tuya/core_common_arduino_libraries/common/Update/Update.cpp.o
Compiling .pio/build/generic-bk7231n-qfn32-tuya/core_common_arduino_libraries/common/Update/UpdateUtil.cpp.o
Compiling .pio/build/generic-bk7231n-qfn32-tuya/core_common_arduino_libraries/common/WiFiClient/LwIPClient.cpp.o
Compiling .pio/build/generic-bk7231n-qfn32-tuya/core_common_arduino_libraries/common/WiFiClient/LwIPRxBuffer.cpp.o
Compiling .pio/build/generic-bk7231n-qfn32-tuya/core_common_arduino_libraries/common/WiFiClient/MbedTLSClient.cpp.o
Compiling .pio/build/generic-bk7231n-qfn32-tuya/core_common_arduino_libraries/common/WiFiServer/LwIPServer.cpp.o
Compiling .pio/build/generic-bk7231n-qfn32-tuya/core_common_arduino_libraries/common/WiFiUdp/LwIPUdp.cpp.o
Compiling .pio/build/generic-bk7231n-qfn32-tuya/core_common_arduino_libraries/common/mDNS/LwIPmDNS.cpp.o
Compiling .pio/build/generic-bk7231n-qfn32-tuya/core_common_arduino_libraries/common/mDNS/mDNS.cpp.o
[... 509 lines cut ...]

  strip.begin();           // INITIALIZE NeoPixel strip object (REQUIRED)
  strip.show();            // Turn OFF all pixels ASAP
  strip.setBrightness(50); // Set BRIGHTNESS to about 1/5 (max = 255)
}

// loop() function -- runs repeatedly as long as board is on ---------------

void loop() {
  // Fill along the length of the strip in various colors...
  colorWipe(strip.Color(255,   0,   0), 50); // Red
  colorWipe(strip.Color(  0, 255,   0), 50); // Green
  colorWipe(strip.Color(  0,   0, 255), 50); // Blue

  // Do a theater marquee effect in various colors...
  theaterChase(strip.Color(127, 127, 127), 50); // White, half brightness
  theaterChase(strip.Color(127,   0,   0), 50); // Red, half brightness
  theaterChase(strip.Color(  0,   0, 127), 50); // Blue, half brightness

  rainbow(10);             // Flowing rainbow cycle along the whole strip
  theaterChaseRainbow(50); // Rainbow-enhanced theaterChase variant
}
```

Help would be greatly appreaciated

kuba2k2, 2023-12-26:
It seems there are some missing functions or macros used by the NeoPixel library. These might be specific to esp32 or esp8266 and are not supported by LibreTiny. To fix this you would have to find what are these functions used for, and where.

hannescam, 2023-12-26:
I think these errors are related to interrupt disabling/enabling, but if I comment them out it uploads but the NeoPixel stays off.
Here are the lines in the code:
https://github.com/adafruit/Adafruit_NeoPixel/blob/master/Adafruit_NeoPixel.cpp#L288
https://github.com/adafruit/Adafruit_NeoPixel/blob/master/Adafruit_NeoPixel.cpp#L3036
I think if it is possible to implement those it would work maybe