libretiny-eu/libretiny #231: Remote receiver not working on a CBU with linptech_g6l_protocol

spupetic, 2023-12-29:
I have installed ESPHome on a CBU based 433 MHz wireless doorbell. It has a doorbell chime chip (that works well via the remote_transmitter component) and a CMT2217B OOK RF receiver, which does not work somewhy. I can confirm that the RF receiver itself works with ESPHome on an ESP32. 

Here is the code for the protocol decoder:

[https://github.com/pauln/esphome-linp-doorbell-g04/tree/main/components/linptech_g6l](https://github.com/pauln/esphome-linp-doorbell-g04/tree/main/components/linptech_g6l)

I don't get any errors or warnings, except that the PIN should be inverted, but when I invert it, that doesn't change anything. By setting the log level to very verbose, nothing useful comes up, it seems like the protocol decoder never sends log messages.

spupetic, 2024-01-03:
I have tried connecting the same RF receiver to an ESP32, and with that, it works fine.

ESP32:
[image]

Taking a look at the remote.raw codes, and comparing the ESP32 and the CBU logs, it seems like the latter interprets the signal differently. Inverting the PIN mode does not help.

CBU:
[image]

spupetic, 2024-01-07:
I have tried with an ESP8266 connected to the receiver, and it works perfectly fine with that too.

BrainP4in, 2024-01-14:
Don't know if this is related, but I get pretty similar logs on my CBU `remote_receiver` with an IR Sensor. It worked on OpenBK, so there should be nothing wrong with the Setup. Is it possible that the problem is not in the `linptech_g6l` protocol?

spupetic, 2024-01-14:
> Don't know if this is related, but I get pretty similar logs on my CBU `remote_receiver` with an IR Sensor. It worked on OpenBK, so there should be nothing wrong with the Setup. Is it possible that the problem is not in the `linptech_g6l` protocol?

Does your receiver work with ESPHome on the CBU? I think that the problem is not with the `linptech_g6l` protocol decoder, as it works fine with an ESP32 and an ESP8266 too. 

If I connect the same receiver to an ESP-based device, it works fine, but with the CBU, it does not, so it should be CBU or libretiny related. The libretiny `remote_receiver` is based on the ESP8266's remote_receiver codebase, and it works with that on an ESP. So, that leaves two possibilities, something is behaving different: the CBU or the implementation.

spupetic, 2024-01-15:
> The receiver only worked in OpenBK. In ESPHome I get raw logs like in your Screenshots. So I think your assumption is right.

What pin do your receiver use? And were there warnings that stated it should be inverted?

BrainP4in, 2024-01-15:
Im using P8 for der Receiver.
And at first the log showed the warning, which went away after adding invert to the pin configuration.

spupetic, 2024-01-15:
> Im using P8 for der Receiver. And at first the log showed the warning, which went away after adding invert to the pin configuration.

In my case, with an ESP-device, the receiver should not be inverted, it does not work when inverted. The problem can also be `P8 `related in that case.

BrainP4in, 2024-01-15:
My problem is that I can't easily change that since it is a "Sonoff" IR Blaster/ Remote

spupetic, 2024-01-15:
> My problem is that I can't easily change that since it is a "Sonoff" IR Blaster/ Remote

Yes, it should not be changed.

spupetic, 2024-01-17:
> Have you only tried receiving messages? Has sending worked on your Device? Maybe related as well.
> 
> Not sure if it's a user error, but it logs that the message was sent but the IR Device ex. TV, AV Receiver don't react.

The `remote_transmitter` works on my device, but it is not IR or RF, it just controls the interconnected doorbell chime chip. So I have a doorbell, which dings when I ask it, but cannot react to the button presses :D

Which pin do you use for the transmitter?

BrainP4in, 2024-01-17:
> The `remote_transmitter` works on my device, but it is not IR or RF, it just controls the interconnected doorbell chime chip. So I have a doorbell, which dings when I ask it, but cannot react to the [...]

Sounds cool ^^

I'm using P7 to transmit the data but as I said, this coud be a error on my side.

spupetic, 2024-01-22:
Did you have success?

BrainP4in, 2024-01-23:
No not really, I also tried input pullup on the receiver and got more realistic logs but almost everything is raw.

spupetic, 2024-01-23:
I have also tried input pullup, pulldown and inverting with all of these combinations, but none of them worked. Can we say that we have ruled out user error?

BrainP4in, 2024-01-23:
I think so, yes.

spupetic, 2024-01-30:
@kuba2k2 Can you give us some advice, what can we do to investigate this further?

olivluca, 2024-08-26:
Probably related to https://github.com/libretiny-eu/libretiny/issues/69#issuecomment-2284071231