libretiny-eu/libretiny #234: Identify deep sleep wake types on lt_get_reboot_reason()

daniel-dona, 2024-01-02:
Currently waking from deep sleep is defined as `REBOOT_REASON_SLEEP` but in BK7231 there are at least two possible "sub-types": GPIO and RTC.

Can we have separate reboot reasons for these cases?

Maybe `REBOOT_REASON_SLEEP_GPIO` and `REBOOT_REASON_SLEEP_RTC`. 

```c
lt_reboot_reason_t lt_get_reboot_reason() {
	switch (bk_misc_get_start_type()) {
		case RESET_SOURCE_POWERON:
			return REBOOT_REASON_POWER;
		case RESET_SOURCE_REBOOT:
			return REBOOT_REASON_SOFTWARE;
		case RESET_SOURCE_WATCHDOG:
			return REBOOT_REASON_WATCHDOG;
		case RESET_SOURCE_CRASH_XAT0:
		case RESET_SOURCE_CRASH_UNDEFINED:
		case RESET_SOURCE_CRASH_PREFETCH_ABORT:
		case RESET_SOURCE_CRASH_DATA_ABORT:
		case RESET_SOURCE_CRASH_UNUSED:
		case RESET_SOURCE_CRASH_PER_XAT0:
			return REBOOT_REASON_CRASH;
		case RESET_SOURCE_DEEPPS_USB:
                    return REBOOT_REASON_SLEEP_USB;
		case RESET_SOURCE_DEEPPS_GPIO:
		    return REBOOT_REASON_SLEEP_GPIO;
		case RESET_SOURCE_DEEPPS_RTC:
		    return REBOOT_REASON_SLEEP_RTC;
			default:
			return REBOOT_REASON_UNKNOWN;
	}
}
```

```c
const char *lt_get_reboot_reason_name(lt_reboot_reason_t reason) {
	if (!reason)
		reason = lt_get_reboot_reason();
	switch (reason) {
		case REBOOT_REASON_POWER:
			return "Power-On";
		case REBOOT_REASON_BROWNOUT:
			return "Brownout";
		case REBOOT_REASON_HARDWARE:
			return "HW Reboot";
		case REBOOT_REASON_SOFTWARE:
			return "SW Reboot";
		case REBOOT_REASON_WATCHDOG:
			return "WDT Reset";
		case REBOOT_REASON_CRASH:
			return "Crash";
		case REBOOT_REASON_SLEEP_GPIO:
			return "Sleep Wakeup (GPIO)";
		case REBOOT_REASON_SLEEP_RTC:
			return "Sleep Wakeup (RTC)";
		case REBOOT_REASON_SLEEP_USB:
			return "Sleep Wakeup (USB)";
		case REBOOT_REASON_DEBUGGER:
			return "Debugger";
		default:
			return "Unknown";
	}
}
```

```c
typedef enum {
	REBOOT_REASON_UNKNOWN  = 1,
	REBOOT_REASON_POWER	   = 2,
	REBOOT_REASON_BROWNOUT = 3,
	REBOOT_REASON_HARDWARE = 4,
	REBOOT_REASON_SOFTWARE = 5,
	REBOOT_REASON_WATCHDOG = 6,
	REBOOT_REASON_CRASH	   = 7,
	REBOOT_REASON_SLEEP_GPIO = 8,
	REBOOT_REASON_SLEEP_RTC  = 9,
	REBOOT_REASON_SLEEP_USB  = 10,
	REBOOT_REASON_DEBUGGER = 11,
	REBOOT_REASON_MAX	   = 12,
} lt_reboot_reason_t;
```

There are a lot of cases were makes sense to check the reason, for example I'm working with some battery powered sensors that use both RTC (for updating battery state every ~24h) and GPIO (sensor triggered).

I can prepare a PR if this sounds good...

kuba2k2, 2024-01-03:
Approved 👍

cap9qd, 2024-02-14:
@daniel-dona I don't want to jump in the middle of your work but do have these changes made and tested in my fork; I needed them for an ESPHome LibreTiny PR for deep_sleep. Judging by your comments maybe  you are also working on something similar with deep sleep or another purpose?

0x3333, 2024-02-21:
Well, I'm in the same situation as @cap9qd, I need deep sleep and I started implementing it, eventually, we could collaborate on this.