libretiny-eu/libretiny #235: Platform RTL8720CF

esnet146, 2024-01-03:
[image]
read:
512 b - ltchiptool_ambz2_2024-01-03_15-34-57_efuse.bin
2 097 152 b - ltchiptool_ambz2_2024-01-03_15-29-55.bin
393 216 b - ltchiptool_ambz2_2024-01-03_15-28-26_rom.bin   

generate esphome config:
rtl87xx:
  board: generic-rtl8720cf-2mb-992k

and write.
restart:

== Rtl8710c IoT Platform ==
Chip VID: 5, Ver: 3
ROM Version: v3.0

== Boot Loader ==
Dec  5 2019:14:02:18

fwx SELE[fffffffe]
fw SELE Bitidx 1, fw1 valid 0, sn 0, fw2 valid 1, sn 1
fw2 USE, return sn 1
[MISC Err]Hash Result Incorrect!
Boot Load Err!

Restart and a0 +3.3v:
== Rtl8710c IoT Platform ==
Chip VID: 5, Ver: 3
ROM Version: v3.0
Test Mode: boot_cfg1=0x20
Test Mode GPIOA[14, 3, 2] = 0x0
[SPIF Err]Invalid ID
[SPIF Err]Invalid ID
[BOOT Err]Flash init error (io_mod=0, pin_sel=0)
StartUp@0x0: Invalid RAM Img Signature!

Any ideas on how to launch it?

kuba2k2, 2024-01-03:
https://github.com/libretiny-eu/libretiny/issues/44#issuecomment-1858636582

esnet146, 2024-01-03:
Yes!

rtl87xx:
  board: wbr3
  family: rtl8720c
  framework:
    version: 0.0.0
    source: https://github.com/libretiny-eu/libretiny.git#feature/realtek-update

 and flash: A_0+RX(GPIOA_13) to 3.3v

kuba2k2, 2024-01-03:
So, does it work now?

To boot the firmware, A0 can't be pulled to 3.3v.

esnet146, 2024-01-03:
it work! 
Only flash  firmware A_0+RX(GPIOA_13) to 3.3v
boot - A0 to gnd or free

closed 2024-01-03 by esnet146

helgek, 2024-03-21:
@esnet146 Thank you for sharing this information. I tried to also read the firmware from a WBR3 (of this humidifier https://www.airbi.cz/en/humidifier-evo-wifi) using BK7231 GUI Flash Tool. Unfortunately I'm not able to activate the flash mode. Would you maybe be able to provide a simplified instruction how you got into flash mode (my technical background is not good enough to be able to fully understand your instruction)? 

I also tried to follow the information from https://developer.tuya.com/en/docs/iot/wbr3-module-datasheet?id=K9dujs2k5nriy and https://developer.tuya.com/en/docs/iot/burn-and-authorize-wbr-series-modules?id=Ka78imt8pf85p . 

This is how I have connected to the WBR3 (I don't have access to A_0, is this maybe the issue? I'm wondering then how I could figure out if there's maybe a pin on the bottom side of the bigger pcb which represents A_0?):

[image]

[image]
[image]
[image]

From the LED activity on the UART I'm pretty confident that the basic setup is correct (I've successfully) flashed a couple of ESP32 before). I also cross-checked swapping TX and RX just to make sure that I didn't make a mistake there. 

Thanks.

kuba2k2, 2024-03-21:
As the name implies, "BK7231 GUI Flash Tool" is for BK7231 chips, not Realtek chips.

The recommended tool is ltchiptool, it works with Realtek and Beken chips.

WBR3 is not yet supported by LibreTiny, I'm not sure what do you want to accomplish. Anyway, to enter download mode, you need to pull A0 to 3.3V while resetting the chip (similar to ESP32 but not to GND). However, on WBR3 this pin is not accessible - it's on the bottom side of the PCB. Accessing it would require desoldering.

helgek, 2024-03-29:
@kuba2k2 Thank you very much for the explanation and clarification. These were the remaining missing information pieces to fully understand how this works. I could have understood from the documentation too but probably I just didn't want to think of the possibility that Tuya designs their chips in a way to completely avoid access possibilities to flash them. 

This week I received another Tuya supported humidifier (EUROM OASIS WIFI)and it revealed a WR3 chip which is in the list of supported chipsets: https://docs.libretiny.eu/boards/wr3/#flash-memory-map and https://docs.libretiny.eu/docs/status/supported/#board-list . Following https://docs.libretiny.eu/docs/platform/realtek-ambz/#flashing I haven't yet been successful to start download mode but this might be because I was not (yet) fulfilling all recommendations listed there. 

Thank you for all the efforts you are putting into this project.

maxdd, 2025-02-01:
@helgek was it sufficient to connect A0 to 3.3V or was A15 also needed?

helgek, 2025-02-03:
@maxdd Sorry I don't remember and I also never successfully put the chip into flash mode. But according to https://github.com/libretiny-eu/libretiny/issues/235#issuecomment-2012328704 it should be A0 only

maxdd, 2025-02-03:
Seems like UART_LOG is needed so you need to desolder, I will try unless I found home assistant + tuya integration sufficient.
I hope tuya-cloudcutter will find another bug to exploit