libretiny-eu/libretiny #237: esphome HLW8012 sensor not working with BK7231N

honzakutil, 2024-01-07:
Hello, I was able to upload esphome to BK7231N (board CB2S). Wifi and GPIOs are working, but I am not able to get working the HLW8012 power sensor (https://esphome.io/components/sensor/hlw8012.html). The esphome HLW8012 seems to need `pulse_counter_sensor.h`, which needs `esphome/core/hal.h`. 

So the question is: Is the `pulse_counter_sensor` somehow tightly connected with ESP32/ESP2866 hardware or is it supposed to work also with BK7231N?

ESPhome config file follows:
```yaml
substitutions:
  device_name: zasuvka02
  friendly_name: Zasuvka 02

esphome:
  name: ${device_name}
#  platform: esp32

bk72xx:
  board: cb2s
  framework:
    version: latest

# Enable logging
logger: 
  level: VERY_VERBOSE
  logs:
    sensor: INFO
    binary_sensor: INFO
    text_sensor: INFO
    api.service: DEBUG
    scheduler: DEBUG

# Enable Home Assistant API
api:
  encryption:
    key: "....."

ota:
  password: !secret ota_psw

wifi:
#  fast_connect: true
  networks:
  - ssid: !secret wifi-IOT
    password: !secret wifi-IOT_psw
    hidden: true
  - ssid: !secret wifi-str
    password: !secret wifi-str_psw
  - ssid: !secret wifi-str2
    password: !secret wifi-str2_psw
  reboot_timeout: 2min

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "${device_name}_Hotspot"
    password: !secret fallback_psw

#captive_portal:

text_sensor:
  - platform: template
    name: "${friendly_name} Uptime Human Readable"
    id: uptime_human
    icon: mdi:clock-start

sensor:
  - platform: uptime
    name: "${friendly_name} Uptime Sensor"
    id: uptime_sensor
    update_interval: 30s
    on_raw_value:
      then:
        - text_sensor.template.publish:
            id: uptime_human
            state: !lambda |-
              int seconds = round(id(uptime_sensor).raw_state);
              int days = seconds / (24 * 3600);
              seconds = seconds % (24 * 3600);
              int hours = seconds / 3600;
              seconds = seconds % 3600;
              int minutes = seconds /  60;
              seconds = seconds % 60;
              return (
                (days ? to_string(days) + "d " : "") +
                (hours ? to_string(hours) + "h " : "") +
                (minutes ? to_string(minutes) + "m " : "") +
                (to_string(seconds) + "s")
              ).c_str();

  - platform: wifi_signal
    name: "${friendly_name} WiFi Sensor"
    update_interval: 30s

  - platform: hlw8012
    model: BL0937
    current_resistor: 0.001
    voltage_divider: 2351
    change_mode_every: 4
    initial_mode: VOLTAGE
    sel_pin:
      number: P6
      inverted: true
    cf_pin: P8
    cf1_pin: P7
    current:
      name: "${friendly_name} Current"
      accuracy_decimals: 6
    voltage:
      name: "${friendly_name} Voltage"
      accuracy_decimals: 6
    power:
      name: "${friendly_name} Power"
      accuracy_decimals: 6
    energy:
      name: "${friendly_name} Energy"
      accuracy_decimals: 6
    update_interval: 5s

# Immax NEO LITE Smart vnitřní zásuvka v2, WiFi
# 07762L

# internal connection
# P8 - BL0937 pin 6 - CF
# P7 - BL0937 pin 7 - CF1
# P6 - BL0937 pin 8 - SEL
# P24 - relay active high
# RX1 - button to gnd
# P26 - blue LED (connected to vcc, active low)

# https://esphome.io/components/libretiny.html#advanced-options
# https://esphome.io/components/sensor/hlw8012.html

switch:
  - platform: gpio
    name: "${friendly_name} Relay"
    id: relay
    pin: P24
    restore_mode: ALWAYS_ON

  - platform: gpio
    name: "${friendly_name} LED"
    id: led
    pin: 
      number: P26
      inverted: true
    restore_mode: ALWAYS_ON

binary_sensor:
  - platform: gpio
    name: "${friendly_name} Button"
    id: button
    pin: RX1
    filters:
      - invert:
    on_click:
      min_length: 50ms
      max_length: 350ms
      then:
        - switch.toggle: relay
        - switch.toggle: led

```

Cossid, 2024-01-07:
You likely need to invert all 3 pins, that has pretty consistently been the case so far on the Beken platform.

honzakutil, 2024-01-07:
thanks for quick reply, inverting all three pins solved the issue. Power meter is now working correctly.

closed 2024-01-07 by honzakutil