libretiny-eu/libretiny #238: arm-none-eabi-g++: not found

hunhejj, 2024-01-08:
I am getting the error `arm-none-eabi-g++: not found` when building a libretiny firmware.

ESPHome: v2023.12.15 (Docker)
Host: Raspberry Pi 4 (4 GB)

Sample device:
```
esphome:
  name: libretiny_test

bk72xx:
  board: generic-bk7231n-qfn32-tuya

api:
  encryption:
    key: !secret api_encryption_key

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
```

Logs:
[logs_libretiny-test_compile.txt](https://github.com/libretiny-eu/libretiny/files/13863367/logs_libretiny-test_compile.txt)

SaschaMuehl, 2024-01-30:
same issue here

hunhejj, 2024-02-24:
Any idea @kuba2k2 ?

hunhejj, 2024-02-24:
I just checked: I am using a 64 bit container (arm64) on a 64bit system (64 bit Raspberry OS) @kuba2k2

hunhejj, 2024-02-26:
ping @kuba2k2

kuba2k2, 2024-02-26:
Actually, I don't have a solution. I don't know what causes the issue exactly, I don't have the right environment to reproduce it. I can only refer you to the issues listed above, hoping that you find a solution there.

juskalalie, 2024-02-29:
I opened #259, I can give you more information if you need.
I am running HAOS Blue on an Odroid N2+ with 4GB RAM.