libretiny-eu/libretiny #242: T102-V1.1 with RTL8710BX boot loop

hansnull, 2024-01-18:
I have bought a WIFI plugin which has a T102_V1.1 board, but not with the RTL8710BN chip... it has the RTL8710BX.
So I tried to use the generic-rtl8710bx-4mb-980k and flashed the device. Now it just boot looping, here the debug output:

I [      0.000] LibreTiny v1.4.1+sha.4cddc01 on generic-rtl8710bx-4mb-980k, compiled at Jan 18 2024 22:25:45, GCC 10.3.1 (-Os)
[I][logger:395]: Log initialized
[C][ota:473]: There have been 4 suspected unsuccessful boot attempts.
[D][lt.preferences:104]: Saving 1 preferences to flash...
[D][lt.preferences:132]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[I][app:029]: Running through setup()...
[C][switch.gpio:011]: Setting up GPIO Switch 'Relay'...
[D][switch:016]: 'Relay' Turning OFF.
[D][switch:055]: 'Relay': Sending state OFF
[D][switch:016]: 'Relay' Turning OFF.
[C][wifi:038]: Setting up WiFi...
[C][wifi:051]: Starting WiFi...
[C][wifi:052]:   Local MAC: C4:82:E1:A7:80:1D
interface 0 is initialized
interface 1 is initialized

Initializing WIFI ...
WIFI initialized
[D][wifi:459]: Starting scan...
RTL8195A[HAL]: Hard Fault Error!!!!
RTL8195A[HAL]: R0 = 0x1b9
RTL8195A[HAL]: R1 = 0x1
RTL8195A[HAL]: R2 = 0x0
RTL8195A[HAL]: R3 = 0x7
RTL8195A[HAL]: R12 = 0x1b9
RTL8195A[HAL]: LR = 0x80151cd
RTL8195A[HAL]: PC = 0x1eb8dcf0
RTL8195A[HAL]: PSR = 0x80000000
RTL8195A[HAL]: BFAR = 0xe000ed38
RTL8195A[HAL]: CFSR = 0x20000
RTL8195A[HAL]: HFSR = 0x40000000
RTL8195A[HAL]: DFSR = 0x0
RTL8195A[HAL]: AFSR = 0x0
RTL8195A[HAL]: PriMask 0x0
RTL8195A[HAL]: BasePri 0x0
RTL8195A[HAL]: SVC priority: 0x00
RTL8195A[HAL]: PendSVC priority: 0xf0
RTL8195A[HAL]: Systick priority: 0xf0
<RTL8195A>ROM:[V0.1]
FLASHRATE:4
BOOT TYPE:0 XTAL:40000000
IMG1 DATA[1168:10002000]
IMG1 ENTRY[8000541:100021ef]
IMG1 ENTER
CHIPID[000000f6]
read_mode idx:3, flash_speed idx:0
calibration_result:[1:9:9][5:d] 
calibration_result:[2:13:7][1:d] 
calibration_result:[3:1:1][1:1] 
calibration_ok:[2:13:7] 
FLASH CALIB[NEW OK]
OTA2 ADDR[80d0000]
OTAx SELE[fffffffc]
OTA1 USE
IMG2 DATA[0x8074290:3824:0x10005000]
IMG2 SIGN[RTKWin(10005008)]
IMG2 ENTRY[0x10005000:0x8033e5d]
System_Init1
System_Init2
I [      0.000] LibreTiny v1.4.1+sha.4cddc01 on generic-rtl8710bx-4mb-980k, compiled at Jan 18 2024 22:25:45, GCC 10.3.1 (-Os)
[I][logger:395]: Log initialized
[C][ota:473]: There have been 5 suspected unsuccessful boot attempts.
[D][lt.preferences:104]: Saving 1 preferences to flash...
[D][lt.preferences:132]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[I][app:029]: Running through setup()...
[C][switch.gpio:011]: Setting up GPIO Switch 'Relay'...
[D][switch:016]: 'Relay' Turning OFF.
[... 299 lines cut ...]

Initializing WIFI ...
WIFI initialized
[D][wifi:459]: Starting scan...
RTL8195A[HAL]: Hard Fault Error!!!!
RTL8195A[HAL]: R0 = 0x8015b01
RTL8195A[HAL]: R1 = 0x0
RTL8195A[HAL]: R2 = 0x10003938
RTL8195A[HAL]: R3 = 0x13d
RTL8195A[HAL]: R12 = 0x3b
RTL8195A[HAL]: LR = 0x802ff13
RTL8195A[HAL]: PC = 0x80151f8
RTL8195A[HAL]: PSR = 0x61000000
RTL8195A[HAL]: BFAR = 0xe000ed38
RTL8195A[HAL]: CFSR = 0x1000000
RTL8195A[HAL]: HFSR = 0x40000000
RTL8195A[HAL]: DFSR = 0x0
RTL8195A[HAL]: AFSR = 0x0
RTL8195A[HAL]: PriMask 0x0
RTL8195A[HAL]: BasePri 0x0
RTL8195A[HAL]: SVC priority: 0x00
RTL8195A[HAL]: PendSVC priority: 0xf0
RTL8195A[HAL]: Systick priority: 0xf0

any help is appreciated : )

Cossid, 2024-01-18:
How are you powering the device?  It usually won't fully boot when powered from a UART device when wifi is enabled, but may when using it's designed power supply.

hansnull, 2024-01-18:
Good hint, i will try an external power supply!

hansnull, 2024-01-18:
That did the trick : ) Thank you very much

closed 2024-01-18 by hansnull