libretiny-eu/libretiny #245: VIKEFON Tuya Power Plug LSPA10 - A10TY Board (?)

ronlut, 2024-01-26:
Hi, thanks a lot for the amazing work.
I must say in advance, I am software guy but here to learn and help if possible.
I have the device in the title, attaching some photos of it.
I am really not sure it's even BK/RTL, but actually I am not sure what it is at all.
It's made by VIKEFON using Tuya ([example from aliexpress with various plug types](https://www.aliexpress.com/item/1005005709815402.html)).

Would like to know/check whether it's supported by LibreTiny, can be supported or supported by any other solution.
I am currently changing all my tuya devices to ESPHome and can't find anything about this one on the web. 
The board says "A10TY", couldn't find anything about it as well.
Hopefully this thread might help others once it has some interesting info :)

[image]
[image]

Thanks a lot!

kuba2k2, 2024-01-26:
Look at the blue PCB at the top of the first photo - there's probably a model number written. If I had to guess, it might be a CB2S or WB2S. After you find the model number, refer to LibreTiny docs for more details.

ronlut, 2024-01-26:
@kuba2k2 WB2S indeed, thanks!
Looks like it's supported, adding to my todo list 👍

closed 2024-01-26 by ronlut