libretiny-eu/libretiny #247: RTL8710BX with 2 MB flash on a T102-V1.1 cannot connect to Wi-Fi network

spupetic, 2024-01-28:
I have generic Tuya smart sockets with a T102-V1.1 board, but it doesn't have the RTL8710BN, but the BX, without the 4 MB flash, according to ltchiptool, it has a 2 MB flash. I have installed ESPHome kickstart (RTL8710BN) on them, but I couldn't get Wi-Fi client mode working. I have tried different APs, changing channels and WPA mode, password, without success. Then, I have compiled ESPHome with the same and different (X-based) board, but it did not help. The fallback AP works though.

kuba2k2, 2024-01-28:
Can you post a photo of the T102 board? That way so I can see the exact chip markings of the RTL and the flash.

spupetic, 2024-01-29:
[image]

This is one of the boards, they are identical.

RF197, 2024-02-05:
I have the same plugs (blew one up though), and it does connect just fine.

Here are my config settings:

substitutions:
  device_name: 'Plug 2'

esphome:
  name: plug-2
  comment: '20A Smart Plug RTL 1'

rtl87xx:
  board: wr2

logger:
  level: DEBUG

wifi:
  networks:
    - ssid: !secret wifi_ssid
      password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Smart Plug 2 AP"
    password: "xxxxxxxxxxxxx"

api:

ota:
  password: "xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx"
  
captive_portal:

web_server:
  port: 80

I had issues getting the BL0937 to work (something with CHANGE intterupts not allowed) though, and PA15 does not switch.

spupetic, 2024-02-05:
How did you blew it up? By does not switch you mean the relay switching does not work?

Raddie12, 2024-02-07:
how did you get it in boot mode?
ESPHOME tells me to connect CEN ro GRD, TX2 to GRD, remove CEN from GRD and then remove TX2 from GRD.
I didn´t manage to do so ...

mabey I unse the wromg (CEN PIN?

spupetic, 2024-02-07:
> how did you get it in boot mode? ESPHOME tells me to connect CEN ro GRD, TX2 to GRD, remove CEN from GRD and then remove TX2 from GRD. I didn´t manage to do so ...
> 
> mabey I unse the wromg (CEN PIN?

I was just shorting TX2 to GND while plugging the USB to Serial adapter into the computer. Probably not the best way to do it, but I worked well for me. So I have not used CEN.

Raddie12, 2024-02-07:
Do you use GPIO_A30/DEBUG_LOG_TX & GPIO_A 29/DEBUG_LOG_RX. And do you use ESPHome or Tasmota? I tried both also both TX/RX but it´s nor working :(

spupetic, 2024-02-07:
> Do you use GPIO_A30/DEBUG_LOG_TX & GPIO_A 29/DEBUG_LOG_RX. And do you use ESPHome or Tasmota? I tried both also both TX/RX but it´s nor working :(
> 
> Do you release the TX2 from GRD after usb-plugging?

It is a bit confusing with the RTL's RX TX situation, you need to use UART2 (RX2 TX2) refered here: [https://docs.libretiny.eu/boards/t102-v1.1/#usage](https://docs.libretiny.eu/boards/t102-v1.1/#usage)

After plugging in the USB, I released GND from TX2.

I am using ESPHome, Tasmota and OpenBeken is not (yet) compatible with this board.

RF197, 2024-02-11:
> How did you blew it up? By does not switch you mean the relay switching does not work?

Errh yes, trying to connect debug UART while the device was in the mains power. Not my brightest moment....

You indeed have to use pins PA29/30 and ground TX while powering up. This only needs to happen the first time by the way.

I'm also using ESPhome.

spupetic, 2024-02-14:
> > How did you blew it up? By does not switch you mean the relay switching does not work?
> 
> Errh yes, trying to connect debug UART while the device was in the mains power. Not my brightest moment.... It also killed the USB ports on my PC, and took my keyboard to the grave with it (plus killed the power in the house). So don't try that at home...
> 
> You indeed have to use pins PA29/30 and ground TX while powering up. This only needs to happen the first time by the way.
> 
> I'm also using ESPhome.
> 
> Update: the relay works if you use the WR2 board (there's a separate issue on that, known bug) (since it's on PA15).
> 
> Enabling BL0937 causes the whole thing to bootloop until it goes into emergency mode. This is due to pin change interrupts not being supported in the chips; though they are needed to use the pulse counter plugin. There may be a workaround using two interrupt sources in one handler (it does have support for rising and falling edge interrupts), but I haven't been able to figure out how to convince esphome to use that....

Thanks for the tips. Sadly, the Wi-Fi issue is still present for me, I have tried OTA update from the fallback hotspot, it works, but the device does not connect to any Wi-Fi (even tried with a different access point).

RF197, 2024-02-15:
> Thanks for the tips. Sadly, the Wi-Fi issue is still present for me, I have tried OTA update from the fallback hotspot, it works, but the device does not connect to any Wi-Fi (even tried with a [...]

As a sanity check - the wifi access points are at 2.4GHz and have DHCP enabled, right?. If so this is very odd....

vladT0, 2024-05-14:
I have the same issue and the same board. According log, scan of WIFI networks is working correctly, but can not connect to WIFI:

```
[D][wifi:462]: Starting scan...
[D][wifi:477]: Found networks:
[I][wifi:520]: - 'ov593' (EE:41:18:EC:31:56) ▂▄▆█
[D][wifi:522]:     Channel: 2
[D][wifi:523]:     RSSI: -59 dB
[D][wifi:525]: - 'BeeOne' (F0:B0:14:E3:B7:ED) ▂▄▆█
[D][wifi:525]: - 'Vodafone Hotspot' (DE:53:1C:9A:54:5D) ▂▄▆█
[D][wifi:525]: - 'ov593c' (D4:DA:21:75:E9:E8) ▂▄▆█
[D][wifi:525]: - 'Unity-KL' (30:D3:2D:C2:B3:CB) ▂▄▆█
[D][wifi:525]: - 'W-2AQA205' (BE:FF:4D:F4:FE:25) ▂▄▆█
[D][wifi:525]: - 'Aloha' (34:31:C4:F3:04:E6) ▂▄▆█
[I][wifi:303]: WiFi Connecting to 'ov593'...
E [     33.201] WIFI: Connection failed; ret=-1
[W][wifi_lt:117]: esp_wifi_connect failed! 4
[E][wifi:349]: wifi_sta_connect_ failed!
[W][wifi_lt:286]: Event: Disconnected ssid='' bssid=00:00:00:00:00:00 reason='Unspecified'

```

vladT0, 2024-05-18:
I had the exact same problem, tried all the changes and could not connect to WIFI. Finally, I examined the board under a microscope and found a solder ball on one of the chip pins. I removed it and everything worked. Second, if you unsolder the module from the main board and use 3.3V from the UART PCB, note that its current is not enough to boot this module normally, it will reboot without stopping (at least in my case) and the WIFI will not work either. Try flashing a very simple PinScan example from the library, but use a module with a good LDO or solder it back to the main board.

didjeru, 2024-12-01:
Good afternoon! I also flashed all my Tuya outlets to 20 amps. Now, out of 6 outlets, only 2 are working properly. The others either have no Wi-Fi signal at all or the signal is only 16-20%. With the original firmware, the signal was 100%. I suspect that after the reflash, the chip switches to an external antenna, which is missing...

smshr, 2024-12-27:
I am having the same issue but with T103_V1.0 module.

[image]

geonux, 2025-03-28:
Probably linked to https://github.com/libretiny-eu/libretiny/issues/211