libretiny-eu/libretiny #248: mDNS not showing all TXT records

chrwei, 2024-01-28:
I'm using a bk7231n cb2s based device with esphome and had already opened an issue there https://github.com/esphome/issues/issues/5435 but it seems the issue is in libretiny's mDNS

to summarize the issue linked, when esphome adds a TXT record that is longer than 44 characters, that record, and any TXT records added after that, do not show up in the mDNS packet.  They do show up in the esphome logs, so it's somewhere in how libretiny's mDNS handles the `MDNS.addServiceTxt(service_type, proto, record.key.c_str(), record.value.c_str());` call.

I've looked at the code and I don't see anything obvious that would cause this, but my C++ skills are basic.

The same yaml works on esp8266, so it's not a non-printable character, I even retyped the line to be sure.  I did attempt to compare the esp8266mDNS code, but it's too different to make a useful comparison.

szupi-ipuzs, 2024-01-29:
I don't know much about mdns and I don't know if this is related, but comparing different esphome implementations of MDNSComponent, there's one thing different with libretiny: lack of the loop() method, which other implementations seem to use to call update() periodically.
And the loop method is missing because it wasn't enabled for cases when USE_LIBRETINY is defined:

```c++
class MDNSComponent : public Component {
 public:
  void setup() override;
  void dump_config() override;

#if (defined(USE_ESP8266) || defined(USE_RP2040)) && defined(USE_ARDUINO)
  void loop() override;
#endif
```

@kuba2k2 Is this by design or was simply forgotten? Could this be related to @chrwei  observerations ?

kuba2k2, 2024-01-29:
It's not related - the loop() method was added by ESPHome after LibreTiny had mDNS working. It's most likely related to RP2040 support which needed that method. I think it's used to call some mDNS callbacks that need to be called periodically, if any.

As to why the original issue happens - I'm not sure. Maybe there is some hardcoded length limit that just breaks things when exceeded. It will need further investigation.

chrwei, 2024-01-29:
that's what i thought too, but unless there's a (char *)malloc limit I don't see one.

chrwei, 2024-02-08:
anyone able to reproduce this?

Cossid, 2024-02-11:
Quickly looking into this, 44 is the length of the value  you're adding into a TXT record, but it looks like the issue is when the full record length is 64 or more characters, which is a much more logical number from a coding perspective.

In your own example
> I've narrowed it to an issue with package_import_url. If I make the url 44 characters or less it works, and 45 or more it fails.
"github://chrwei/esphome_templates/test.yaml@" works and "github://chrwei/esphome_templates/test.yaml@m" fails.

the first would have a value of
`package_import_url=github://chrwei/esphome_templates/test.yaml@` which is an overall length of 63
the second failed value of
`package_import_url=github://chrwei/esphome_templates/test.yaml@m` would be an overall length of 64

It does appear I have a few devices that are not broadcasting mdns as well, but I am unsure of why at this point, I don't believe I have any fields that exceed a total length of 63/64.  The issue for mine might be the PTR record exceeding 31/32.

chrwei, 2024-02-11:
I hadn't though of that, and that would overflow an int8 length, I just don't see anywhere that would be the issue

Cossid, 2024-02-11:
A little further investigating, neither 44/45 nor 63/64 appear to be the issue.

I had no field that long, but I shorted one of my longest strings until I started receiving a packet.  Then I shortened a different TXT record, and I was able to expand the one I previously had to shorten by that same amount.

It seems the maximum data length of all TXT data is capped at 213 bytes, which again, is a very unusual/non-programmatic number.

szupi-ipuzs, 2024-02-24:
Since I don't know much about mdns, I decided to try this from a different angle.
I've extracted the mdns code from esphome/libretiny and built it as a separate app on ubuntu with gtest framework (mocking the actual lwip implementation). 
Then I run this with valgrind and it did find some problems:

```
~/projects/mdns_test/.build$ valgrind --leak-check=full --show-leak-kinds=all  test/lt_mdns_test
==2714346== Memcheck, a memory error detector
==2714346== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
==2714346== Using Valgrind-3.18.1 and LibVEX; rerun with -h for copyright info
==2714346== Command: test/lt_mdns_test
==2714346== 
Running main() from /home/szupi/projects/mdns_test/.build/_deps/googletest-src/googletest/src/gtest_main.cc
[==========] Running 1 test from 1 test suite.
[----------] Global test environment set-up.
[----------] 1 test from MDNSComponent
[ RUN      ] MDNSComponent.aaa
Starting (TEST_NAME)
GMOCK WARNING:
Uninteresting mock function call - returning directly.
    Function call: mdns_resp_register_name_result_cb(0x1fa027)
NOTE: You can safely ignore the above warning unless this call should not happen.  Do not suppress it by blindly adding an EXPECT_CALL() if you don't mean to enforce the call.  See https://github.com/google/googletest/blob/main/docs/gmock_cook_book.md#knowing-when-to-expect-useoncall for details.

GMOCK WARNING:
Uninteresting mock function call - returning directly.
    Function call: mdns_resp_init()
NOTE: You can safely ignore the above warning unless this call should not happen.  Do not suppress it by blindly adding an EXPECT_CALL() if you don't mean to enforce the call.  See https://github.com/google/googletest/blob/main/docs/gmock_cook_book.md#knowing-when-to-expect-useoncall for details.
==2714346== Invalid write of size 1
==2714346==    at 0x484EE8E: strcpy (in /usr/libexec/valgrind/vgpreload_memcheck-amd64-linux.so)
==2714346==    by 0x1F9CED: ensureUnderscore(char const*) (mDNS.cpp:9)
==2714346==    by 0x1F9D65: mDNS::addService(char*, char*, unsigned short) (mDNS.cpp:20)
==2714346==    by 0x2024CB: mDNS::addService(char const*, char const*, unsigned short) (mDNS.h:94)
==2714346==    by 0x2023A3: esphome::mdns::MDNSComponent::setup() (mdns_libretiny.cpp:31)
==2714346==    by 0x1DB22D: MDNSComponent_aaa_Test::TestBody() (lt_mdns_test.cpp:82)
==2714346==    by 0x258AE8: void testing::internal::HandleSehExceptionsInMethodIfSupported<testing::Test, void>(testing::Test*, void (testing::Test::*)(), char const*) (gtest.cc:2612)
==2714346==    by 0x250CAC: void testing::internal::HandleExceptionsInMethodIfSupported<testing::Test, void>(testing::Test*, void (testing::Test::*)(), char const*) (gtest.cc:2648)
==2714346==    by 0x22ABFD: testing::Test::Run() (gtest.cc:2687)
==2714346==    by 0x22B718: testing::TestInfo::Run() (gtest.cc:2836)
==2714346==    by 0x22C11E: testing::TestSuite::Run() (gtest.cc:3015)
==2714346==    by 0x23C645: testing::internal::UnitTestImpl::RunAllTests() (gtest.cc:5920)
==2714346==  Address 0x4e25de5 is 0 bytes after a block of size 5 alloc'd
==2714346==    at 0x4848899: malloc (in /usr/libexec/valgrind/vgpreload_memcheck-amd64-linux.so)
==2714346==    by 0x1F9CB9: ensureUnderscore(char const*) (mDNS.cpp:7)
==2714346==    by 0x1F9D65: mDNS::addService(char*, char*, unsigned short) (mDNS.cpp:20)
==2714346==    by 0x2024CB: mDNS::addService(char const*, char const*, unsigned short) (mDNS.h:94)
==2714346==    by 0x2023A3: esphome::mdns::MDNSComponent::setup() (mdns_libretiny.cpp:31)
==2714346==    by 0x1DB22D: MDNSComponent_aaa_Test::TestBody() (lt_mdns_test.cpp:82)
==2714346==    by 0x258AE8: void testing::internal::HandleSehExceptionsInMethodIfSupported<testing::Test, void>(testing::Test*, void (testing::Test::*)(), char const*) (gtest.cc:2612)
==2714346==    by 0x250CAC: void testing::internal::HandleExceptionsInMethodIfSupported<testing::Test, void>(testing::Test*, void (testing::Test::*)(), char const*) (gtest.cc:2648)
==2714346==    by 0x22ABFD: testing::Test::Run() (gtest.cc:2687)
==2714346==    by 0x22B718: testing::TestInfo::Run() (gtest.cc:2836)
==2714346==    by 0x22C11E: testing::TestSuite::Run() (gtest.cc:3015)
==2714346==    by 0x23C645: testing::internal::UnitTestImpl::RunAllTests() (gtest.cc:5920)
==2714346== 
==2714346== Invalid write of size 1
==2714346==    at 0x484EE8E: strcpy (in /usr/libexec/valgrind/vgpreload_memcheck-amd64-linux.so)
==2714346==    by 0x1F9CED: ensureUnderscore(char const*) (mDNS.cpp:9)
==2714346==    by 0x1F9E22: mDNS::addServiceTxt(char*, char*, char*, char*) (mDNS.cpp:29)
==2714346==    by 0x20250C: mDNS::addServiceTxt(char const*, char const*, char const*, char const*) (mDNS.h:102)
==2714346==    by 0x202420: esphome::mdns::MDNSComponent::setup() (mdns_libretiny.cpp:33)
==2714346==    by 0x1DB22D: MDNSComponent_aaa_Test::TestBody() (lt_mdns_test.cpp:82)
==2714346==    by 0x258AE8: void testing::internal::HandleSehExceptionsInMethodIfSupported<testing::Test, void>(testing::Test*, void (testing::Test::*)(), char const*) (gtest.cc:2612)
==2714346==    by 0x250CAC: void testing::internal::HandleExceptionsInMethodIfSupported<testing::Test, void>(testing::Test*, void (testing::Test::*)(), char const*) (gtest.cc:2648)
==2714346==    by 0x22ABFD: testing::Test::Run() (gtest.cc:2687)
==2714346==    by 0x22B718: testing::TestInfo::Run() (gtest.cc:2836)
==2714346==    by 0x22C11E: testing::TestSuite::Run() (gtest.cc:3015)
[... cut ...]
2) There's a memory leak when destroying the MDNS object. Shouldn't the pointers in the vector be freed?

@chrwei if we setup all your strings in MDNSComponent::compile_records_() exactly how they appear in your device, we might actually be able to debug this issue. As long as it is reproducible on a PC :) But at least we'll know.

Attached is (almost) all that is needed to build and run this test. Additionally you need to install liblwip (sudo apt install liblwip-dev).
[mdns_test.tar.gz](https://github.com/libretiny-eu/libretiny/files/14391136/mdns_test.tar.gz)

szupi-ipuzs, 2024-03-05:
Ok, after a lot of debugging and chasing ghosts in the code I came to the same conclusion as you guys: this is an issue with [hard limit in lwip implementation](https://github.com/lwip-tcpip/lwip/blob/d0efd9ef7ba08e54b46b1060e2b4629a4907391b/src/apps/mdns/mdns_domain.c#L82) - max 64 characters for single txt label (length field not included).
The mdns_resp_add_service_txtitem() function does return error and this breaks the loop, so the remaining txt entries are not processed, even if they might fit in the buffer. I think this can be improved. But the 64 char limit is something that needs to be changed in LwIP. Or libretiny can switch to a different implementation of mdns :)

> Since I don't know much about mdns

Hey, I've learned something today :)

kuba2k2, 2024-03-05:
We could include a fixup for `"lwip/apps/mdns.h"` that will first include the original file, then `#undef` the `MDNS_LABEL_MAXLEN` and `#define` it again with a higher value. Fixups are placed before any other directory in the search path. At least that's the case for SDK include paths.

szupi-ipuzs, 2024-03-07:
That might work providing lwip always use this constant consequently - otherwise it's risky. I would first ask the authors opinion before patching it like that.
Or we can actually ask them to change it - by creating a pull request. At least then they would (hopfully) tell us if this is a correct approach.

szupi-ipuzs, 2024-03-08:
I've just read in [the RFC](https://datatracker.ietf.org/doc/html/rfc6762#appendix-C) that this limit is imposed by the specification... so shouldn't this also NOT work with other implementations?

Cossid, 2024-03-08:
I haven't test your changes yet, but that's just it, none of the strings I was testing with personally should have even been hitting that 63 character length.  So it's possible I was running into the underflow error, and OP is running into the max length issue.

I honestly don't even know the difference between what is a label and what is a name, I have little knowledge of the core of mDNS.

Cossid, 2024-03-08:
The more I test, the more I think we were just running into random underruns that would just kill the entire mDNS send.

The value I was testing with before (which in the TXT key=value format was only a length of 59 and was previously failing) is now working, and the only thing that has changed in the mDNS packet (that I am aware of) is the ESPHome version string in a different TXT record.

Cossid, 2024-03-08:
Unfortunately, it looks like the underflow was not our issue.  I tried deploying it to a problematic device, and it is no different.

For me, the only reproducible case is when the data size of all TXT records combined exceeds a data length of 208, the entire mDNS packet is dropped/not sent.

Working example
[image]

And it will begin to fail if I add any characters into either `project_name` or `project_version`
If I reduce characters in either field, I can add them to the other field, and as long as the total data length of the TXT type stays <= 208, it seems to work.

And it's not like we're running into packet size issues, as the entire mDNS packet is only 475 bytes when the TXT data length is 208.

None of my individual TXT records are exceeding 63 characters, so that's not the limiting factor either.

chrwei, 2024-03-09:
Which brings up, what's esp8266mDNS doing different that the same ESPHome config works on esp8266 devices?
based on my packet caps it seems esp8266mDNS splits it into 3 packets:
host.local, "QU" question A
host._http._tcp.local, "QU" question PTR
host._esphomelib._tcp.local, "QU" question PTR

lwip seems to try to shove it all into one packet, but it is separate TXT records, so I'm not sure why that matters.

wireshark capture zip file (2KB) https://www.dropbox.com/scl/fi/gx8n34zqkvf3spbje2pxs/mdnscap.zip?rlkey=gju38y2ejj2oqjszva2ck7d4t&dl=0

.232 is the esp8266 and .181 is the bk72xx. capture is power on one, then power on the other.
Add this filter in wireshark to limit to just the standard query: _ws.col.info contains "Standard query 0x0000"

Cossid, 2024-03-09:
I'm not fully convinced it's related to packet size either, as the initial packet is large, but subsequent packets sent later are smaller (on devices that don't have this issue).  The issue on LibreTiny is if it doesn't send the first packet, it also never sends any subsequent packets, so it appears the entire mDNS module fails and stops.

I've been testing remotely so far, without serial, so I haven't had a chance to enable LT debug logs yet.  I'll try it on a dev board with serial and see what I can find there.

Cossid, 2024-03-09:
Serial logging only gives one more tiny clue, but not smoking gun like an error like I was hoping.

The following is the full LT mdns output when the packet _is not sent_:
```
D [      7.960] MDNS: Starting (devboard-cbu-123456789012345)
D [      7.966] MDNS: Starting mDNS on netif 1
D [      7.970] MDNS: Added IGMP to netif 1
D [      7.978] MDNS: mDNS started on netif 1, announcing it to network
D [      7.982] MDNS: Starting mDNS on netif 0
D [      7.988] MDNS: Added IGMP to netif 0
D [      7.994] MDNS: mDNS started on netif 0, announcing it to network
D [      8.000] MDNS: Add service: netif 1 / devboard-cbu-123456789012345 / _esphomelib / 1 / 6053
D [      8.010] MDNS: Add service: netif 0 / devboard-cbu-123456789012345 / _esphomelib / 1 / 6053
D [      8.020] MDNS: Add service: netif 1 / devboard-cbu-123456789012345 / _http / 1 / 80
D [      8.028] MDNS: Add service: netif 0 / devboard-cbu-123456789012345 / _http / 1 / 80
```
and the following is the output when the packet _is_ sent:
```
D [     11.952] MDNS: Starting (devboard-cbu-123456789012345)
D [     11.954] MDNS: Starting mDNS on netif 1
D [     11.958] MDNS: Added IGMP to netif 1
D [     11.966] MDNS: mDNS started on netif 1, announcing it to network
D [     11.970] MDNS: Starting mDNS on netif 0
D [     11.976] MDNS: Added IGMP to netif 0
D [     11.982] MDNS: mDNS started on netif 0, announcing it to network
D [     11.988] MDNS: Add service: netif 1 / devboard-cbu-123456789012345 / _esphomelib / 1 / 6053
D [     11.998] MDNS: Add service: netif 0 / devboard-cbu-123456789012345 / _esphomelib / 1 / 6053
D [     12.008] MDNS: Add service: netif 1 / devboard-cbu-123456789012345 / _http / 1 / 80
D [     12.016] MDNS: Add service: netif 0 / devboard-cbu-123456789012345 / _http / 1 / 80
D [     12.964] MDNS: Status: netif 1, status 1
D [     12.992] MDNS: Status: netif 0, status 1
```
So we get 2 status lines at the end if successful, but nothing if not.

chrwei, 2024-03-09:
yeah you'd need to add more debug output to the code. the MDNS.addServiceTxt(service_type, proto, record.key.c_str(), record.value.c_str()); call is the issue, so the extra debug would need to be inside that addServiceTxt method to see where it's "crashing"

Cossid, 2024-03-09:
The issue isn't in addServiceTxt, but rather the callbacks

I modified mdnsTxtCallback to log every time it called mdns_resp_add_service_txtitem

Here is the output from a working case (cleaned up)
```
D [      6.000] MDNS: Starting (devboard-cbu-123456789012345)
D [      6.004] MDNS: Starting mDNS on netif 1
D [      6.010] MDNS: Added IGMP to netif 1
D [      6.016] MDNS: mDNS started on netif 1, announcing it to network
D [      6.022] MDNS: Starting mDNS on netif 0
D [      6.026] MDNS: Added IGMP to netif 0
D [      6.032] MDNS: mDNS started on netif 0, announcing it to network
D [      6.038] MDNS: Add service: netif 1 / devboard-cbu-123456789012345 / _esphomelib / 1 / 6053
D [      6.054] MDNS: Add service: netif 0 / devboard-cbu-123456789012345 / _esphomelib / 1 / 6053
D [      6.058] MDNS: mDNS::addServiceTxtImpl - item: friendly_name=Testing1234567890123456789012 service: _esphomelib, size: 1
D [      6.070] MDNS: mDNS::addServiceTxtImpl - item: version=2024.2.2 service: _esphomelib, size: 1
D [      6.080] MDNS: mDNS::addServiceTxtImpl - item: mac=d8d6685cad59 service: _esphomelib, size: 1
D [      6.090] MDNS: mDNS::addServiceTxtImpl - item: platform=BK7231N service: _esphomelib, size: 1
D [      6.100] MDNS: mDNS::addServiceTxtImpl - item: board=generic-bk7231n-qfn32-tuya service: _esphomelib, size: 1
D [      6.110] MDNS: mDNS::addServiceTxtImpl - item: network=wifi service: _esphomelib, size: 1
D [      6.120] MDNS: mDNS::addServiceTxtImpl - item: project_name=EarthBulb.A19_WB8P_SM2135_RGBWW service: _esphomelib, size: 1
D [      6.132] MDNS: mDNS::addServiceTxtImpl - item: project_version=0.0.1 service: _esphomelib, size: 1
D [      6.136] MDNS: mdnsTxtCallback adding record for service  record friendly_name=Testing1234567890123456789012 strlen 43
D [      6.146] MDNS: mdnsTxtCallback adding record for service +friendly_name=Testing1234567890123456789012 record version=2024.2.2 strlen 16
D [      6.160] MDNS: mdnsTxtCallback adding record for service +friendly_name=Testing1234567890123456789012version=2024.2.2 record mac=d8d6685cad59 strlen 16
D [      6.176] MDNS: mdnsTxtCallback adding record for service +friendly_name=Testing1234567890123456789012version=2024.2.2mac=d8d6685cad59 record platform=BK7231N strlen 16
D [      6.192] MDNS: mdnsTxtCallback adding record for service +friendly_name=Testing1234567890123456789012version=2024.2.2mac=d8d6685cad59platform=BK7231N record board=generic-bk7231n-qfn32-tuya strlen 32
D [      6.210] MDNS: mdnsTxtCallback adding record for service +friendly_name=Testing1234567890123456789012version=2024.2.2mac=d8d6685cad59platform=BK7231N board=generic-bk7231n-qfn32-tuya record network=wifi strlen 12
D [      6.232] MDNS: mdnsTxtCallback adding record for service +friendly_name=Testing1234567890123456789012version=2024.2.2mac=d8d6685cad59platform=BK7231N board=generic-bk7231n-qfn32-tuya
network=wifi record project_name=EarthBulb.A19_WB8P_SM2135_RGBWW strlen 44
D [      6.258] MDNS: mdnsTxtCallback adding record for service  record friendly_name=Testing1234567890123456789012 strlen 43
D [      6.268] MDNS: mdnsTxtCallback adding record for service +friendly_name=Testing1234567890123456789012 record version=2024.2.2 strlen 16
D [      6.280] MDNS: mdnsTxtCallback adding record for service +friendly_name=Testing1234567890123456789012version=2024.2.2 record mac=d8d6685cad59 strlen 16
D [      6.296] MDNS: mdnsTxtCallback adding record for service +friendly_name=Testing1234567890123456789012version=2024.2.2mac=d8d6685cad59 record platform=BK7231N strlen 16
D [      6.312] MDNS: mdnsTxtCallback adding record for service +friendly_name=Testing1234567890123456789012version=2024.2.2mac=d8d6685cad59platform=BK7231N record board=generic-bk7231n-qfn32-tuya strlen 32
D [      6.332] MDNS: mdnsTxtCallback adding record for service +friendly_name=Testing1234567890123456789012version=2024.2.2mac=d8d6685cad59platform=BK7231N board=generic-bk7231n-qfn32-tuya record network=wifi strlen 12
D [      6.352] MDNS: mdnsTxtCallback adding record for service +friendly_name=Testing1234567890123456789012version=2024.2.2mac=d8d6685cad59platform=BK7231N board=generic-bk7231n-qfn32-tuya
network=wifi record project_name=EarthBulb.A19_WB8P_SM2135_RGBWW strlen 44
D [      6.382] MDNS: Add service: netif 1 / devboard-cbu-123456789012345 / _http / 1 / 80
D [      6.392] MDNS: Add service: netif 0 / devboard-cbu-123456789012345 / _http / 1 / 80
D [      6.582] MDNS: mdnsTxtCallback adding record for service  record friendly_name=Testing1234567890123456789012 strlen 43
D [      6.590] MDNS: mdnsTxtCallback adding record for service +friendly_name=Testing1234567890123456789012 record version=2024.2.2 strlen 16
D [      6.604] MDNS: mdnsTxtCallback adding record for service +friendly_name=Testing1234567890123456789012version=2024.2.2 record mac=d8d6685cad59 strlen 16
D [      6.620] MDNS: mdnsTxtCallback adding record for service +friendly_name=Testing1234567890123456789012version=2024.2.2mac=d8d6685cad59 record platform=BK7231N strlen 16
D [      6.636] MDNS: mdnsTxtCallback adding record for service +friendly_name=Testing1234567890123456789012version=2024.2.2mac=d8d6685cad59platform=BK7231N record board=generic-bk7231n-qfn32-tuya strlen 32
D [      6.654] MDNS: mdnsTxtCallback adding record for service +friendly_name=Testing1234567890123456789012version=2024.2.2mac=d8d6685cad59platform=BK7231N board=generic-bk7231n-qfn32-tuya record network=wifi strlen 12
[... cut ...]
D [     15.390] MDNS: mdnsTxtCallback adding record for service +friendly_name=Testing1234567890123456789012version=2024.2.2mac=d8d6685cad59platform=BK7231N board=generic-bk7231n-qfn32-tuya
network=wifi record project_name=EarthBulb.A19_WB8P_SM2135_RGBWW_Bulb strlen 49
D [     15.416] MDNS: mdnsTxtCallback adding record for service +friendly_name=Testing1234567890123456789012version=2024.2.2mac=d8d6685cad59platform=BK7231N board=generic-bk7231n-qfn32-tuya
network=wifi1project_name=EarthBulb.A19_WB8P_SM2135_RGBWW_Bulb record project_version=0.0.1 strlen 21
```
This is truncated, but it would keep going on forever.

dwmw2, 2024-09-03:
I think it's just hitting the LwIP mDNS limits from [mdns.h](https://git.savannah.nongnu.org/cgit/lwip.git/tree/src/include/lwip/apps/mdns.h#n59):
```
#define MDNS_LABEL_MAXLEN  63
#define MDNS_DOMAIN_MAXLEN 256
```

I think those are too small. [RFC6763 §6.2](https://datatracker.ietf.org/doc/html/rfc6763#section-6.2) says:
```
[6.2](https://datatracker.ietf.org/doc/html/rfc6763#section-6.2).  DNS-SD TXT Record Size

   The total size of a typical DNS-SD TXT record is intended to be small
   -- 200 bytes or less.

   In cases where more data is justified (e.g., LPR printing [[BJP](https://datatracker.ietf.org/doc/html/rfc6763#ref-BJP)]),
   keeping the total size under 400 bytes should allow it to fit in a
   single 512-byte DNS message [[RFC1035](https://datatracker.ietf.org/doc/html/rfc1035)].

   In extreme cases where even this is not enough, keeping the size of
   the TXT record under 1300 bytes should allow it to fit in a single
   1500-byte Ethernet packet.
```
I can increase `MDNS_LABEL_MAXLEN` and that seems to allow long individual strings, but increasing `MDNS_DOMAIN_MAXLEN` just seems to make it crash.

dwmw2, 2024-09-03:
Turns out I can increase MDNS_LABEL_MAXLEN to 314, but not 315.

It's actually the size of the `name[]` array in `struct mdns_domain` which causes the crash; if I increase that alone, it crashes.