libretiny-eu/libretiny #257: Crash "RTL8195A Hard Fault Error" with LT 1.5.0

hn, 2024-02-25:
LT 1.5.0 crashes on EMW3080 (RTL8710BN clone) as reported here: https://github.com/hn/ginlong-solis/issues/22

LT 1.4.1 works fine on this MCU.

The crash is caused by https://github.com/libretiny-eu/libretiny/commit/1d80b5fff72cc31cba49f54fe7dd56357842031a from https://github.com/libretiny-eu/libretiny/pull/226 (checked by adding (cherry-picking) only this patch to 1.4.1 -> MCU crashes).

kuba2k2, 2024-02-25:
@szupi-ipuzs 
Something broke, I think. I will try to find the source of the issue.

kuba2k2, 2024-02-26:
[Here](https://github.com/libretiny-eu/libretiny/blob/1d80b5fff72cc31cba49f54fe7dd56357842031a/cores%2Frealtek-amb%2Farduino%2Flibraries%2FWiFi%2FWiFiScan.cpp) is the modified Realtek WiFi code. It seems that scanAlloc() returns the amount of allocated items, then the Realtek code tries to assign to the last item. The index in [] is wrong though, it tries to assign to one item past the last one.

Can you try adding `last--;` right after the if() that checks if the allocation succeeded? That should fix the issue.

hn, 2024-02-26:
```
rtl87xx:
  board: generic-rtl8710bx-4mb-980k
  framework:
    version: 0.0.0
    source: https://github.com/hn/libretiny#fix-257-wifi-scan
```

Et voilà ... it works :) MCU boots flawlessly with `Version: v1.5.0+sha.72d46c9 on generic-rtl8710bx-4mb-980k`

kuba2k2, 2024-02-26:
PRs are welcome :)

Also, since you're using the generic BX version in your project, we can add that EMW board natively to LibreTiny. It could even have its own flash layout and pinout drawing to best match that board.

PR #258 (2024-02-26): [realtek-ambz] Fix crash after WiFi scan
```
[C][wifi:038]: Setting up WiFi...
[C][wifi:051]: Starting WiFi...
[C][wifi:052]:   Local MAC: 80:A0:36:A7:F7:1F
interface 0 is initialized
                          interface 1 is initialized

Initializing WIFI ...
WIFI initialized
                [D][wifi:459]: Starting scan...
RTL8195A[HAL]: Hard Fault Error!!!!
RTL8195A[HAL]: R0 = 0x1b9
```

As explained by @kuba2k2: "It seems that scanAlloc() returns the amount of allocated items, then the Realtek code tries to assign to the last item. The index in [] is wrong though, it tries to assign to one i [... cut]
closed 2024-02-26 by kuba2k2