libretiny-eu/libretiny #262: support for BK7231M(CB2S)

shzlww, 2024-03-08:
**Describe the bug**
I have a smart plug(not Tuya version) with cb2s module ,but the module is using a BK7231M.
I try to flash it with ltchiptool,but i get CRC error at 28%, never success,then I try to flash it with bk7231flasher of OpenBeken, When selecting BK7231N, the Module would flash just fine, but when power cycling it, it would not boot and the AP would not come up.

So, Could BK7231M be support by libretiny?

**Screenshots**
[image]

**Additional context**
Original firmware dump by bk7231flasher_1.1.6
[readResult_BK7231N_QIO_2024-02-3-18-21-04.bin.gz](https://github.com/openshwprojects/OpenBK7231T_App/files/14491950/readResult_BK7231N_QIO_2024-02-3-18-21-04.bin.gz)

logs from bk7231flasher_1.1.6:

> Backup name has not been set, so output file will only contain flash type/date.
> 
> Starting read!
> Now is: 2024年3月2日 18:17:31.
> Flasher mode: BK7231N
> Going to open port: COM3.
> Serial port open!
> Getting bus... (now, please do reboot by CEN or by power off/on)
> Getting bus failed, will try again - 0/100!
> Getting bus success!
> Going to set baud rate setting (921600)!
> Failed to set baud rate!
> There was no result to save.
> Backup name has not been set, so output file will only contain flash type/date.
> Starting read!
> Now is: 2024年3月2日 18:17:50.
> Flasher mode: BK7231N
> Going to open port: COM3.
> Serial port open!
> Getting bus... (now, please do reboot by CEN or by power off/on)
> Getting bus success!
> Going to set baud rate setting (115200)!
> Will try to read device flash MID (for unprotect N):
> Flash MID loaded: 1560EB
> Will now search for Flash def in out database...
> Flash def found! For: 1560EB
> Flash information: mid: 1560EB, icName: TH25Q_16HB, manufacturer: TH, szMem: 1000000, szSR: 2, cwUnp: 0, cwEnp: 7, cwMsk: 407C, sb: 2, lb: 5, cwdRd: 05-35-FF-FF, cwdWr: 01-FF-FF-FF
> Entering SetProtectState(True)...
> sr: 4
> sr: 4004
> final sr: 4004
> msk: 407c
> cw: 0, sb: 2, lb: 5
> bfd: 0
> sr: 0
> sr: 0
> final sr: 0
> msk: 407c
> cw: 0, sb: 2, lb: 5
> bfd: 0
> SetProtectState(True) success!
> Going to start reading at offset 0x00...
> Reading 0x00... Ok! Reading 0x1000... Ok! Reading 0x2000... Ok! Reading 0x3000... Ok! Reading 0x4000... Ok! Reading 0x5000... Ok! Reading 0x6000... Ok! Reading 0x7000... Ok! Reading 0x8000... Ok! Reading 0x9000... Ok! Reading 0xA000... Ok! Reading 0xB000... Ok! Reading 0xC000... Ok! Reading 0xD000... Ok! Reading 0xE000... Ok! Reading 0xF000... Ok! Reading 0x10000... Ok! Reading 0x11000... Ok! Reading 0x12000... Ok! Reading 0x13000... Ok! Reading 0x14000... Ok! Reading 0x15000... Ok! Reading 0x16000... Ok! Reading 0x17000... Ok! Reading 0x18000... Ok! Reading 0x19000... Ok! Reading 0x1A000... Ok! Reading 0x1B000... Ok! Reading 0x1C000... Ok! Reading 0x1D000... Ok! Reading 0x1E000... Ok! Reading 0x1F000... Ok! Reading 0x20000... Ok! Reading 0x21000... Ok! Reading 0x22000... Ok! Reading 0x23000... Ok! Reading 0x24000... Ok! Reading 0x25000... Ok! Reading 0x26000... Ok! Reading 0x27000... Ok! Reading 0x28000... Ok! Reading 0x29000... Ok! Reading 0x2A000... Ok! Reading 0x2B000... Ok! Reading 0x2C000... Ok! Reading 0x2D000... Ok! Reading 0x2E000... Ok! Reading 0x2F000... Ok! Reading 0x30000... Ok! Reading 0x31000... Ok! Reading 0x32000... Ok! Reading 0x33000... Ok! Reading 0x34000... Ok! Reading 0x35000... Ok! Reading 0x36000... Ok! Reading 0x37000... Ok! Reading 0x38000... Ok! Reading 0x39000... Ok! Reading 0x3A000... Ok! Reading 0x3B000... Ok! Reading 0x3C000... Ok! Reading 0x3D000... Ok! Reading 0x3E000... Ok! Reading 0x3F000... Ok! Reading 0x40000... Ok! Reading 0x41000... Ok! Reading 0x42000... Ok! Reading 0x43000... Ok! Reading 0x44000... Ok! Reading 0x45000... Ok! Reading 0x46000... Ok! Reading 0x47000... Ok! Reading 0x48000... Ok! Reading 0x49000... Ok! Reading 0x4A000... Ok! Reading 0x4B000... Ok! Reading 0x4C000... Ok! Reading 0x4D000... Ok! Reading 0x4E000... Ok! Reading 0x4F000... Ok! Reading 0x50000... Ok! Reading 0x51000... Ok! Reading 0x52000... Ok! Reading 0x53000... Ok! Reading 0x54000... Ok! Reading 0x55000... Ok! Reading 0x56000... Ok! Reading 0x57000... Ok! Reading 0x58000... Ok! Reading 0x59000... Ok! Reading 0x5A000... Ok! Reading 0x5B000... Ok! Reading 0x5C000... Ok! Reading 0x5D000... Ok! Reading 0x5E000... Ok! Reading 0x5F000... Ok! Reading 0x60000... Ok! Reading 0x61000... Ok! Reading 0x62000... Ok! Reading 0x63000... Ok! Reading 0x64000... Ok! Reading 0x65000... Ok! Reading 0x66000... Ok! Reading 0x67000... Ok! Reading 0x68000... Ok! Reading 0x69000... Ok! Reading 0x6A000... Ok! Reading 0x6B000... Ok! Reading 0x6C000... Ok! Reading 0x6D000... Ok! Reading 0x6E000... Ok! Reading 0x6F000... Ok! Reading 0x70000... Ok! Reading 0x71000... Ok! Reading 0x72000... Ok! Reading 0x73000... Ok! Reading 0x74000... Ok! Reading 0x75000... Ok! Reading 0x76000... Ok! Reading 0x77000... Ok! Reading 0x78000... Ok! Reading 0x79000... Ok! Reading 0x7A000... Ok! Reading 0x7B000... Ok! Reading 0x7C000... Ok! Reading 0x7D000... Ok! Reading 0x7E000... Ok! Reading 0x7F000... Ok! Reading 0x80000... Ok! Reading 0x81000... Ok! Reading 0x82000... Ok! Reading 0x83000... Ok! Reading 0x84000... Ok! Reading 0x85000... Ok! Reading 0x86000... Ok! Reading 0x87000... Ok! Reading 0x88000... Ok! Reading 0x89000... Ok! Reading 0x8A000... Ok! Reading 0x8B000... Ok! Reading 0x8C000... Ok! Reading 0x8D000... Ok! Reading 0x8E000... Ok! Reading 0x8F000... Ok! Reading 0x90000... Ok! Reading 0x91000... Ok! Reading 0x92000... Ok! Reading 0x93000... Ok! Reading 0x94000... Ok! Reading 0x95000... Ok! Reading 0x96000... Ok! Reading 0x97000... Ok! Reading 0x98000... Ok! Reading 0x99000... Ok! Reading 0x9A000... Ok! Reading 0x9B000... Ok! Reading 0x9C000... Ok! Reading 0x9D000... Ok! Reading 0x9E000... Ok! Reading 0x9F000... Ok! Reading 0xA0000... Ok! Reading 0xA1000... Ok! Reading 0xA2000... Ok! Reading 0xA3000... Ok! Reading 0xA4000... Ok! Reading 0xA5000... Ok! Reading 0xA6000... Ok! Reading 0xA7000... Ok! Reading 0xA8000... Ok! Reading 0xA9000... Ok! Reading 0xAA000... Ok! Reading 0xAB000... Ok! Reading 0xAC000... Ok! Reading 0xAD000... Ok! Reading 0xAE000... Ok! Reading 0xAF000... Ok! Reading 0xB0000... Ok! Reading 0xB1000... Ok! Reading 0xB2000... Ok! Reading 0xB3000... Ok! Reading 0xB4000... Ok! Reading 0xB5000... Ok! Reading 0xB6000... Ok! Reading 0xB7000... Ok! Reading 0xB8000... Ok! Reading 0xB9000... Ok! Reading 0xBA000... Ok! Reading 0xBB000... Ok! Reading 0xBC000... Ok! Reading 0xBD000... Ok! Reading 0xBE000... Ok! Reading 0xBF000... Ok! Reading 0xC0000... Ok! Reading 0xC1000... Ok! Reading 0xC2000... Ok! Reading 0xC3000... Ok! Reading 0xC4000... Ok! Reading 0xC5000... Ok! Reading 0xC6000... Ok! Reading 0xC7000... Ok! Reading 0xC8000... Ok! Reading 0xC9000... Ok! Reading 0xCA000... Ok! Reading 0xCB000... Ok! Reading 0xCC000... Ok! Reading 0xCD000... Ok! Reading 0xCE000... Ok! Reading 0xCF000... Ok! Reading 0xD0000... Ok! Reading 0xD1000... Ok! Reading 0xD2000... Ok! Reading 0xD3000... Ok! Reading 0xD4000... Ok! Reading 0xD5000... Ok! Reading 0xD6000... Ok! Reading 0xD7000... Ok! Reading 0xD8000... Ok! Reading 0xD9000... Ok! Reading 0xDA000... Ok! Reading 0xDB000... Ok! Reading 0xDC000... Ok! Reading 0xDD000... Ok! Reading 0xDE000... Ok! Reading 0xDF000... Ok! Reading 0xE0000... Ok! Reading 0xE1000... Ok! Reading 0xE2000... Ok! Reading 0xE3000... Ok! Reading 0xE4000... Ok! Reading 0xE5000... Ok! Reading 0xE6000... Ok! Reading 0xE7000... Ok! Reading 0xE8000... Ok! Reading 0xE9000... Ok! Reading 0xEA000... Ok! Reading 0xEB000... Ok! Reading 0xEC000... Ok! Reading 0xED000... Ok! Reading 0xEE000... Ok! Reading 0xEF000... Ok! Reading 0xF0000... Ok! Reading 0xF1000... Ok! Reading 0xF2000... Ok! Reading 0xF3000... Ok! Reading 0xF4000... Ok! Reading 0xF5000... Ok! Reading 0xF6000... Ok! Reading 0xF7000... Ok! Reading 0xF8000... Ok! Reading 0xF9000... Ok! Reading 0xFA000... Ok! Reading 0xFB000... Ok! Reading 0xFC000... Ok! Reading 0xFD000... Ok! Reading 0xFE000... Ok! Reading 0xFF000... Ok! Reading 0x100000... Ok! Reading 0x101000... Ok! Reading 0x102000... Ok! Reading 0x103000... Ok! Reading 0x104000... Ok! Reading 0x105000... Ok! Reading 0x106000... Ok! Reading 0x107000... Ok! Reading 0x108000... Ok! Reading 0x109000... Ok! Reading 0x10A000... Ok! Reading 0x10B000... Ok! Reading 0x10C000... Ok! Reading 0x10D000... Ok! Reading 0x10E000... Ok! Reading 0x10F000... Ok! Reading 0x110000... Ok! Reading 0x111000... Ok! Reading 0x112000... Ok! Reading 0x113000... Ok! Reading 0x114000... Ok! Reading 0x115000... Ok [... cut ...] ector 0x120000... ok! Writing sector 0x121000... ok! Writing sector 0x122000... ok! Writing sector 0x123000... ok! Writing sector 0x124000... ok! Writing sector 0x125000... ok! Writing sector 0x126000... ok! Writing sector 0x127000... ok! Writing sector 0x128000... ok! Writing sector 0x129000... ok! Starting CRC check for 298 sectors, starting at offset 00
> CRC matches 0x5A99C1C1!
> 
> NOTE: the OBK config writing is disabled, so not writing anything extra.
> Writing file data to chip successs.

Cossid, 2024-03-08:
Well, that certainly is an oddity.  That firmware is unencrypted, and appears to have the BK7231N 1.0.11 bootloader, and doesn't have Tuya firmware, but some other 3rd party.  I'm not sure if the `M` is a misprint or not, but there are references to both `BK7231N` and `BK7231M` in the strings.

I'm guessing this would already work with a custom coeff for unencrypted, but I'm unsure what that looks like (and the corresponding ota key/iv and possible ota offset), so I'll leave that to @kuba2k2 to advise on.

kuba2k2, 2024-03-08:
This device indeed appears to be unencrypted. ESPHome config for this device:

```yaml
esphome:
  name: my-device
  platformio_options:
    board_build.bkcrypt_coeffs: 00000000000000000000000000000000
    board_build.bkota.key: 0123456789ABCDEF0123456789ABCDEF
    board_build.bkota.iv: 0123­456789ABCDEF
    board_flash.download: "0x132000+0xAE000"
```

shzlww, 2024-03-08:
compile and flash well,but can not run.No wifi connection and  No ap .
```yaml
Linking .pioenvs/bk7231m/raw_firmware.elf
|-- Image 1: raw_firmware.elf
|   |-- image_bk7231n_app.0x011000.rbl
|   |   |-- raw_firmware.bin
|   |-- image_bk7231n_app.0x011000.crc
|   |-- image_bk7231n_app.0x129F0A.rblh
|   |-- image_bk7231n_app.ota.rbl
|   |-- image_bk7231n_app.ota.ug.bin
RAM:   [====      ]  35.8% (used 93944 bytes from 262144 bytes)
Flash: [=======   ]  72.3% (used 783240 bytes from 1083136 bytes)
Building UF2 OTA image
|-- esphome_2023.12.5_generic-bk7231n-qfn32-tuya_bk7231n_lt1.4.1.uf2
|-- firmware.uf2
|-- firmware.bin
```
```yaml
esphome:
  name: bk7231m
  platformio_options:
    board_build.bkcrypt_coeffs: "00000000000000000000000000000000"
    board_build.bkota.key: "0123456789ABCDEF0123456789ABCDEF"
    board_build.bkota.iv: "0123456789ABCDEF"
    board_flash.download: "0x132000+0xAE000"
    #board_flash.download: "0x1E0000"
bk72xx:
  board: generic-bk7231n-qfn32-tuya

logger:

web_server:

captive_portal:

mdns:

api:
  password: ""

ota:
  password: ""
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  ap:
```

kuba2k2, 2024-03-08:
How did you flash the firmware, and what file did you flash?

Please connect UART to TX2 and capture log outputs from the device. Otherwise we can't even tell if it's running or not.

shzlww, 2024-03-09:
I try to flash the bk7231m by using many ways, same result: no wifi,no ap,no uart log
1.flash bk7231m.uf2 by ltchiptool 
2.flash bk7231m-0x011000.rbl by bk7231_flasher
finally,I flash it back to the original factory firmware by bk7231_flasher, I got the uart log
```yaml
[0;34m[D ELEC][bll_elec_manage.c:246]upload CHECK
[0mbk_misc_init_start_type 0 0
prvHeapInit-start addr:0x40c328, size:146648
[Flash]id:0xeb6015
[Flash]init over
sctrl_sta_ps_init
SDK Rev: 3.0.56 fa12785
[THD]app:[tcb]40ccf8 [stack]40c4f0-40ccf0:2048:5
[THD]extended_app:[tcb]40d568 [stack]40cd60-40d560:2048:4
[THD]idle:[tcb]40d9d8 [stack]40d5d0-40d9d0:1024:0
[THD]timer_thd:[tcb]40e360 [stack]40db58-40e358:2048:2
OSK Rev: F-3.0.35 fa12785
cset:0 0 0 0
bandgap_calm_in_efuse=0x3e
[load]bandgap_calm=0x20->0x1e,vddig=4->5
[FUNC]rwnxl_init
chip id=7231c device id=20521028
IP Rev: W4-3.0.56-P0
txdesc flush
[FUNC]intc_init
[FUNC]calibration_main
get rfcali_mode:1
device_id=0x20521028
bk_misc_init_start_type 0 0
prvHeapInit-start addr:0x40c328, size:146648
[Flash]id:0xeb6015
[Flash]init over
sctrl_sta_ps_init
SDK Rev: 3.0.56 fa12785
[THD]app:[tcb]40ccf8 [stack]40c4f0-40ccf0:2048:5
[THD]extended_app:[tcb]40d568 [stack]40cd60-40d560:2048:4
[THD]idle:[tcb]40d9d8 [stack]40d5d0-40d9d0:1024:0
[THD]timer_thd:[tcb]40e360 [stack]40db58-40e358:2048:2
OSK Rev: F-3.0.35 fa12785
cset:0 0 0 0
bandgap_calm_in_efuse=0x3e
[load]bandgap_calm=0x20->0x1e,vddig=4->5
[FUNC]rwnxl_init
chip id=7231c device id=20521028
IP Rev: W4-3.0.56-P0
txdesc flush
[FUNC]intc_init
[FUNC]calibration_main
get rfcali_mode:1
device_id=0x20521028
calibration_main over
flash txpwr table:0x1f
dif g and n20 ID in flash:4
dif g and n40 ID in flash:4
load txpwr tabe n20 in flash
read txpwr tab from flash success
uncali adc value:[00 00 00]
temp in flash is:321
[THD]temp_detct:[tcb]40e950 [stack]40e548-40e948:1024:3
lpf_i & q in flash is:113, 107
found flash XTAL:66
xtal in flash is:66
xtal_cali:66
--init_xtal = 66
[FUNC]ps_init
int watchdog enabled, period=10000
task watchdog enabled, period=60000
[FUNC]func_init_extended OVER!!!

start_type:0
[... 387 lines cut ...]
[0m[0;34m[D MQTT][bll_dev_mqtt.c:636]FreeHeapSize 56816
[0mMQTT RECV {"data":{"params":[{"paramCode":"powerConsumptionClear","paramValue":"1"}]},"function":"Control","deviceId":"CMCC-590384-BCFD0C785DDE","seqId":"HJQI_17605323062_93f56815"}
[0;34m[D MPRT][fml_mqtt_protocol.c:115]funcion: Control
[0m[0;34m[D MPRT][fml_mqtt_protocol.c:134]paramCode: powerConsumptionClear, paramValue: 1
[0m[0;34m[D MQTT][bll_dev_mqtt.c:748]handle: 3
[0m[0;32m[I  MNG][bll_dev_common.c:657]energy set: 0
[0m[0;32m[I ELEC][bll_elec.c:318]WRITE node, part: 9, num: 3, int_val: 0, f_val: 0
[0m[0;32m[I ELEC][bll_elec.c:318]WRITE node, part: 8, num: 3, int_val: 0, f_val: 0
[0m[0;32m[I MQTT][bll_dev_mqtt.c:529]MQTT send: 4, i: 0
[0mlen: [218]
{"data":{"params":[{"paramCode":"powerConsumption","paramValue":"0.000"}]},"deviceId":"CMCC-590384-BCFD0C785DDE","childDeviceId":"","eventType":"ParamChange","timestamp":1.709958359e+12,"seqId":"DEV_28098265890595085"}
packet_type:4
[0;34m[D MQTT][bll_dev_mqtt.c:181]mqtt nitice event 2
[0m[0;32m[I MQTT][bll_dev_mqtt.c:623]MQTT send ret: 0
[0m[0;34m[D MQTT][bll_dev_mqtt.c:636]FreeHeapSize 56752
[0m[0;32m[I MQTT][bll_dev_mqtt.c:529]MQTT send: 4, i: 0
[0mlen: [219]
{"data":{"params":[{"paramCode":"powerConsumptionClear","paramValue":"1"}]},"deviceId":"CMCC-590384-BCFD0C785DDE","childDeviceId":"","eventType":"ParamChange","timestamp":1.709958359e+12,"seqId":"DEV_28098265897682905"}
packet_type:4
[0;34m[D MQTT][bll_dev_mqtt.c:181]mqtt nitice event 2
[0m[0;32m[I MQTT][bll_dev_mqtt.c:623]MQTT send ret: 0
[0m[0;34m[D MQTT][bll_dev_mqtt.c:636]FreeHeapSize 56816
[0m[0;34m[D ELEC][bll_elec_manage.c:246]upload CHECK
[0m
```

kuba2k2, 2024-03-09:
I didn't mean UART log of the stock firmware, it's not useful. If it's not Tuya, it's not unusual that it doesn't use AP mode to connect.

We need UART logs after flashing either the UF2 file by ltchiptool or the RBL file by bkflasher. So, choose either of these methods (one that works), flash the file, then dump the flash again. Attach here the UART log after flashing and the resulting dump file after flashing.

shzlww, 2024-03-09:
@kuba2k2 no uart log by normal flash the uf2 or rbl . So I did some hex trick with the dumped flash bin file (pre-flashed with uf2), I copy the stock firmware bin  [0x0~ 0x00011115]  to the dumped uf2[0x0 ~ 0x00011115] ,It actually started running!!!!!
log:
```yaml
[1;32mI [      0.000] [0mLibreTiny v1.4.1 on generic-bk7231n-qfn32-tuya, compiled at Mar  8 2024 21:48:43, GCC 10.3.1 (-O1)
[0;32m[I][logger:399]: Log initialized[0m
[0;35m[C][ota:473]: There have been 2 suspected unsuccessful boot attempts.[0m
[0;36m[D][lt.preferences:104]: Saving 1 preferences to flash...[0m
[0;36m[D][lt.preferences:132]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed[0m
[0;32m[I][app:029]: Running through setup()...[0m
[0;35m[C][wifi:038]: Setting up WiFi...[0m
[0;35m[C][wifi:051]: Starting WiFi...[0m
[0;35m[C][wifi:052]:   Local聽MAC: C8:47:8C:42:00:48[0m
€[1;32mI [      0.000] [0mLibreTiny v1.4.1 on generic-bk7231n-qfn32-tuya, compiled at Mar  8 2024 21:48:43, GCC 10.3.1 (-O1)
[0;32m[I][logger:399]: Log initialized[0m
[0;35m[C][ota:473]: There have been 3 suspected unsuccessful boot attempts.[0m
[0;36m[D][lt.preferences:104]: Saving 1 preferences to flash...[0m
[0;36m[D][lt.preferences:132]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed[0m
[0;32m[I][app:029]: Running through setup()...[0m
[0;35m[C][wifi:038]: Setting up WiFi...[0m
[0;35m[C][wifi:051]: Starting WiFi...[0m
[0;35m[C][wifi:052]:   Local聽MAC: C8:47:8C:42:00:48[0m
[0;35m[C][wifi:250]: Setting up AP...[0m
[0;35m[C][wifi:252]:   AP SSID: 'bk7231m hotspot'[0m
[0;35m[C][wifi:253]:   AP Password: '******'[0m
[0;35m[C][wifi:262]:   IP Address: 192.168.4.1[0m
[0;36m[D][wifi:455]: Starting scan...[0m
[0;35m[C][web_server:123]: Setting up web server...[0m
[0;35m[C][ota:097]: Over-The-Air Updates:[0m
[0;35m[C][ota:098]:   Address: bk7231m.local:8892[0m
[0;33m[W][ota:106]: Last Boot was an unhandled reset, will proceed to safe mode in 7 restarts[0m
[0;35m[C][api:025]: Setting up Home Assistant API server...[0m
[0;32m[I][app:062]: setup() finished successfully![0m
[0;32m[I][app:102]: ESPHome version 2023.12.5 compiled on Mar  8 2024, 21:47:48[0m
[0;35m[C][logger:443]: Logger:[0m
[0;35m[C][logger:444]:   Level: DEBUG[0m
[0;35m[C][logger:445]:   Log Baud Rate: 115200[0m
[0;35m[C][logger:447]:   Hardware UART: UART2[0m
[0;35m[C][captive_portal:088]: Captive Portal:[0m
[0;35m[C][wifi:573]: WiFi:[0m
[0;35m[C][wifi:405]:   Local MAC: C8:47:8C:42:00:48[0m
[0;35m[C][wifi:410]:   SSID: [5m''[6m[0m
[0;35m[C][wifi:411]:   IP Address: 0.0.0.0[0m
[0;35m[C][wifi:412]:   BSSID: [5m00:00:00:00:00:00[6m[0m
[0;35m[C][wifi:414]:   Hostname: 'bk7231m'[0m
[0;35m[C][wifi:416]:   Signal strength: 0 dB [0;32m鈻傗杽鈻嗏枅[0m[0m
[0;35m[C][wifi:420]:   Channel: 0[0m
[0;35m[C][wifi:421]:   Subnet: 0.0.0.0[0m
[0;35m[C][wifi:422]:   Gateway: 0.0.0.0[0m
[0;35m[C][wifi:423]:   DNS1: 0.0.0.0[0m
[0;35m[C][wifi:424]:   DNS2: 0.0.0.0[0m
[0;35m[C][web_server:168]: Web Server:[0m
[0;35m[C][web_server:169]:   Address: bk7231m.local:80[0m
[0;35m[C][mdns:115]: mDNS:[0m
[0;35m[C][mdns:116]:   Hostname: bk7231m[0m
[0;35m[C][ota:097]: Over-The-Air Updates:[0m
[0;35m[C][ota:098]:   Address: bk7231m.local:8892[0m
[0;33m[W][ota:106]: Last Boot was an unhandled reset, will proceed to safe mode in 7 restarts[0m
[0;35m[C][api:139]: API Server:[0m
[0;35m[C][api:140]:   Address: bk7231m.local:6053[0m
[0;35m[C][api:144]:   Using noise encryption: NO[0m
[0;35m[C][lt.component:013]: LibreTiny:[0m
[0;35m[C][lt.component:014]:   Version: v1.4.1 on generic-bk7231n-qfn32-tuya, compiled at Mar  8 2024 21:47:00, GCC 10.3.1 (-O1)[0m
[0;35m[C][lt.component:015]:   Loglevel: 3[0m
[1;33mW [    181.394] [0mrx timeout 4
[0;32m[I][captive_portal:031]: Captive Portal Requested WiFi Settings Change:[0m
[0;32m[I][captive_portal:032]:   SSID='******'[0m
[0;32m[I][captive_portal:033]:   Password=[5m'******'[6m[0m
[0;36m[D][lt.preferences:104]: Saving 1 preferences to flash...[0m
[0;36m[D][lt.preferences:132]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed[0m
[0;36m[D][wifi:455]: Starting scan...[0m
[... 26 lines cut ...]
[0;36m[D][wifi:518]: - [5m'Jiang'[6m [5m(34:F7:16:E5:3F:98) [6m[0;33m鈻傗杽[0;37m鈻嗏枅[0m[0m
[0;36m[D][wifi:518]: - [5m'CU_XDDY'[6m [5m(E0:A1:CE:0D:D2:98) [6m[0;33m鈻傗杽[0;37m鈻嗏枅[0m[0m
[0;36m[D][wifi:518]: - [5m'ChinaNet-R5xz'[6m [5m(C0:B1:01:89:00:78) [6m[0;31m鈻?[0;37m鈻勨枂鈻?[0m[0m
[0;32m[I][wifi:300]: WiFi Connecting to '******'...[0m
[0;33m[W][component:214]: Component wifi took a long time for an operation (0.16 s).[0m
[0;33m[W][component:215]: Components should block for at most 20-30ms.[0m
[0;32m[I][wifi:587]: WiFi Connected![0m
[0;35m[C][wifi:405]:   Local MAC: C8:47:8C:42:00:48[0m
[0;35m[C][wifi:410]:   SSID: [5m'******'[6m[0m
[0;35m[C][wifi:411]:   IP Address: 192.168.1.125[0m
[0;35m[C][wifi:412]:   BSSID: [5m6A:DB:54:E6:04:93[6m[0m
[0;35m[C][wifi:414]:   Hostname: 'bk7231m'[0m
[0;35m[C][wifi:416]:   Signal strength: -42 dB [0;32m鈻傗杽鈻嗏枅[0m[0m
[0;35m[C][wifi:420]:   Channel: 11[0m
[0;35m[C][wifi:421]:   Subnet: 255.255.255.0[0m
[0;35m[C][wifi:422]:   Gateway: 192.168.1.1[0m
[0;35m[C][wifi:423]:   DNS1: 192.168.1.1[0m
[0;35m[C][wifi:424]:   DNS2: 192.168.1.1[0m
[0;36m[D][wifi:596]: Disabling AP...[0m
[0;33m[W][component:214]: Component wifi took a long time for an operation (0.06 s).[0m
[0;33m[W][component:215]: Components should block for at most 20-30ms.[0m
[0;32m[I][ota:117]: Boot seems successful, resetting boot loop counter.[0m
[0;36m[D][lt.preferences:104]: Saving 1 preferences to flash...[0m
[0;36m[D][lt.preferences:132]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed[0m
```

kuba2k2, 2024-03-09:
Interesting, the UF2 file doesn't modify the bootloader area. Have you been flashing OpenBeken QIO binary in the meantime? The QIO contains an encrypted bootloader, which will not work on your device and will prevent any firmware from running, unless you reflash the stock bin file.

Also, can you attach a screenshot of "Read chip info" from ltchiptool? This might help troubleshoot the CRC errors you were getting.

shzlww, 2024-03-09:
@kuba2k2 
[image]

shzlww, 2024-03-09:
> Have you been flashing OpenBeken QIO binary in the meantime?
OpenBeken QIO binary do not work,same as uf2 binary, no wifi/no ap/no uart log

kuba2k2, 2024-03-09:
You should be able to flash the .bin file by ltchiptool if you uncheck the "auto detect" box and clear the start address, skip offset and writing length boxes. Also set the chip family to BK72xx.

That being said, because your chip is unrecognized, it will fail all CRC checks because BK7231N does this differently. There is a pending update of bk7231tools that addresses this issue.

shzlww, 2024-03-09:
@kuba2k2 

> You should be able to flash the .bin file by ltchiptool if you uncheck the "auto detect" box and clear the start address, skip offset and writing length boxes. Also set the chip family to BK72xx.

I just use ltchiptool to flash back to stock firmware ,process 100%,get crc error,but the stock firmware can work normally.May be  the way of crc different between ltchiptool and bkflasher,this stock firmware was dumped by bkflasher, I think this crc error does not matter.
[image]

shzlww, 2024-03-10:
@kuba2k2 After re-flashed a new device with bk7231m, I think my previous description about flashing was wrong.
I should re-describe about it:
1.At the beginning, I use ltchiptool to flash .uf2 into bk7231m,but always get an CRC error.
2.Then, I use bkflasher to flash OpenBK7231N_QIO.bin into it , flash successfully ,but it can not work.
3.After then,I use ltchiptool to flash .uf2, flash successfully without CRC error. it can not work,either.
4.In step 3, I notice that ltchiptool flash .uf2 starting from 0x011000, so I think the hex data before 0x011000 should be bootloader,then I dump the flash of bk7231m by bkflasher.
5.I copy the bootloader of stock firmware (0x0~0x010FFF)  to the same area of dumped binary file.
6.At last,I flash this strange combination binary file into bk7231m.Out of my expectation,all components work well, including wifi,ap,dhcp,uartlog,webserver,ota,etc..
7.The only issue is that all mac addresses of bk7231m are the same after flashed. I solved it by using bkflasher(Restore RF) to generate a new random mac address.

kuba2k2, 2024-03-10:
In step 2, where you flashed OpenBeken QIO - that's the mistake. OpenBeken's files currently will not work with your device. Doing that, you overwrite the bootloader (0x0-0x11000) with the wrong one, which makes your device fail to boot.

ad. 7.: your device might be using a different RF partition address. BkFlasher's "Restore RF" function might indeed generate a random MAC address, however it will **NOT** write the correct Wi-Fi calibration data. You might get reduced Wi-Fi range and/or performance, or other issues.

The correct solution would be to update ESPHome/LibreTiny's partition layout:
```yaml
esphome:
  name: my-device
  platformio_options:
    board_build.bkcrypt_coeffs: 00000000000000000000000000000000
    board_build.bkota.key: 0123456789ABCDEF0123456789ABCDEF
    board_build.bkota.iv: 0123­456789ABCDEF
    board_flash.download: "0x132000+0xA6000"
    board_flash.kvs: "0x1D8000+0x8000"
    board_flash.calibration: "0x1E0000+0x1000"
    board_flash.tlv: "0x1E1000+0x1000"
    board_flash.net: "0x1E2000+0x1000"
    board_flash.userdata: "0x1E3000+0x1D000"
```
(taken from BK7231T default partition layout, offsets match your device)

If that doesn't work, restore the stock firmware and try again.

shzlww, 2024-03-11:
Should I modify  the board_flash.app? I found there are some data before 0x132000 in the stock firmware binary file.
[image]
[image]

kuba2k2, 2024-03-11:
No, this is part of the app partition. This is the RBL header, it's not even important for running the app. You don't need to change the app partition.

shzlww, 2024-03-12:
@kuba2k2  it always get a CRC error when flash the uf2 file with ltchiptool, so I copy all data of bk7231m-1-0x011000.rbl to the stock firmware(2M),and then flash this new 2Mb file to the bk7231m.
 I just tried this method: ap work ok, but it can not scan wifi.
[image]

UART LOG : local mac address is correct,but wifi scan timeout.
```yaml
[1;32mI [      0.000] [0mLibreTiny v1.4.1 on generic-bk7231n-qfn32-tuya, compiled at Mar 12 2024 02:00:25, GCC 10.3.1 (-O1)
[0;32m[I][logger:399]: Log initialized[0m
[0;35m[C][status_led:014]: Setting up Status LED...[0m
[0;35m[C][ota:473]: There have been 0 suspected unsuccessful boot attempts.[0m
[0;36m[D][lt.preferences:104]: Saving 1 preferences to flash...[0m
[0;36m[D][lt.preferences:132]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed[0m
[0;32m[I][app:029]: Running through setup()...[0m
[0;35m[C][switch.gpio:011]: Setting up GPIO Switch 'Relay 1'...[0m
[0;36m[D][switch:016]: 'Relay 1' Turning OFF.[0m
[0;36m[D][switch:055]: 'Relay 1': Sending state OFF[0m
[0;36m[D][light:036]: 'id_light_relay' Setting:[0m
[0;36m[D][light:041]:   Color mode: [0m
[0;36m[D][switch:016]: 'Relay 1' Turning OFF.[0m
[0;36m[D][binary_sensor:034]: 'binary_switch': Sending initial state OFF[0m
[0;35m[C][light:035]: Setting up light 'id_light_relay'...[0m
[0;36m[D][light:036]: 'id_light_relay' Setting:[0m
[0;35m[C][hlw8012:014]: Setting up HLW8012...[0m
[0;35m[C][wifi:038]: Setting up WiFi...[0m
[0;35m[C][wifi:051]: Starting WiFi...[0m
[0;35m[C][wifi:052]:   Local聽MAC: BC:FD:0C:78:77:8C[0m
[0;35m[C][wifi:250]: Setting up AP...[0m
[0;35m[C][wifi:252]:   AP SSID: 'bk7231m-1'[0m
[0;35m[C][wifi:253]:   AP Password: ''[0m
[0;35m[C][wifi:262]:   IP Address: 192.168.4.1[0m
[0;36m[D][wifi:455]: Starting scan...[0m
[0;35m[C][web_server:123]: Setting up web server...[0m
[0;35m[C][ota:097]: Over-The-Air Updates:[0m
[0;35m[C][ota:098]:   Address: bk7231m-1.local:8892[0m
[0;35m[C][api:025]: Setting up Home Assistant API server...[0m
[0;32m[I][app:062]: setup() finished successfully![0m
[0;36m[D][sensor:093]: 'Uptime': Sending state 0.44000 s with 0 decimals of accuracy[0m
[0;36m[D][sensor:093]: 'Heap Free': Sending state 25240.00000 B with 0 decimals of accuracy[0m
[0;36m[D][sensor:093]: 'Heap Max Block': Sending state 0.00000 B with 0 decimals of accuracy[0m
[0;32m[I][app:102]: ESPHome version 2023.12.5 compiled on Mar 12 2024, 01:59:52[0m
[0;35m[C][logger:443]: Logger:[0m
[0;35m[C][logger:444]:   Level: DEBUG[0m
[0;35m[C][logger:445]:   Log Baud Rate: 115200[0m
[0;35m[C][logger:447]:   Hardware UART: UART2[0m
[0;35m[C][status_led:019]: Status LED:[0m
[0;35m[C][status_led:020]:   Pin: 11[0m
[0;35m[C][gpio.output:010]: GPIO Binary Output:[0m
[0;35m[C][gpio.output:011]:   Pin: 10[0m
[0;35m[C][switch.gpio:068]: GPIO Switch 'Relay 1'[0m
[0;35m[C][switch.gpio:090]:   Restore Mode: always OFF[0m
[0;35m[C][switch.gpio:031]:   Pin: 8[0m
[0;35m[C][uptime.sensor:031]: Uptime Sensor 'Uptime'[0m
[0;35m[C][uptime.sensor:031]:   Device Class: 'duration'[0m
[0;35m[C][uptime.sensor:031]:   State Class: 'total_increasing'[0m
[0;35m[C][uptime.sensor:031]:   Unit of Measurement: 's'[0m
[0;35m[C][uptime.sensor:031]:   Accuracy Decimals: 0[0m
[0;35m[C][uptime.sensor:031]:   Icon: 'mdi:timer-outline'[0m
[0;35m[C][gpio.binary_sensor:015]: GPIO Binary Sensor 'binary_switch'[0m
[0;35m[C][gpio.binary_sensor:016]:   Pin: 24[0m
[0;35m[C][light:103]: Light 'id_light_relay'[0m
[0;35m[C][restart:068]: Restart Switch 'Restart Switch'[0m
[0;35m[C][restart:070]:   Icon: 'mdi:restart'[0m
[0;35m[C][restart:090]:   Restore Mode: always OFF[0m
[0;35m[C][hlw8012:037]: HLW8012:[0m
[0;35m[C][hlw8012:038]:   SEL Pin: 26[0m
[0;35m[C][hlw8012:039]:   CF Pin: 6[0m
[0;35m[C][hlw8012:040]:   CF1 Pin: 7[0m
[0;35m[C][hlw8012:041]:   Change measurement mode every 8[0m
[0;35m[C][hlw8012:042]:   Current resistor: 1.0 m鈩?[0m
[0;35m[C][hlw8012:043]:   Voltage Divider: 800.0[0m
[... 192 lines cut ...]
[0;35m[C][debug:070]:     Accuracy Decimals: 0[0m
[0;35m[C][debug:070]:     Icon: 'mdi:counter'[0m
[0;36m[D][debug:076]: ESPHome version 2023.12.5[0m
[0;36m[D][debug:080]: Free Heap Size: 16464 bytes[0m
[0;36m[D][debug:356]: LibreTiny Version: 1.4.1[0m
[0;36m[D][debug:357]: Chip: BK7231N (7b1c) @ 120 MHz[0m
[0;36m[D][debug:358]: Chip ID: 0x8C7778[0m
[0;36m[D][debug:359]: Board: generic-bk7231n-qfn32-tuya[0m
[0;36m[D][debug:360]: Flash: 2048 KiB / RAM: 256 KiB[0m
[0;36m[D][debug:361]: Reset Reason: WDT Reset[0m
[0;36m[D][text_sensor:064]: 'Device Info': Sending state '2023.12.5|Version: v1.4.1 on generic-bk7231n-qfn32-tuya, compiled at Mar 12 2024 01:58:53, GCC 10.3.1 (-O1)|Reset Reason: WDT Reset|Chip Name: BK7231N|Chip ID: 0x008c7778|Flash: 2048 KiB|RAM: 256 KiB'[0m
[0;32m[I][captive_portal:031]: Captive Portal Requested WiFi Settings Change:[0m
[0;32m[I][captive_portal:032]:   SSID='Linksys03265'[0m
[0;32m[I][captive_portal:033]:   Password=[5m'19881025'[6m[0m
[0;36m[D][lt.preferences:104]: Saving 1 preferences to flash...[0m
[0;36m[D][lt.preferences:132]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed[0m
[0;36m[D][wifi:455]: Starting scan...[0m
[1;33mW [     41.462] [0mWIFI: Scan timeout, discarding
[1;31m[E][wifi:463]: Scan timeout![0m
[0;36m[D][wifi:455]: Starting scan...[0m
[1;33mW [     76.490] [0mWIFI: Scan timeout, discarding
[0;33m[W][component:214]: Component wifi took a long time for an operation (5.00 s).[0m
[0;33m[W][component:215]: Components should block for at most 20-30ms.[0m

```

kuba2k2, 2024-04-07:
@shzlww are you able to test this BK7231M with the latest version of ltchiptool? In particular, I'm curious if the flashing succeeds now, as well as the "Chip ID" from "Read chip info". I'd appreciate a screenshot of the chip info dialog.

shzlww, 2024-04-20:
@kuba2k2 
1.screenshot of the chip info.
[image]

2.The stock firmware plug ,Flash uf2 success,no CRC error.Here is the yaml
```yaml
esphome:
  name: bk7231m-1
  name_add_mac_suffix: false
  platformio_options:
    board_build.bkcrypt_coeffs: "00000000000000000000000000000000"
    board_build.bkota.key: "0123456789ABCDEF0123456789ABCDEF"
    board_build.bkota.iv: "0123456789ABCDEF"
    board_flash.download: "0x132000+0xA6000"
    board_flash.kvs: "0x1D8000+0x8000"
    board_flash.calibration: "0x1E0000+0x1000"
    board_flash.tlv: "0x1E1000+0x1000"
    board_flash.net: "0x1E2000+0x1000"
    board_flash.userdata: "0x1E3000+0x1D000"

bk72xx:
  board: generic-bk7231n-qfn32-tuya

logger:

web_server:

captive_portal:

mdns:

api:
  reboot_timeout: 0s

ota:

wifi:
  reboot_timeout: 0s
  ap:
```
3.It can launch ap hotspot ,but no wifi results found in 192.168.4.1

78757838, 2024-05-05:
有没有后续了？