libretiny-eu/libretiny #264: PlatformIO is Broken with "KeyError: \"Invalid board option 'build.family'\""

cap9qd, 2024-03-12:
I have been testing the i2c (Wire) implementation for BK72XX chips (so far so good BTW just jamming it into ESPHome i2c component) and wanted to see if I could get SoftWire working on a Temperature-Humidity board before diving into ESPHome. I found that trying to develop in PlatformIO under VSCode is broken if I leave libretiny in the .platformio directory. I thought this was something to do with my fork and developing for ESPHome but does not seem to be the case. If I remove libretiny from C:\User\user\\.platformio\platforms\ then PIO goes back to working.

```
PS C:\Users\user> cd '.\Documents\Projects\BK7233N_SoftWire\'              
PS C:\Users\user\Documents\Projects\BK7233N_SoftWire> pio project init --board generic-bk7231n-qfn32-tuya
KeyError: Traceback (most recent call last):
  File "C:\Users\user\.platformio\penv\lib\site-packages\platformio\__main__.py", line 103, in main       
    cli()  # pylint: disable=no-value-for-parameter
  File "C:\Users\user\.platformio\penv\lib\site-packages\click\core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
  File "C:\Users\user\.platformio\penv\lib\site-packages\click\core.py", line 1055, in main
    rv = self.invoke(ctx)
  File "C:\Users\user\.platformio\penv\lib\site-packages\platformio\cli.py", line 85, in invoke
    return super().invoke(ctx)
  File "C:\Users\user\.platformio\penv\lib\site-packages\click\core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "C:\Users\user\.platformio\penv\lib\site-packages\click\core.py", line 1655, in invoke
    sub_ctx = cmd.make_context(cmd_name, args, parent=ctx)
  File "C:\Users\user\.platformio\penv\lib\site-packages\click\core.py", line 920, in make_context        
    self.parse_args(ctx, args)
  File "C:\Users\user\.platformio\penv\lib\site-packages\click\core.py", line 1378, in parse_args
    value, args = param.handle_parse_result(ctx, opts, args)
  File "C:\Users\user\.platformio\penv\lib\site-packages\click\core.py", line 2360, in handle_parse_result
    value = self.process_value(ctx, value)
  File "C:\Users\user\.platformio\penv\lib\site-packages\click\core.py", line 2322, in process_value
    value = self.callback(ctx, self, value)
  File "C:\Users\user\.platformio\penv\lib\site-packages\platformio\project\commands\init.py", line 39, in validate_boards
    pm.board_config(id_)
  File "C:\Users\user\.platformio\penv\lib\site-packages\platformio\package\manager\platform.py", line 136, in board_config
    for manifest in self.get_installed_boards():
  File "C:\Users\user\.platformio\penv\lib\site-packages\platformio\util.py", line 56, in wrapper
    self.cache[key] = (time.time(), func(*args, **kwargs))
  File "C:\Users\user\.platformio\penv\lib\site-packages\platformio\package\manager\platform.py", line 112, in get_installed_boards
    for config in p.get_boards().values():
  File "C:\Users\user\.platformio\platforms\libretiny\platform.py", line 235, in get_boards
    result[key] = self.update_board(value)
  File "C:\Users\user\.platformio\platforms\libretiny\platform.py", line 250, in update_board
    family = board.get("build.family")
  File "C:\Users\user\.platformio\penv\lib\site-packages\platformio\platform\board.py", line 47, in get
    raise KeyError("Invalid board option '%s'" % path)
KeyError: "Invalid board option 'build.family'"

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `python -m pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq/index.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================
```

I noticed you pushed some changes to https://registry.platformio.org/platforms/kuba2k2/libretiny/boards recently so maybe I am just jumping into the middle of something that's a work in progress.

I tried updating PIO but my PIO version is latest at v6.1.14b1.

I also tried installing libretiny from PIO CLI command line with the same result.

With the microcontroller stuff I am half useful...but here I am next to useless tracking this down.

cap9qd, 2024-03-12:
I found the issue by adding some `print(...)` statements in `platform.py`; I have an old custom esp32 board defined from before the esp32-s3 was included in the esp32 platform...which doesn't have the build.family in the JSON.

kuba2k2, 2024-03-12:
Oh. I hate that PIO seems to call platform.py files with OTHER platforms' boards... this is a mess. Anyway, LT should be able to handle such a case (e.g. filter out/ignore all non-LibreTiny boards), so I'll reopen this as a bug. Thanks for finding the issue!