libretiny-eu/libretiny #266: Wifi chip from Smart Plug reports as Rtl8710c How to flash or dump it?

raufaser, 2024-03-15:
I desoldered the Wifi module from a Gosund EP2.
I managed to hook it up and It talks to me via UART.
But i have no idea how to put it into download mode and what kind of firmware i need.

This is the comm PCB: Front with W701 chip
[image]

And back:
[image]

I hooked it up like this:
[image]

And got this:
```
== Rtl8710c IoT Platform ==
Chip VID: 5, Ver: 3
ROM Version: v3.0

== Boot Loader ==
Dec  5 2019:14:02:18

fwx SELE[fffffffc]
fw SELE Bitidx 2, fw1 valid 1, sn 0, fw2 valid 1, sn 100
fw1 USE, return sn 0

Boot Loader <==

== RAM Start ==

Build @ 15:38:41, Jan 29 2023

interface 0 is initialized

interface 1 is initialized
...
...
See the txt for the whole capture
```

[capture.txt](https://github.com/libretiny-eu/libretiny/files/14621539/capture.txt)

Can someone help?

raufaser, 2024-03-16:
I traced the chip pins to the outer pins. This is what i got:
[image]
I guessed a little and it seems to be the same as this one:
[image]

So i pulled A0 and CEN to 3,3V. Now i get this message:

```
== Rtl8710c IoT Platform ==
Chip VID: 5, Ver: 3
ROM Version: v3.0
Test Mode: boot_cfg1=0x20
Download Image over UART2[tx=16,rx=15] baud=115200
```

I am already communication via pins 16 and 15. Do you have any idea how to dump the firmware? Or what firmware i could flash?

kuba2k2, 2024-03-16:
Try ltchiptool, it can dump and flash firmware of the chip. A ROM dump could be useful if you can share it.

As for firmware, LibreTiny has preliminary support for RTL8720C. Things like OTA are not implemented yet.

raufaser, 2024-03-16:
I am a little bit confused, because my device reports as Rtl87**1**0c IoT Platform == ... is it different from RTL87**2**0C ?

No success yet with dumping.

kuba2k2, 2024-03-16:
No, it's fundamentally the same thing. 8710 is the generic family name, also 8710 is a version of 8720 without Bluetooth. What matters is the "C" after the number.