libretiny-eu/libretiny #268: BK7231N : The compiled Firmware file size exceeds the flash memory

TuanAIoT, 2024-03-22:
Flash memory is 2MB, however when I compile the program it exceeds 2048kb,
despite only using 66% of memory (used 715616 bytes from 1083136 bytes)
[image]
[image]

Cossid, 2024-03-22:
That is completely normal, uf2 is a wrapping format with extra information about partitions, writing instructions, and validations.  The file is streamed and processed, and only the appropriate bits will be written as determined by the information streamed and processed.  You must use LTChipTool (or the OTA process which also implements uf2 depending on platform), and it will be handled correctly.

closed 2024-03-22 by Cossid

TuanAIoT, 2024-03-22:
@Cossid ,
Exceeding the size of 2048KB causes an error when writing the firmware.bin file to the BK7231N chip
[image]

kuba2k2, 2024-03-22:
Firmware.bin is exactly the same UF2 file. It's named this way so that people used to ESP platforms choose this file for flashing.

As Cossid said, it only works with LibreTiny flashing tools, which understand this file format. OpenBeken's flasher tool only understands raw binaries, so it will not be able to write UF2 files.

TuanAIoT, 2024-03-22:
yes, thank you everyone, The problem has been resolved