libretiny-eu/libretiny #270: instructions contradictory in ltchiptool?

helgek, 2024-04-01:
Hi, 

when I tried to connect to the RTL8710B (so far unsuccessfully but I will try again with a better setup soon) these are the instructions shown in ltchiptool 4.11.1 (dev):
```

I: Found new device: ttyS4 - n/a
I: Found new device: ttyUSB0 - USB2.0-Ser! - None (1A86/7523)
I: Connect UART2 of the Realtek chip to the USB-TTL adapter:
I: 
I:     --------+        +---------------------
I:          PC |        | RTL8710B            
I:     --------+        +---------------------
I:          RX | ------ | TX2 (Log_TX / PA30) 
I:          TX | ------ | RX2 (Log_RX / PA29) 
I:             |        |                     
I:         GND | ------ | GND                 
I:     --------+        +---------------------
I:  
I: Using a good, stable 3.3V power supply is crucial. Most flashing issues
I: are caused by either voltage drops during intensive flash operations,
I: or bad/loose wires.
I:  
I: The UART adapter's 3.3V power regulator is usually not enough. Instead,
I: a regulated bench power supply, or a linear 1117-type regulator is recommended.
I:  
I: In order to flash the chip, you need to enable download mode.
**I: This is done by pulling CEN to GND briefly, while still keeping the TX2 pin
I: connected to GND.**
I:  
I: Do this, in order:
I:  - connect CEN to GND
I:  - connect TX2 to GND
I:  - release CEN from GND
I:  - release TX2 from GND

```
The part that I highlighted in bold letters is not mentioned at https://docs.libretiny.eu/docs/platform/realtek-ambz/#wiring and it also seems to not work together with the instructions in the paragraph following. Was this part of the instruction maybe inserted by mistake / not removed by mistake?

Thanks,
Helge

kuba2k2, 2024-04-01:
The docs say:
> This is done by resetting the chip while pulling down the TX2 pin.

ltchiptool says:
> This is done by pulling CEN to GND briefly, while still keeping the TX2 pin connected to GND.

These two statements are equivalent:
- Pulling down = connecting to GND
- Resetting = pulling CEN to GND.

as explained in the docs:
>"CEN" pin is the RESET pin - connecting it to GND will keep the chip in "reset" state.

---
Unfortunately I don't have any ideas for better wording of the whole CEN/reset thing. People seem to not understand what that means, but I don't know a better way to phrase it.

kuba2k2, 2024-04-01:
Indeed, the order is different than in the detailed sentence (the 1st and 2nd steps). But that doesn't matter - as long as CEN is released before TX2, it will work the same way.

helgek, 2024-04-01:
That's right. The only thing that doesn't work in this order is to pull CEN to GND briefly. I don't know if this is relevant for this chipset but from your explanation I assume it's not. I think I saw some discussion before (don't remember which chipset it was about) where the duration of the CEN to GND was discussed, that's why I had assumed that "briefly" in your instruction might be of relevance.

helgek, 2024-04-01:
Thank you, I was now able to read the flash memory with an improved setup using a FT232RL based UART device. A bench power supply wasn't needed in my case to read the WR3.

closed 2024-04-01 by helgek