libretiny-eu/libretiny #271: Further clarification to what 1.5M baud rate refers to

helgek, 2024-04-01:
Hi,

at https://docs.libretiny.eu/docs/platform/realtek-ambz/#flashing it shows the following information:

**Tip

You need a good USB<->UART adapter for the process. Some chips may not support 1.5M baud rate, required by the ROM for the initial handshake. Widespread PL2303 is currently known not to work, at least under Windows. FT232RL is verified to work reliably.**

In ltchiptool 1.5M baud rate is not available. Possibly I don't understand something fundamental regarding the technical background of this information, nevertheless I wanted to suggest to maybe add some additional information for people like me without the technical background going this deep. 

Thanks,
Helge

kuba2k2, 2024-04-01:
1.5M baud rate is 1,500,000 baud. It isn't available in ltchiptool, because the available baudrates only apply for the actual flashing process. The 1.5M baud rate is only used at the start of the process:

> required by the ROM for the **initial handshake**.

helgek, 2024-04-01:
Thank you, yes this is what I had also assumed, I just wasn't sure about it because I'm not familiar with the technical depth of the connection initiation process. Although you provide an explanation it would maybe help others to add an additional sentence for half-knowledge people like me :D

closed 2024-04-01 by helgek