libretiny-eu/libretiny #272: Tuya device firmware read from WR3 can't be analyzed in UPK2ESPHome

helgek, 2024-04-01:
Hi,

the firmware file for this humidifier device https://eurom.nl/en/product/climate-control/humidifiers/oasis-303-wifi/ is not readable in UPK2ESPHome. I never booted the device before reading the flash memory, so the file attached is original without any personal settings.

When I open the file in text editor I can see Tuya references and generally there's a lot of of readable text.
[Luftbefeuchter ebay ltchiptool_ambz_2024-04-02_00-45-27.zip](https://github.com/libretiny-eu/libretiny/files/14829553/Luftbefeuchter.ebay.ltchiptool_ambz_2024-04-02_00-45-27.zip)

Thanks,
Helge

Cossid, 2024-04-02:
Realtek isn't officially supported for UPK2ESPHome, only Beken is confirmed/designed.  When it comes to Realtek, it can only work when a couple conditions are met.  The first being the SDK is new enough and uses the same encrypted storage schema as the Beken SDKs, and the second being the device has a configuration and isn't an MCU or device-specific firmware.

In this case, the firmware type is `rtlbn_tls_common_9600` which indicates this is likely a TuyaMCU device that communicates at baud 9600, and there would be no configuration in storage, just RX/TX connected to what LibreTiny would have mapped as RX0 and TX0

helgek, 2024-04-02:
Thank you. I'm new to this and trying to understand how I should proceed to eventually be able to flash a working ESPHome config.

From reading through the different docs of libretiny, libchiptool, etc. it seems like a way forward could be to:

- use ESPHome-Kickstart to analyze button clicks and other data being coming through the internal serial interface (it's a humidifier with buttons, display, humidity sensor, water level sensor)

- retrieve tuya DP's ( https://community.home-assistant.io/t/discovering-localtuya-keys-and-dps/588634 ) but I'm not sure if this information would help to write an ESPHome config

Really appreciate any guidance how to continue. Thank you.

Cossid, 2024-04-02:
Usually when there is a TuyaMCU, all of the IO is connected to the TuyaMCU, not the wifi module.  Kickstart doesn't really help a lot for that since you know the pins the TuyaMCU would be connected to.  You'd need to install regular ESPHome with a `tuya:` configuration, and parse the boot message to see what DPIDs to map with other tuya sensors/components.

Sometimes there us a TuyaMCU + IO on the wifi, but that is quite rare.