libretiny-eu/libretiny #273: Problems when configuring wifi for devices using the BK7231N chip and connecting the device to the App

TuanAIoT, 2024-04-02:
Chip : BK7231N
Module : CB3S, CBU
I know that ESP8266 and ESP32 have the function of configuring wifi information through "ESP Touch" APP.
So is there a way for the App to directly configure wifi for the device, then the App will save the device's MAC Address information?

I want to simplify the work for users of products using the current BK7231N chip. Just like Tuya products are doing, the devices will automatically be paired with the "Tuya Smart" APP. Users just need to put the device and App into pairing mode and they will automatically pair with each other.

kuba2k2, 2024-04-02:
LibreTiny doesn't have any Tuya Cloud related code. I don't know why would you want to use the Tuya Smart app with LibreTiny (you can just use the stock firmware instead). It's not possible to use the Tuya Smart app with devices that are not connected to the Tuya cloud.

If you want to mimic the Tuya app configuration procedure, you can probably create an AP manually and connect to that AP using an app on the smartphone. There's no built-in way to do that, because it misses the point of the project. You will have to write the AP logic and the app manually.

However, if you're developing a device that you want to manufacture or sell, keep in mind that LibreTiny might not be production-ready. Also make sure you read and understood the license agreement of the project and related libraries.

closed 2024-04-12 by TuanAIoT

TuanAIoT, 2024-04-12:
https://docs.bekencorp.com/armino/bk7256/zh_CN/latest/examples/bluetooth/ble_boarding_demo.html