libretiny-eu/libretiny #279: unable to write to BK7231M on CB2S module

gdtv, 2024-04-18:
I have two CB2S modules, the left one has a BK7231M, the right one has a metal cover so I don't know what's inside.
The right one works fine, I can write the firmware with ltchiptool-v4.11.1 and it runs correctly.
But the left one cannot write the firmware. The following is a screenshot of the error.

[image]

[image]

[image]

```
W: Erasing failure @ 0x11000 (Erase failed - flash protected; found non-0xFF bytes @ 0x11000), retrying (attempt 0)
W: Erasing failure @ 0x11000 (Erase failed - flash protected; found non-0xFF bytes @ 0x11000), retrying (attempt 1)
W: Erasing failure @ 0x11000 (Erase failed - flash protected; found non-0xFF bytes @ 0x11000), retrying (attempt 2)
W: Erasing failure @ 0x11000 (Erase failed - flash protected; found non-0xFF bytes @ 0x11000), retrying (attempt 3)
E: An error has occurred
E: ValueError: Erase failed - flash protected; found non-0xFF bytes @ 0x11000
E: |-- File "ltchiptool\gui\work\base.py", line 24, in run
E: |-- File "ltchiptool\gui\work\flash.py", line 73, in run_impl
E: |-- File "ltchiptool\gui\work\flash.py", line 126, in _do_write
E: |-- File "ltchiptool\soc\bk72xx\flash.py", line 314, in flash_write_uf2
E: |-- File "ltchiptool\util\streams.py", line 177, in update_from
E: |-- File "bk7231tools\serial\cmd_hl_flash.py", line 185, in program_flash
E: |-- File "bk7231tools\serial\cmd_ll_flash.py", line 229, in flash_erase_block
E: |-- File "bk7231tools\serial\cmd_ll_flash.py", line 209, in do_erase_verify
```

kuba2k2, 2024-04-18:
Please post full flashing logs in verbose mode.

gdtv, 2024-04-18:
> Please post full flashing logs in verbose mode.

this is logs
[error-log.txt](https://github.com/libretiny-eu/libretiny/files/15025209/error-log.txt)

ntzb, 2024-05-20:
exact same issue for me, but the path was different.
my cb2s was flashed with the generic `kickstart-bk7231n-2023-12-15.uf2` firmware, and after that, it does not show up as AP, and does not accept any more flashing, with the same error as above:
```
W: Erasing failure @ 0x11000 (Erase failed - flash protected; found non-0xFF bytes @ 0x11000), retrying (attempt 0)
W: Erasing failure @ 0x11000 (Erase failed - flash protected; found non-0xFF bytes @ 0x11000), retrying (attempt 1)
W: Erasing failure @ 0x11000 (Erase failed - flash protected; found non-0xFF bytes @ 0x11000), retrying (attempt 2)
W: Erasing failure @ 0x11000 (Erase failed - flash protected; found non-0xFF bytes @ 0x11000), retrying (attempt 3)
```

tried to power off, power on, reconnect rx,tx a few times, and finally it was working again, and could flash the compiled firmware

kuba2k2, 2024-05-20:
Do you also have a BK7231M? CB2S modules usually come with a BK7231N, so that's likely where the issue comes from.

ntzb, 2024-05-20:
oh didn't see the N vs M thing.
mine was N, so it's you're probably right.

gdtv, 2024-05-21:
Solution: 
https://bbs.hassbian.com/forum.php?mod=viewthread&tid=25311
Sorry for my poor English, I can't translate it into English

kuba2k2, 2024-05-21:
That could be a solution for BK7231M. BK7231Flasher program uses a different algorithm for unprotecting the flash chip, so it might have better success on other bootloaders.

The config shown in the forum post works only for this particular device - it uses unencrypted firmware. It is important to first specify the encryption parameters, then compile the firmware.

The OpenBK "burn" step is mostly unnecessary, since it overwrites the bootloader with one that won't work on that device. However, I don't think there is another way to unprotect the flash using BK7231Flasher - the tool is only written to flash OpenBK.