libretiny-eu/libretiny #281: Feature Request : Maintain constant state through reboot

BenJamesAndo, 2024-04-29:
It would be nice when doing OTA updates in ESPHome for Beken chips for the GPIOs not to reset. This is undesirable if you have something like a server or router plugged in and your plug briefly turns off.

From [ESPHome page](https://esphome.io/components/esp8266.html#configuration-variables)
`early_pin_init (Optional, boolean): Specifies whether pins should be initialised as early as possible to known values. Recommended value is false where switches are involved, as these will toggle when updating the firmware or when restarting the device. Defaults to true.`

Cossid, 2024-04-29:
This might be a hardware limitation, as the state is lost before ESPHome can init it.  This would probably need something similar to ESP32's `gpio_hold_en`/`gpio_hold_dis` which the BDK doesn't appear to have.

This is difficult to verify how Tuya handled it, as firmware upgrades are so rare on their platform.

Kaldek, 2024-08-14:
> this platform (Beken BK7231) is an APP/OTA style, which is different than the usual APP/APP, which means part of the boot process is applying the OTA which is not handled by an OS, but rather the bootloader

Somewhat off topic here, but could this be related to the reason why sometimes the OTA image flashes successfuly but doesn't update the App partition?  I've had a bunch of WB2S modules lately not apply the OTA'd firmware reliably.  A forum post over at Elektroda suggested grounding TX1 while applying power to the device (more specifically they said hold the external button which is connected to TX1 down whilst removing power, and keeping it held while applying power).  Seems to work reliably but I haven't found any material yet that explains the correct method to force apply the OTA.