libretiny-eu/libretiny #283: AssertionError: Error: Missing site-packages directory `/home/$USER/.platformio/penv/.libretiny/lib/python3.11/site-packages`:

GijsTimmers, 2024-05-25:
I might very well be missing something. After installing esphome in a virtual environment with this guide: https://esphome.io/guides/installing_esphome.html#linux and trying to compile a `.yaml` file I get:

```
AssertionError: Error: Missing site-packages directory `/home/$USER/.platformio/penv/.libretiny/lib/python3.11/site-packages`:
  File "/home/$USER/Scripts/esphome/iot/lib/python3.11/site-packages/platformio/builder/main.py", line 173:
    env.SConscript("$BUILD_SCRIPT")
  File "/home/$USER/.platformio/packages/tool-scons/scons-local-4.7.0/SCons/Script/SConscript.py", line 612:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/home/$USER/.platformio/packages/tool-scons/scons-local-4.7.0/SCons/Script/SConscript.py", line 279:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "/home/$USER/.platformio/platforms/libretiny/builder/main.py", line 24:
    env.ConfigurePythonVenv()
  File "/home/$USER/.platformio/packages/tool-scons/scons-local-4.7.0/SCons/Util/envs.py", line 251:
    return self.method(*nargs, **kwargs)
  File "/home/$USER/.platformio/platforms/libretiny/builder/python-venv.py", line 73:
    assert (
```

.yaml snippet:

```
bk72xx:
  board: generic-bk7231t-qfn32-tuya
```

kuba2k2, 2024-05-26:
Do you have venv installed? Try "python -m venv" to see if it works. If not, you'll need to install it using apt or something.

GijsTimmers, 2024-06-07:
Yes, I have that one installed:
```
$ python3 -m venv    
usage: venv [-h] [--system-site-packages] [--symlinks | --copies] [--clear] [--upgrade] [--without-pip] [--prompt PROMPT]
            [--upgrade-deps]
            ENV_DIR [ENV_DIR ...]
venv: error: the following arguments are required: ENV_DIR
```

GijsTimmers, 2024-06-07:
I found the culprit. Something had gone wrong installing esphome or platformio, probably by doing a system wide installation instead of a local one in the virtual environment. Removing `~/.local/bin/esphome` and `~/.platformio` did the trick. Thanks for the help anyway. Closing now.

closed 2024-06-07 by GijsTimmers