libretiny-eu/libretiny #285: Incorrect input format ltchiptool

Mit4el, 2024-06-01:
Hi, can you help with the error when linking to platformio
At the same time, the uf2 files do not appear.
What could be the problem?
[image]
I need "Full installation" ltchiptool?
the command outputs the following:
PS ....> python -m ltchiptool uf2 write -h
.....\AppData\Local\Microsoft\WindowsApps\PythonSoftwareFoundation.Python.3.11_qbz5n2kfra8p0\python.exe: No module named ltchiptool

I am trying to redo a large project, here is my platformio.ini
https://github.com/Mit4el/IoTManager/blob/52fb21ee99337c83807bed8d96a9bc87a1d4b9bf/platformio.ini#L380

kuba2k2, 2024-06-01:
You don't need to have a "full" installation of ltchiptool, LibreTiny does it automatically. It has its own isolated environment.

You're using Python from Microsoft Store, which is very discouraged and could be an issue. It would be best if you uninstalled Python from Microsoft Store and installed it the classic way.

Can you run the PlatformIO command in verbose mode? It's either `PIOVERBOSE=1` environment variable, or `-v`/`--verbose` command line switch - I don't remember, consult the PIO documentation. It should at least print what command it runs to get that error.

Mit4el, 2024-06-01:
env_uf2ota([".pio\build\tiny\firmware.uf2"], [".pio\build\tiny\raw_firmware.elf"])
[image]

kuba2k2, 2024-06-01:
That unfortunately doesn't help. Please edit
`~/.platformio/platforms/libretiny/builder/utils/ltchiptool-util.py`
and add
`print(cmd)`
on line 58, right before
`env.Execute(" ".join(cmd))`.

Mit4el, 2024-06-01:
['@${LTCHIPTOOL} uf2 write', '--output "${BUILD_DIR}\\iotm_tiny_0.0.1_${VARIANT}_${MCULC}_lt1.5.1.uf2"', '--output "${BUILD_DIR}\\firmware.uf2"', '--output "${BUILD_DIR}\\firmware.bin"', '--board ${BOARD_JSON}', '--lt-version 1.5.1', '--fw "iotm_tiny:0.0.1"', '--date 1717260523', '--legacy', '"${BUILD_DIR}/image_${MCULC}_app.${FLASH_APP_OFFSET}.crc=flasher:app"', '"${BUILD_DIR}/image_${MCULC}_app.${FLASH_RBL_OFFSET}.rblh+1150730=flasher:app"', '"${BUILD_DIR}/image_${MCULC}_app.ota.rbl=device:download"']
[image]

kuba2k2, 2024-06-01:
What's your full project directory? You might be having issues if you have spaces in the project directory path.

If you don't have spaces, can you try without the custom_fw_name? It's rarely used, maybe it's the one causing issues.

Mit4el, 2024-06-01:
c:\USER_BMW\IoTManager_4dev\IoTManager+C6+Tiny\
There were + , removed it worked. Thanks

Mit4el, 2024-06-01:
By the way, HttpClient swears at strptime in the file
If you add extern "C" to the lt_posix_api.h file, it will get better))

Mit4el, 2024-06-01:
Everything is fine build and upload.
But UART2 is silent and does not output anything, the firmware does not start wifi AP (((
UART in libertine works, does not require additional define in platformio.ini?

kuba2k2, 2024-06-01:
The lines in your config:
```ini
	-DLT_LOGLEVEL=LT_LEVEL_NONE
	-DLT_UART_SILENT_ALL=1
```
pretty much disable all possible UART output.

closed 2024-06-07 by Mit4el