libretiny-eu/libretiny #289: Configure OTA on wblc5 (BK7231T)

raulzanardo, 2024-07-16:
How do configure the platformio.ini file to use the OTA upload protocol?

Cossid, 2024-07-16:
You can use the `generic-bk7231t-qfn32-tuya` board for the WBLC5 board, https://docs.libretiny.eu/boards/generic-bk7231t-qfn32-tuya/

raulzanardo, 2024-07-16:
Ok, but how do configure the upload protocol for it to be OTA?

I cant find any documentation about the upload_protocol of the platformio.ini file

Cossid, 2024-07-16:
I don't know that OTA in particular is a part of platformio, as it is more an implementation detail, not low-level.

platformio is expecting UART; https://docs.libretiny.eu/docs/flashing/platformio/#using-ltchiptool-wired-via-uart

I'll defer to @kuba2k2 to expand further.

kuba2k2, 2024-07-16:
If by "OTA" you're referring to the wireless update over Wi-Fi, then no, PlatformIO doesn't have that built-in.
In fact, to best of my knowledge, PlatformIO doesn't have that feature on any supported platform.

If you want to upload your firmware via Wi-Fi, you will need to use the `Update` Arduino library. The library is very similar to ESP32's equivalent, it's meant to be a drop-in replacement.
See docs:
- https://docs.libretiny.eu/ltapi/class_update_class/
- https://docs.libretiny.eu/ltapi/class_libre_tiny_o_t_a/
- https://docs.libretiny.eu/docs/dev/lt-api/#ota
There are many tutorials online about the `Update` library, you will definitely also find a way to e.g. pull firmware updates via HTTP.

The `upload_protocol` is not used and shouldn't be changed - it defaults to `uart`, no other protocol is currently supported.

raulzanardo, 2026-01-14:
I figure it out!

Now it works the same way as it would work with an ESP32 or ESP8266. 

I made a repo exameple of how it works. 

https://github.com/raulzanardo/libretiny-pio-ota

closed 2026-01-14 by raulzanardo