libretiny-eu/libretiny #293: BW 15 Bluetooth

NathanKuo0821, 2024-09-05:
Dear Sir,

I'm currently developing with the BW15, but I'm unable to use the standard Arduino BLE library. How should I proceed?

Nathan Kuo

kuba2k2, 2024-09-05:
RTL8720CF is not fully supported yet. Furthermore, there is no BLE support in LibreTiny at all.

See #44.

vinhpn96, 2025-07-28:
> RTL8720CF is not fully supported yet. Furthermore, there is no BLE support in LibreTiny at all.
> 
> See [#44](https://github.com/libretiny-eu/libretiny/issues/44).

Hi @kuba2k2 
there is no BLE support in LibreTiny at all.
it means that LibreTiny can not support BLE or LibreTiny have not support yet?
if it can not support, could you please explain why?
let me know because I have to implement BLE and if not, I have to use another platform

Thanks