libretiny-eu/libretiny #296: PlatformIO registry board info links are broken

rob040, 2024-09-25:
at https://registry.platformio.org/platforms/kuba2k2/libretiny/boards
all board info links result in 404 page not found

It might be related to issue #264, but that one is half a year old.

kuba2k2, 2024-09-25:
That is because PlatformIO registry assume that a docs page exists in PIO docs. LibreTiny has its own docs pages:
https://docs.libretiny.eu/
This is not changeable in PlatformIO.

Psi-nz, 2024-10-12:
Anyone have a solution?

Currently if you install libretiny in PlatformIO (at least on windows), it instantly breaks the board list from loading and throws errors every time you click on various pages in platformio.  It breaks the board list from loading for every platform you have installed, not just libretiny boards.  **I'm not blaming libretiny for this**.  Just saying its a pretty big issue that needs a solution because currently libretiny cannot be used at all in PlatformIO.  And once it's installed you can't even uninstall libretiny from the GUI because it throws board list errors. You have to actually go delete the libretiny folder manually to get PlatformIO working again.

kuba2k2, 2024-10-12:
That issue is a different one than the OP had.

I've had such a problem in the past, where it would display a KeyError message. IIRC, what I've found is that it was a problem with either the ESP32 or esp8266 platform, which were scanning LibreTiny boards and found them incompatible. Try uninstalling one of them.

However, PlatformIO can be used without the board list. You can still create a project, either via PlatformIO Home GUI or the command line. Compiling a project also works perfectly fine - I have both LibreTiny and ESP platforms installed (as well as several others) and I have no problems currently.