libretiny-eu/libretiny #301: bk7231n ADC on PIN9 not available in ESPHome 2024.11.2

majuss, 2024-12-03:
```
esphome:
  name: trafficlight1
  friendly_name: trafficlight1

bk72xx:
  board: generic-bk7231n-qfn32-tuya

wifi:

logger:

api:

ota:
  platform: esphome

web_server:

captive_portal:

output:
  - platform: libretiny_pwm
    id: red
    frequency: 1000 Hz
    pin: PWM5
  - platform: libretiny_pwm
    id: blue
    frequency: 1000 Hz
    pin: PWM0
  - platform: libretiny_pwm
    id: green
    frequency: 1000 Hz
    pin: PWM4
  - platform: libretiny_pwm
    id: cw
    frequency: 1000 Hz
    pin: PWM2
  - platform: libretiny_pwm
    id: ww
    frequency: 1000 Hz
    pin: PWM1

light:
  - platform: rgb
    name: "Living Room Lights"
    red: red
    green: green
    blue: blue

  - platform: cwww
    name: whitelight
    cold_white: cw
    warm_white: ww
    cold_white_color_temperature: 6536 K
    warm_white_color_temperature: 2000 K

binary_sensor:
  - platform: gpio
    pin: P14
    name: "PIR Sensor"
    device_class: motion
  - platform: gpio
    pin:
      number: P20
      inverted: true
    name: "Button"

sensor:
  - platform: adc
    name: sensor
    pin: P9
    filters:
      - multiply: 3.3
```
I try to use P9 on an ADC, my device has a photoresistor connected to it. The datasheet mentions that P9 is reserved for "VBAT" which also indicates an ADC pin. But still the board definition available in ESPHome thinks that P9 is not a valid ADC pin.

majuss, 2024-12-03:
Okay thank you for clarification!
I took my multimeter and measured. It's connected to P28 where the red X is.

[image]

kuba2k2, 2024-12-03:
The pin you marked is ADC3, also known as P28 (you can use any of these names in your config). It is an analog input pin.

See full pinout here:
https://docs.libretiny.eu/boards/cbu/?h=cbu#pinout

Cossid, 2024-12-03:
It's usually P23 that is ADC3, is P28 also supported?

kuba2k2, 2024-12-03:
Ah, my bad, I meant P23 as well. Corrected now.

majuss, 2024-12-03:
So I tried ADC3 but I only get 0.07 V no matter how I will shine the light on the resistor. I measured 1 M Ohm at darkness and 600k Ohm with flashlight.

Cossid, 2024-12-03:
I'm guessing you're falling victim to the default polling rate of 60 seconds for sensors.
You'd need to set `update_interval` to a lower value like `1s` to get more real-time updates.

This looks like the YD001 nightlight, and I can confirm the raw values on ADC3 vary between 0 and 636
There is also a binary sensor for light sense on P9 (unsure what its threshold is)

majuss, 2024-12-04:
That was exactly it!

closed 2024-12-04 by majuss