libretiny-eu/libretiny #314: WBR2 module with RTL8720CF chip - unable to run simple code

christescu, 2025-01-21:
Hello,
I have a WBR2 module with RTL8720CF chip, and I would like to play with it using PlatformIO and LibreTiny.
I started with a simple Blinking / Hello world program.

platformio.ini:
```
[env:generic-rtl8720cf-2mb-992k]
platform = libretiny
board = generic-rtl8720cf-2mb-992k
framework = arduino
```
main.cpp:
```
#include <Arduino.h>
pin_size_t led1 = 20;

void setup() {
  pinMode(led1, OUTPUT);
  Serial.begin(115200);
}

void loop() {
  digitalWrite(led1, HIGH);
  delay(500);
  digitalWrite(led1, LOW);
  delay(500);
  Serial.println("Hello world!");
}
```
Obviously, I have an led on GPIO 20.
I upload the code, everything fine.
But after reboot, the led is not blinking and on serial monitor I have:
```
== Rtl8710c IoT Platform ==
Chip VID: 5, Ver: 3
ROM Version: v3.0

== Boot Loader ==
Aug 30 2024:04:22:00
[MISC Err]boot_get_load_fw_idx: failed!
[MISC Err]boot_load: Get load FW index err
Boot Load Err!

```
Any idea what am I doing wrong?

As an info, on the same chip I wrote OpenRTL87X0C (from OpenBeken project) and it works fine, I can also control the led. So there is no hardware or flashing issue.

kuba2k2, 2025-01-28:
The RTL8720C chip has a partition table inside the flash. Different bootloaders expect different partition tables. OpenBeken also replaces these partitions with their own.

There is an error in the UF2 implementation in ltchiptool that sometimes prevents it from writing the bootloader/partition table properly. To get around that, try flashing these files manually:

[bw15.zip](https://github.com/user-attachments/files/18577183/bw15.zip)

This is the bootloader and partition table that LibreTiny expects. Flashing it with ltchiptool (to the correct offsets, as indicated by the filename) should make it possible to flash your own firmware.

christescu, 2025-01-29:
I confirm that after flashing the 2 files, now it works like a charm.
Thank you so much for this workaround.

closed 2025-01-29 by christescu