libretiny-eu/libretiny #315: ESPHome: fatal error: lwip/sockets.h: No such file or directory when using LibreTiny 1.7.0 or above

DougEdey, 2025-03-10:
I'm trying to configure some newly flashed devices with ESPHome, I have the kickstart firmware loaded and it's connected to my HomeAssistant instance and adopted in ESPHome

[image]

I used the web based ltchiptool: https://upk.libretiny.eu/?profile=globe-electric-35804-rgb-e26-led-bulb

Since the current Python GUI doesn't have this feature supported (at least not that I can see)

I've tried the default libretiny @1.7.0 and @1.8.0

Here's my yaml config:

```yaml
esphome:
  name: denrgbcct
  name_add_mac_suffix: true

bk72xx:
  board: generic-bk7231t-qfn32-tuya
  framework: 
    version: 1.8.0

logger:

web_server:

captive_portal:

mdns:

api:

ota:
  - platform: esphome
    id: ota_esphome

wifi:
  ap:

# text_sensor:
#   - platform: libretiny
#     version:
#       name: LibreTiny Version

sm2135:
  clock_pin: P8
  data_pin: P26
  rgb_current: 20mA
  cw_current: 55mA

output:
  - platform: sm2135
    id: output_red
    channel: 2
  - platform: sm2135
    id: output_green
    channel: 1
  - platform: sm2135
    id: output_blue
    channel: 0
  - platform: sm2135
    id: output_cold
    channel: 4
  - platform: sm2135
    id: output_warm
    channel: 3

light:
  - platform: rgbww
    id: light_rgbww
    name: Light
    color_interlock: true
    cold_white_color_temperature: 6500 K
    warm_white_color_temperature: 2700 K
    red: output_red
    green: output_green
    blue: output_blue
    cold_white: output_cold
    warm_white: output_warm
```

```
INFO ESPHome 2025.2.2
INFO Reading configuration /config/esphome/den-rgb-cct.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing denrgbcct (board: generic-bk7231t-qfn32-tuya; framework: arduino; platform: libretiny @ 1.8.0)
--------------------------------------------------------------------------------
Warning! Non-Git installations are NOT SUPPORTED.
HARDWARE: BK7231T 120MHz, 256KB RAM, 1.03MB Flash
 - framework-arduino-api @ 2022.8.24+sha.237b10a 
 - framework-beken-bdk @ 0.0.0+v2021.06.07.sha.6491b8c 
 - library-flashdb @ 1.2.0+sha.d5c892f 
 - library-freertos @ 9.0.0+sha.cea0a88 
 - library-freertos-port @ 2023.5.23+sha.a917d93 
 - library-lwip @ 2.2.0-bdk+sha.48da4a2 
 - library-printf @ 6.1.0+sha.28a79bd 
 - library-uf2ota @ 5.0.0+sha.f955412 
PLATFORM VERSIONS:
 - libretiny @ 1.8.0
 - ltchiptool @ 4.11.6
CUSTOM OPTIONS:
 - fw_name = esphome
 - fw_version = 2025.2.2
Dependency Graph
|-- AsyncTCP-esphome @ 2.1.4
|-- ESPAsyncWebServer-esphome @ 3.2.2
|-- DNSServer @ 1.1.0
|-- ArduinoJson @ 6.18.5
Compiling .pioenvs/denrgbcct/src/esphome/components/api/api_connection.cpp.o
Compiling .pioenvs/denrgbcct/src/esphome/components/api/api_frame_helper.cpp.o
Compiling .pioenvs/denrgbcct/src/esphome/components/api/api_server.cpp.o
Compiling .pioenvs/denrgbcct/src/esphome/components/api/list_entities.cpp.o
In file included from src/esphome/components/socket/socket.h:6,
                 from src/esphome/components/api/api_frame_helper.h:14,
                 from src/esphome/components/api/api_connection.h:5,
                 from src/esphome/components/api/api_connection.cpp:1:
src/esphome/components/socket/headers.h:131:10: fatal error: lwip/sockets.h: No such file or directory
  131 | #include "lwip/sockets.h"
      |          ^~~~~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs/denrgbcct/src/esphome/components/api/api_connection.cpp.o] Error 1
In file included from src/esphome/components/socket/socket.h:6,
                 from src/esphome/components/api/api_frame_helper.h:14,
                 from src/esphome/components/api/api_frame_helper.cpp:1:
src/esphome/components/socket/headers.h:131:10: fatal error: lwip/sockets.h: No such file or directory
  131 | #include "lwip/sockets.h"
      |          ^~~~~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs/denrgbcct/src/esphome/components/api/api_frame_helper.cpp.o] Error 1
In file included from src/esphome/components/socket/socket.h:6,
                 from src/esphome/components/api/api_server.h:8,
                 from src/esphome/components/api/api_server.cpp:1:
src/esphome/components/socket/headers.h:131:10: fatal error: lwip/sockets.h: No such file or directory
  131 | #include "lwip/sockets.h"
      |          ^~~~~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs/denrgbcct/src/esphome/components/api/api_server.cpp.o] Error 1
In file included from src/esphome/components/socket/socket.h:6,
                 from src/esphome/components/api/api_frame_helper.h:14,
                 from src/esphome/components/api/api_connection.h:5,
                 from src/esphome/components/api/list_entities.cpp:3:
src/esphome/components/socket/headers.h:131:10: fatal error: lwip/sockets.h: No such file or directory
  131 | #include "lwip/sockets.h"
      |          ^~~~~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs/denrgbcct/src/esphome/components/api/list_entities.cpp.o] Error 1
========================== [FAILED] Took 4.99 seconds ==========================
```

Since the issue goes away with 1.6.0, I think this PR may have caused the issue: https://github.com/libretiny-eu/libretiny/pull/292/files

This is using ESPHome 2025.2.2

@dwmw2 Is there an extra step that needs to be done for ESPHome since your PR changed? Sorry if I missed it somewhere

kuba2k2, 2025-03-10:
This issue is unlikely to be caused by LibreTiny, however you are right that lwIP was upgraded to 2.2.0 with this PR.

Try changing it back to 2.1.3 with this code:
```yaml
esphome:
  platformio_options:
    custom_versions.lwip: 2.1.3-bdk
```

DougEdey, 2025-03-10:
That doesn't fix the compile issue when moving LibreTiny up to 1.7.0:

```
INFO ESPHome 2025.2.2
INFO Reading configuration /config/esphome/den-rgb-cct.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing denrgbcct (board: generic-bk7231t-qfn32-tuya; framework: arduino; platform: libretiny @ 1.7.0)
--------------------------------------------------------------------------------
Warning! Non-Git installations are NOT SUPPORTED.
HARDWARE: BK7231T 120MHz, 256KB RAM, 1.03MB Flash
 - framework-arduino-api @ 2022.8.24+sha.237b10a 
 - framework-beken-bdk @ 0.0.0+v2021.06.07.sha.6491b8c 
 - library-flashdb @ 1.2.0+sha.d5c892f 
 - library-freertos @ 9.0.0+sha.cea0a88 
 - library-freertos-port @ 2023.5.23+sha.a917d93 
 - library-lwip @ 2.2.0-bdk+sha.48da4a2 
 - library-printf @ 6.1.0+sha.28a79bd 
 - library-uf2ota @ 5.0.0+sha.f955412 
PLATFORM VERSIONS:
 - libretiny @ 1.7.0
 - ltchiptool @ 4.11.6
CUSTOM OPTIONS:
 - fw_name = esphome
 - fw_version = 2025.2.2
 - versions:
     - lwip = 2.1.3-bdk
Dependency Graph
|-- AsyncTCP-esphome @ 2.1.4
|-- ESPAsyncWebServer-esphome @ 3.2.2
|-- DNSServer @ 1.1.0
|-- ArduinoJson @ 6.18.5
Compiling .pioenvs/denrgbcct/src/esphome/components/api/api_connection.cpp.o
Compiling .pioenvs/denrgbcct/src/esphome/components/api/api_frame_helper.cpp.o
Compiling .pioenvs/denrgbcct/src/esphome/components/api/api_pb2.cpp.o
Compiling .pioenvs/denrgbcct/src/esphome/components/api/api_pb2_service.cpp.o
In file included from src/esphome/components/socket/socket.h:6,
                 from src/esphome/components/api/api_frame_helper.h:14,
                 from src/esphome/components/api/api_frame_helper.cpp:1:
src/esphome/components/socket/headers.h:131:10: fatal error: lwip/sockets.h: No such file or directory
  131 | #include "lwip/sockets.h"
      |          ^~~~~~~~~~~~~~~~
compilation terminated.
In file included from src/esphome/components/socket/socket.h:6,
                 from src/esphome/components/api/api_frame_helper.h:14,
                 from src/esphome/components/api/api_connection.h:5,
                 from src/esphome/components/api/api_connection.cpp:1:
src/esphome/components/socket/headers.h:131:10: fatal error: lwip/sockets.h: No such file or directory
  131 | #include "lwip/sockets.h"
      |          ^~~~~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs/denrgbcct/src/esphome/components/api/api_connection.cpp.o] Error 1
*** [.pioenvs/denrgbcct/src/esphome/components/api/api_frame_helper.cpp.o] Error 1
========================= [FAILED] Took 11.84 seconds =========================
```

Cossid, 2025-03-10:
What environment are you compiling on?  I do not receive any errors with 2025.2.2 against framework 1.8.0 compiling on x64, including with your config.

DougEdey, 2025-03-10:
This is with ESPHome Builder running as an addon for HAOS 2025.3.1 on x86 (proxmox)

Cossid, 2025-03-10:
It sounds to me like you may have cancelled (or timed out) in the middle of downloading the lwip package for 1.7.0+ and with it partially extracted, it doesn't try to download again.  I'd suggest deleting your platformio cache and packages folders in your esphome builder cache directory.

DougEdey, 2025-03-10:
Thanks for the direction, clearing the platform io folder in the docker container forced the dependencies to be downloaded again which fixed the issue. I'm surprised there's no solution in ESPHome to do this, I may look to see if I can contribute since the flow I had to use isn't great

I'm putting this here for anyone else who hits this:

* Install the `Advanced SSH & Web Terminal` addon
* _Disable_ protected mode
* Open the terminal
* Get into the container: `docker exec -it $(docker ps --format '{{.Names}}'| grep esphome) /bin/bash`
* `rm -rf /data/cache/platformio/`

Go back to ESPHome builder and rebuild the target

Cossid, 2025-03-10:
I agree it is odd there is no way to do that in ESPHome builder, perhaps create a [feature request](https://github.com/esphome/feature-requests)