libretiny-eu/libretiny #317: [Undocummented feature?] Flashing beken without 3.3v connection

voed, 2025-03-23:
Hi guys. Not really an issue, but i have discovered something weird and maybe useful: seems like beken chips can be flashed without 3.3v. What's even more interesting it does not require shorting CEN pin.
I have no idea why it works, but it is:
1. Connect only TX/RX and GND pins to usb-ttl adapter
2. Open ltchiptool, select read or write
3. Press start and it will proceed the operation without power source and without manipulations with CEN

This dump was obtained like that.

[ltchiptool_bk72xx_2025-03-23_15-02-34.zip](https://github.com/user-attachments/files/19409902/ltchiptool_bk72xx_2025-03-23_15-02-34.zip)

kuba2k2, 2025-03-23:
It is possible if the device is connected to another power source (like batteries or USB input). Otherwise I don't see how would that work - microcontrollers need power supply to work.

voed, 2025-03-23:
No, it was not connected to any power source. I dunno how it works aswell, but it is. I was thinking it's some charge on the capacitor, but it does not explains why it does not requires CEN like that.

closed 2026-02-07 by kuba2k2