libretiny-eu/libretiny #334: Additional ADC pins for BK7231N

Savjee, 2025-09-25:
I recently bought an Tuya WiFi Thermostat that uses a BK7231N chip ([AliExpress listing](https://nl.aliexpress.com/item/1005009574418508.html)).

I was able to flash ESPHome onto the device by soldering leads onto the TX and RX pins.

I was also able to get in touch with the manufacturer who shared the list of PINs they used to connect everything to the MCU.

From this, I learned that the NTC sensor is attached to P21, which according to their diagram is ADC6 on the BK7231N.

[image]

This is odd, because I can't find any datasheets of the BK7231N with ADC6 marked on it. While inspecting the PCB, I can confirm that the NTC is indeed connected to P21.

Using this pin in my ESPHome configuration leads to an error. I patched ESPhome and libretiny to allow P21 as an ADC input.

In ESPHome's [`boards.py`](https://github.com/esphome/esphome/blob/dev/esphome/components/bk72xx/boards.py) file, I added ADC6: 
```json
"generic-bk7231n-qfn32-tuya": {
     "ADC6": 21
}
```

In libretiny `boards/_base/ic/bk7231-qfn32.json` I added the pin as well:
```json
{
	"pcb": {
		"ic": {
			"19": {
				"C_NAME": "GPIO21",
				"GPIO": "P21",
				"IRQ": null,
				"I2C": "1_SDA",
				"JTAG": "TMS",
				"I2S": "MCLK",
				"FLASH": "^FCS",
				"ADC": 6
			},
        ...
```

This got rid of the errors, but the ADC value is always 0. Even when I detach the NTC, which should make the ADC float.

Relevant parts of my ESPHome configuration: 

```
---
substitutions:
  devicename: "office-thermostat"
  friendly_name: "Office Thermostat"

packages:
  esphome: !include common/esphome.yaml
  api: !include common/api.yaml
  logger: !include common/logger.yaml
  wifi: !include common/wifi.yaml

bk72xx:
  board: "generic-bk7231n-qfn32-tuya"
  framework:
    version: 0.0.0
    source: https://github.com/Savjee/libretiny.git#master

sensor:
  - platform: wifi_signal
    name: "WiFi Signal Sensor"
    update_interval: 60s

  - platform: ntc
    sensor: resistance_sensor
    name: "Office Temperature"
    calibration:
      b_constant: 3380
      reference_temperature: 25°C
      reference_resistance: 10kOhm

  - platform: resistance
    id: resistance_sensor
    sensor: source_sensor
    configuration: DOWNSTREAM
    resistor: 100
    name: Resistance Sensor

  - platform: adc
    id: source_sensor
    # pin: P23
    pin: ADC6
    name: "ADC Raw value"
    unit_of_measurement: "V"
    update_interval: 1s

switch:
  - platform: gpio
    id: relay
    name: thermostat_relay
    pin: P22
```

Any thoughts on how I could fix this? Are there other changes I have to make to either ESPHome or Libretiny?

kuba2k2, 2025-09-25:
Interesting. I never noticed that BK7231N had some extra ADCs compared to BK7231T and others.

Where did you get that diagram from? It looks like part of a datasheet, yet we never found a BK7231N datasheet anywhere.

---

The `boards/_base/ic/bk7231-qfn32.json` file is not used on runtime, it is used to generate documentation and board support files. You'll need to modify a couple more files instead:

1. `cores/beken-72xx/arduino/src/wiring_analog.c`:
```c
#if CFG_SOC_NAME == SOC_BK7231N
static GPIO_INDEX adcToGpio[] = {
	-1,		 // ADC0 - VBAT
	GPIO26, // ADC1
	GPIO24, // ADC2
	GPIO23, // ADC3
	GPIO28, // ADC4
	GPIO22, // ADC5
	GPIO21, // ADC6
	GPIONUM, // ADC7
};
#else
```

2. `boards/variants/generic-bk7231n-qfn32-tuya.c`: (modify the relevant GPIOs only)
```c
	{GPIO26, PIN_GPIO | PIN_IRQ | PIN_ADC | PIN_PWM,            PIN_NONE, 0},
	{GPIO24, PIN_GPIO | PIN_IRQ | PIN_ADC | PIN_PWM,            PIN_NONE, 0},
	{GPIO23, PIN_GPIO | PIN_IRQ | PIN_ADC | PIN_JTAG,           PIN_NONE, 0},
	{GPIO28, PIN_GPIO | PIN_IRQ | PIN_ADC,                                PIN_NONE, 0},
	{GPIO22, PIN_GPIO | PIN_IRQ | PIN_ADC | PIN_JTAG,                     PIN_NONE, 0},
	{GPIO21, PIN_GPIO | PIN_IRQ | PIN_ADC | PIN_I2C | PIN_I2S | PIN_JTAG, PIN_NONE, 0},
```

kuba2k2, 2025-09-25:
Yes, this is pretty much non-invasive, it won't break any existing configurations, since it merely only enables support for the other ADCs.

Feel free to create a PR 😄 Just make sure to run `boardgen ltci` to regenerate the variant files after updating the JSON.

divadiow, 2025-09-25:
also referenced in the Tuya T34 doc

[T34 Module Datasheet_Tuya Developer Platform_Tuya Developer Platform.pdf](https://github.com/user-attachments/files/22547263/T34.Module.Datasheet_Tuya.Developer.Platform_Tuya.Developer.Platform.pdf)

divadiow, 2025-09-25:
`GPIONUM, // ADC7`

how come 'NUM'?

divadiow, 2025-09-25:
I see. I'm probably missing something in here, but isn't ADC7 IO13 for BK7231N?

[gpio_1.zip](https://github.com/user-attachments/files/22547569/gpio_1.zip)

kuba2k2, 2025-09-26:
Well, BK7231N doesn't even have IO13 on the pins. That's probably why there is no `#ifdef` in the SDK code, but judging by the assignment of other ADCs on non-N chips, it's probably only applicable to those.

It's interesting that all 7 ADCs have been properly connected to GPIOs in LibreTiny's code, but I'm not sure where I got that assignment from. Possibly BK7252 datasheet.